home-assistant/operating-system #3703: Ugreen Bluetooth 5.4 USB Dongle with external antenna not working

felipecrs, 2024-11-30:
### Describe the issue you are experiencing

Hi there,

I bought this Bluetooth adapter for use with my Home Assistant, but apparently Home Assistant does not support it yet.

https://aliexpress.com/item/1005006972424488.html

[image]

It is very cheap, yet it is very good. When plugged into Windows 11, Windows will download the driver automatically and it appears in the device manager as _BARROT Bluetooth 5.4 Adapter_. Its range is ended better than my previous Intel AX210 m.2 adapter.

Given it's very cheap and has long range support, I'd say it's a great fit for Home Assistant users.

The VID PID is: `33fa:0010` (VID_33FA&PID_0010)

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

14.0.rc2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Intel J4125

### Steps to reproduce the issue

https://github.com/user-attachments/assets/f5ed40be-07c2-47ca-b24c-1ea6df0240f0

### Anything in the Supervisor logs that might be useful for us?

```txt
No.
```

### Anything in the Host logs that might be useful for us?

```txt
There are some bluetooth mentions here:

2024-11-30 21:43:36.688 homeassistant kernel: hassio: port 14(veth8a73295) entered blocking state
2024-11-30 21:43:36.688 homeassistant kernel: hassio: port 14(veth8a73295) entered forwarding state
2024-11-30 21:43:36.694 homeassistant kernel: hassio: port 14(veth8a73295) entered disabled state
2024-11-30 21:43:36.698 homeassistant NetworkManager[440]: <info>  [1733003016.6988] manager: (veth8a73295): new Veth device (/org/freedesktop/NetworkManager/Devices/40)
2024-11-30 21:43:36.792 homeassistant kernel: audit: type=1325 audit(1733003016.791:884): table=nat:102 family=2 entries=1 op=nft_register_rule pid=9833 subj=unconfined comm="iptables"
2024-11-30 21:43:36.792 homeassistant kernel: audit: type=1300 audit(1733003016.791:884): arch=c000003e syscall=46 success=yes exit=508 a0=3 a1=7ffd10259d40 a2=0 a3=7ffd10259d2c items=0 ppid=533 pid=9833 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-nft-multi" subj=unconfined key=(null)
2024-11-30 21:43:36.792 homeassistant kernel: audit: type=1327 audit(1733003016.791:884): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D74006E6174002D4100444F434B4552002D7000746370002D6400302F30002D2D64706F72740033303031002D6A00444E4154002D2D746F2D64657374696E6174696F6E003137322E33302E33332E383A333030310000002D690068617373696F
2024-11-30 21:43:36.812 homeassistant kernel: audit: type=1325 audit(1733003016.807:885): table=nat:103 family=2 entries=1 op=nft_register_rule pid=9836 subj=unconfined comm="iptables"
2024-11-30 21:43:36.812 homeassistant kernel: audit: type=1300 audit(1733003016.807:885): arch=c000003e syscall=46 success=yes exit=600 a0=3 a1=7ffec6ee54b0 a2=0 a3=7ffec6ee549c items=0 ppid=533 pid=9836 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-nft-multi" subj=unconfined key=(null)
2024-11-30 21:43:36.812 homeassistant kernel: audit: type=1327 audit(1733003016.807:885): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D74006E6174002D4100504F5354524F5554494E47002D7000746370002D73003137322E33302E33332E38002D64003137322E33302E33332E38002D2D64706F72740033303031002D6A004D415351554552414445
2024-11-30 21:43:36.832 homeassistant kernel: audit: type=1325 audit(1733003016.831:886): table=filter:104 family=2 entries=1 op=nft_register_rule pid=9838 subj=unconfined comm="iptables"
2024-11-30 21:43:37.076 homeassistant systemd[1]: Started libcontainer container ea105bbdb1ba3888306a9f9061599e9125d0543332b07838fb3f98717e8d7b8d.
2024-11-30 21:43:37.295 homeassistant kernel: eth0: renamed from vetha129be4
2024-11-30 21:43:37.304 homeassistant kernel: hassio: port 14(veth8a73295) entered blocking state
2024-11-30 21:43:37.304 homeassistant kernel: hassio: port 14(veth8a73295) entered forwarding state
2024-11-30 21:43:37.311 homeassistant NetworkManager[440]: <info>  [1733003017.3107] device (veth8a73295): carrier: link connected
2024-11-30 21:43:41.685 homeassistant kernel: kauditd_printk_skb: 42 callbacks suppressed
2024-11-30 21:43:41.686 homeassistant kernel: audit: type=1325 audit(1733003021.683:901): table=filter:10 family=2 entries=3 op=nft_register_chain pid=10758 subj=docker-default comm="iptables"
2024-11-30 21:43:41.686 homeassistant kernel: audit: type=1300 audit(1733003021.683:901): arch=c000003e syscall=46 success=yes exit=432 a0=3 a1=7fffba50e210 a2=0 a3=0 items=0 ppid=10757 pid=10758 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/sbin/xtables-nft-multi" subj=docker-default key=(null)
2024-11-30 21:43:41.686 homeassistant kernel: audit: type=1327 audit(1733003021.683:901): proctitle=69707461626C6573002D4100464F5257415244002D6900776730002D6A00414343455054
2024-11-30 21:43:41.690 homeassistant kernel: audit: type=1325 audit(1733003021.688:902): table=filter:11 family=2 entries=1 op=nft_register_rule pid=10847 subj=docker-default comm="iptables"
2024-11-30 21:43:41.690 homeassistant kernel: audit: type=1300 audit(1733003021.688:902): arch=c000003e syscall=46 success=yes exit=288 a0=3 a1=7ffc29754940 a2=0 a3=0 items=0 ppid=10757 pid=10847 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/sbin/xtables-nft-multi" subj=docker-default key=(null)
2024-11-30 21:43:41.690 homeassistant kernel: audit: type=1327 audit(1733003021.688:902): proctitle=69707461626C6573002D4100464F5257415244002D6F00776730002D6A00414343455054
[... 278 lines cut ...]

oldest_recorder_run | November 22, 2024 at 3:08 PM
-- | --
current_recorder_run | November 30, 2024 at 6:42 PM
estimated_db_size | 735.32 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

<details><summary>Sonoff</summary>

version | 3.8.1 (ffa7e22)
-- | --
cloud_online | 2 / 2
local_online | 2 / 2

</details>

### Additional information

Bluetooth integration debug logs: 
[log.txt](https://github.com/user-attachments/files/17967099/log.txt)

Apparently, I'm not alone: https://github.com/home-assistant/core/issues/127539

sairon, 2024-12-04:
The device is not compatible with Linux (in fact, it's even mentioned in the product listing). Please refer to the [Bluetooth integration docs](https://www.home-assistant.io/integrations/bluetooth/#installing-a-usb-bluetooth-adapter) when selecting a Bluetooth adapter.

closed 2024-12-04 by sairon

felipecrs, 2024-12-04:
Thank you @sairon. I understand. I'll hope that maybe Linux will get this driver someday.

I pushed a PR to improve the docs: https://github.com/home-assistant/home-assistant.io/pull/36151