home-assistant/operating-system #3713: Stuck in Grub bootloader after update to OS14

steuerlexi, 2024-12-04:
### Describe the issue you are experiencing

After I updated to OS14 I am stuck on grub bootloader freezed after selection of the usual Slot B. 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

14.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

vmware esxi host 8.02

### Steps to reproduce the issue

1. upgrade to 14.0
2. hangs in slot - b
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
NA
```

### Anything in the Host logs that might be useful for us?

```txt
NA
```

steuerlexi, 2024-12-04:
rescue shell on slot b is not working either

steuerlexi, 2024-12-12:
[image]

dhrto, 2025-01-28:
I had this issue since HAOS 13.2, but finally found the problem in my case. Hopefully it can be of help for your troubleshooting.

I have two seemingly identical setups: generic x86-64 on ESXi 8.0, both with a Conbee II USB passthrough attached for Zigbee stuff.
One setup updated without any problems from 13.1 to 13.2 > 14.0 > 14.1 > 14,2.
The other setup refused to boot after going from 13.1 to 13.2. Later I tried going to the newer release version from 13.1. No dice every time, system hangs at boot slot screen (never have I appreciated the snapshot function of ESXi more). 

It turned out the culprit was that on this particular VM, I had somehow enabled the option 'Expose IOMMU to the guest OS' under the VM config > CPU. Easily missed, as I was certain both setups had exactly the same settings, and this option is only visible when the drop down menu is expanded. Turning this off and the upgrade went through without a hitch. Running 14.2 now.

I was doing some housekeeping of older Win10 VM's and by accident one of the VM's with this option enabled refused to boot from the setup .iso. Might be totally unrelated, but it did steer me in the right direction with HAOS.

steuerlexi, 2025-01-28:
I tried this, and yes, it solved the issue, but I'm curious about why this works. My research indicates that exposing the IOMMU doesn't provide noticeable advantages unless you're also exposing the GPU to the VM. Thanks for pointing me in that direction; it's very much appreciated.

dhrto, 2025-01-28:
Yeah, I also don't understand it, but I'm glad it solved the mystery for me and apparently also for you!

Seems to make no sense that IOMMU causes the lockup this early in the boot phase. 
I had it enabled by accident I guess. To my understanding IOMMU is only 'a thing' if you're messing with passthrough devices, like a GPU. I believe the USB passthrough I have for the Conbee II works on a different level.

closed 2025-03-24 by steuerlexi