home-assistant/operating-system #3714: Updating HAOS VM (since 13.1 -> 13.2) results in the VM never rebooting automatically

pcmike, 2024-12-04:
### Describe the issue you are experiencing

Ever since attempting to update my HAOS VM (running on proxmox) from 13.1 to 13.2 I've been having issues. The first time I attempted 13.1 -> 13.2 the system never restarted (I think there is a post somewhere about that) and I had no clue how to troubleshoot so I just restored a previous backup. Just last night I attempted the update again and this time I noted it hung on "target reached: system reboot." I manually rebooted the VM and 13.2 loaded just fine. I subsequently tried "cli host reboot" and the system rebooted just fine back into 13.2 without issue. I also didn't note anything of interest in the system logs.

This morning I saw the release of HAOS 14 and attempted to update, making sure to make a snapshot right before. This time the 13.2 -> 14 update got stuck at "Unmounting HassOS data partition..." and after about 5mins I just threw in the towel and rolled back to my snapshot. Per [https://github.com/home-assistant/operating-system/discussions/3711#discussioncomment-11461465](url) I'm raising this issue and providing the output of `blkid`

```
# blkid
/dev/zram1: LABEL="hassos-var" UUID="ae654cc8-45c2-4Ъ02-8447-f1cbd1414991" BL0CK_SI2E="4096" TYPЕ="ext4" /dev/zram2: LABEL="hassos-tmp" UUID="9f9f06f2-5bab-4cfa-b792-8ff5c05445ba" BLOCK_SI2E="4096" TYPE="ext4"
/dev/sda4: BLOCK_SI2E="131072" TYPE="squashfs" PARTLABEL="hassos-kerne11" PARTUUID="{c0Za4f0-5350-406f-93a2-56cbed636b5f"
/dev/sda2: BLOCK_SIZE="131072"
TYPE="squashfs" PARTLABEL="hassos-kerne10" PARTUUID="26700fc6-b0bc-4ccf-9837-ea1a4cba3e65"'
‹deu/sda?: LABEL="hassos-overlay" UUID="c10d6845-1927-460e-8b8b-5ec?7eaae9bc" BLOCK_SI2E="1024" TYPE="ext4"'
PARTLABEL="hassos-over lay" PARTUUID="{1326040-5236-4
Oeb-b683-aaa100a9afcf"
/dev/sda5: UUID="4f02096b-a917-49ad-8bf8-67e6739201b6"'
BLOCK_S1ZE="4096" TYPE="erofs" PARTLABEL="hassos-system1" PARTUUID="a3ec664e-32ce-4665-95ea-7ae90ce9aa20"
/deu/sda3: UUID="919de407-1b0d-451a-a765-3707c25925fd"
BLOCK_SIZE="4096" TYPE="erofs" PARTLABEL="hassos-system0" PARTUUID="8d3d53e3-6d49-4c38-8349-aff6859e82fd"
/dev/sda1: SEC_TYPE="msdos" LABEL_FATBOOT="hassos-boot" LABEL="hassos-boot" UUID="DC7E-DBF8" BLOCK_SIZE="512"
0952-733c-4c88-8cba-cab9b8b4377f"*
TYPE="ufat" PARTLABEL="hassos-boot" PARTUUID="b3dd
deu/sdaß: LABEL="hassos-data" UUID="584c1524-4afa-49bd-b590-c9d53f718dd5" BLOCK_SIZE="4096" TYPE="ext4" PARTLABEL="hassos-data" PARTUUID="a52a4597-fa3a-4851-ae
fd-2fbe9f849079"
/deu/sda6: PARTLABEL="hassos-bootstate" PARTUUID="33236519-7f32-4dff-8002-3390b62c309d"
#
```
@agne

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

13.2

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Proxmox 8.2.7

### Steps to reproduce the issue

1. Update HAOS from 13.2 to 14
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
[core-ssh ~]$ ha supervisor logs
2024-12-04 06:15:36.959 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
2024-12-04 06:15:36.959 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
2024-12-04 06:15:36.959 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
2024-12-04 06:15:36.959 INFO (MainThread) [supervisor.resolution.check] System checks complete
2024-12-04 06:15:36.959 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
2024-12-04 06:15:36.997 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-12-04 06:15:36.997 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2024-12-04 06:15:36.997 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
2024-12-04 06:16:44.077 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-12-04 06:46:44.150 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-12-04 07:15:37.004 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
2024-12-04 07:15:37.004 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
2024-12-04 07:15:37.004 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
2024-12-04 07:15:37.042 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
[... 258 lines cut ...]
version_api | ok
installed_addons | File editor (5.8.0), Matter Server (6.6.1), Mosquitto broker (6.4.1), Glances (0.21.1), Piper (1.5.2), Whisper (2.3.0), ESPHome Device Compiler (2024.11.2), Govee to MQTT Bridge (2024.07.13-82ddc6e9), Terminal & SSH (9.15.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 6
-- | --
resources | 10
views | 6
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | November 24, 2024 at 2:44 PM
-- | --
current_recorder_run | December 3, 2024 at 1:15 PM
estimated_db_size | 518.59 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

pcmike, 2024-12-04:
So, I literally just did the same procedure I used to successfully update from 13.1 to 13.2 and this time the update succeeded. Essentially, all I did was.. "host reboot" from the CLI and allow the VM to reboot successfully (it did) and then immediately take a snapshot (just incase) and then attempt an update to HAOS 14 and this time the update completed and rebooted successfully. Obviously there is entirely no way to capture everything that's printed in the console during the reboot, but I did make a screen recording of the console while that was doing its thing.

steuerlexi, 2024-12-04:
This is basically the same as my issue that i raised earlier today. 

#3713

gky008, 2024-12-05:
[image]
help me

closed 2025-03-15 by github-actions[bot]