home-assistant/operating-system #3717: HAOS wiped out network

jonlove234, 2024-12-04:
### Describe the issue you are experiencing

Via the GUI, updated to HAOS 14

Rebooted and no connectivity. Same when reverting back to 13.X

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

14.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

HAOS x86 running in Unraid VM

### Steps to reproduce the issue

1. Update to HAOS 14.0
2. Reboot

### Anything in the Supervisor logs that might be useful for us?

```txt
Attached network related screenshots
```

### Anything in the Host logs that might be useful for us?

```txt
Can't access CLI
```

### Additional information

[image]
[image]
[image]
[image]

jonlove234, 2024-12-04:
Restored a VM backup and worked, so going to close this and assume it was something unrelated to HAOS

jonlove234, 2024-12-04:
Can't repo

closed 2024-12-04 by jonlove234

matvit92, 2024-12-06:
I have the same issue, any other way to repair without restoring VM backup?