home-assistant/operating-system #3718: VirtualBox 7.1.2 / HA OS 13.2 - Bluetooth adapter causing lock up?

JonLaliberte, 2024-12-04:
### Describe the issue you are experiencing

I've been running this VM in VirtualBox for a while now, 2 or 3 years.

I don't remember why exactly, but I unfortunately ended up upgrading to HA OS 13.2 around the same time I switched from VirtualBox 6.1.50 (the latest v6 release) to 7.1.2 (the latest v7 release).

I may have accidentally updated VirtualBox when updating other things... I don't really remember now.
I tried downgrading back to 6.1.50, but I couldn't get the VM to boot. I tried all of the things with the extension pack, uninstalled everything, reinstalled, made sure my user was in the vboxusers group, etc... Just keeps failing.
I also tried moving over to KVM, but after configuring everything, converting the disk image, etc, it wouldn't boot at all and I gave up on it.

So, I'm stuck on 7.1+ for now.

Anyway, I've been having HA lock up. It happens 0-3 times per week.
Whenever it does, there are a some bluetooth related errors logged, but can't say for certain they are related.

Bluetooth adapter: https://www.amazon.com/gp/product/B09KGVFX1K
Note that I have had this adapter for over 2 years and never had this happen before the past month.

These are 3 different crashes:

[image]
[image]
[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

13.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

VirtualBox 7.1.2 on Ubuntu

### Steps to reproduce the issue

Unknown

### Anything in the Supervisor logs that might be useful for us?

```txt
Checked, nothing seemed important.
```

### Anything in the Host logs that might be useful for us?

```txt
[host_2024-12-04T20-05-21.696Z.txt](https://github.com/user-attachments/files/18013686/host_2024-12-04T20-05-21.696Z.txt)
```

### System information

Core 2024.11.3
Supervisor 2024.11.4
Operating System 13.2

### Additional information

VM logs from one of the crashes:
```
00:19:45.379872 Console: Machine state changed to 'Running'
160:10:46.242698 TM: Switching TSC mode from 'RealTSCOffset' to 'VirtTSCEmulated'
160:10:46.249130 OHCI: Software reset
160:10:46.290770 ACPI: Reset initiated by ACPI
[... 186 lines cut ...]
160:10:48.500059 PIIX3 ATA: Ctl#1: RESET, DevSel=0 AIOIf=0 CmdIf0=0xa0 (-1 usec ago) CmdIf1=0x00 (-1 usec ago)
160:10:48.500104 PIIX3 ATA: Ctl#1: finished processing RESET
160:10:49.232689 OHCI: USB Reset
160:10:49.291043 OHCI: Software reset
160:10:49.292575 OHCI: USB Operational
160:10:49.293645 EHCI: Hardware reset
160:10:49.293762 EHCI: USB Operational
160:10:49.301577 IEM: rdmsr(0xc0011029) -> #GP(0)
160:10:49.692952 EHCI: USB Suspended
160:10:50.998630 VMMDev: Guest Log: vboxguest: host-version: 7.1.4r165100 0x8000000f
160:10:50.999924 VMMDev: Guest Additions information report: Version 6.0.0 r127566 '6.0.0'
160:10:50.999967 VMMDev: Guest Additions information report: Interface = 0x00010004 osType = 0x00053100 (Linux >= 2.6, 64-bit)
160:10:51.000211 GUI: UISession::sltAdditionsChange: GA state doesn't really changed, still notifying listeners.
160:10:51.000220 GUI: UIMachineLogicNormal::sltCheckForRequestedVisualStateType: Requested-state=1, Machine-state=6
160:10:51.000317 VMMDev: Guest Additions capability report: (0x0 -> 0x0) seamless: no, hostWindowMapping: no, graphics: no
160:10:51.000350 VMMDev: Heartbeat checking timer has been stopped (rc=VINF_SUCCESS)
160:10:51.000362 VMMDev: Heartbeat flatline timer set to trigger after 4 000 000 000 ns
160:10:51.000413 GUI: UISession::sltAdditionsChange: GA state doesn't really changed, still notifying listeners.
160:10:51.000416 GUI: UIMachineLogicNormal::sltCheckForRequestedVisualStateType: Requested-state=1, Machine-state=6
160:10:51.000666 VMMDev: Guest Log: vbg_heartbeat_init: Setting up heartbeat to trigger every 2000 milliseconds
160:10:51.002670 GUI: UISession::sltAdditionsChange: GA state really changed, notifying listeners.
160:10:51.002693 GUI: UIMachineViewNormal::adjustGuestScreenSize: Adjust guest-screen size if necessary
160:10:51.002706 GUI: UIMachineLogicNormal::sltCheckForRequestedVisualStateType: Requested-state=1, Machine-state=6
190:57:48.131656 VMMDev: vmmDevHeartbeatFlatlinedTimer: Guest seems to be unresponsive. Last heartbeat received 4 seconds ago
```

JonLaliberte, 2024-12-04:
Couldn't seem to upload to the Host logs field, attached here.

[host_2024-12-04T20-05-21.696Z.txt](https://github.com/user-attachments/files/18013698/host_2024-12-04T20-05-21.696Z.txt)

I'm also testing with the bluetooth adapter disconnected now to see if stability improves.

JonLaliberte, 2024-12-17:
@sairon 
Thanks for the follow up.

I'm not 100% sure whether the problem started when I went from 13.1 to 13.2, or if I went from 12.4 to 13.1 to 13.2 in short succession (I usually skip .0 releases) and it happened with the 12 to 13 change. I can only (easily) switch back to 13.1 though.

Since removing the bluetooth adapter I haven't had any hangs, and since I have a bunch of ESP bluetooth proxies now, I'm not too worried about the direct connected adapter anymore.

Stan-Gobien, 2024-12-19:
The same thing happened to me with my Realtek bluetooth USB stick. I've had to unplug it to get HAOS to run stable.

I'm running on vmware ESXi btw.

closed 2025-03-28 by github-actions[bot]