home-assistant/operating-system #3722: Changing a fixed IP address also changes the subnetmask

GHBLoos, 2024-12-06:
### Describe the issue you are experiencing

I changed the (DHCP) IP address to a fixed IP address using the web ui. The setting was:
`IPv4 address  172.20.16.248`
`Subnetmask 255.255.255.0`
I changed the IPv4 address to `172.20.16.15`
The subnetmask changed to `255.255.255.255`
I didn't notice the change before I saved the new settings. After that I wasn't able to get back into the webui. 

Bug: unnecessary and unexpected change of the subnetmask.

[image]

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

14

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

RPi 5, 8GB, 
Ethernet connected
WLAN not connected

### Steps to reproduce the issue

1. new install using Raspberry Pi imager using the menu to install Home Assistant
2. Log in for the first time
3. Go to system->network
4. Change IPv4 to fixed addres (changing only the last digit of the IP address)
That's the moment the subnetmask changes

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

sairon, 2024-12-10:
I can confirm the issue is there but the Operating System has nothing to do with that. Would you mind filing it as a frontend bug please?

closed 2024-12-10 by github-actions[bot]

GHBLoos, 2024-12-10:
https://github.com/home-assistant/frontend/issues/23246