home-assistant/operating-system #3724: Frequent Freeze of HA Core since HAOS Update to 14.0

DerAutomatiker, 2024-12-08:
### Describe the issue you are experiencing

Since I've updated my HAOS Version from 13.2 to 14.0 my Home Assistant freezes multiple times a day. I've thought first it has something to do with HASS 2024.12.0 and reverted back to 2024.11.3 but the problem persisted.

I can access it with SSH and when I send ha core restart it comes back to live.

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

14.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Home Assistant Yellow
-Aeotec Z-Stick 7

### Steps to reproduce the issue

1. Start Home Assistant
2. Wait 2-24h hours
3. Home Assistant isnt reachable locally with IP or via Cloudflared but IP i s pingable
4. Send ha core restart or pull the power plug and insert again
5. Works again for 2-24h
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-12-08 12:55:52.583 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
2024-12-08 12:55:52.584 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
2024-12-08 12:55:52.654 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
2024-12-08 12:55:52.655 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
2024-12-08 12:55:52.655 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
2024-12-08 12:55:52.656 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
2024-12-08 12:55:52.656 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
2024-12-08 12:55:52.657 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
2024-12-08 12:55:52.658 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
2024-12-08 12:55:52.659 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
2024-12-08 12:55:52.660 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
2024-12-08 12:55:52.680 INFO (MainThread) [supervisor.resolution.checks.base] Run check for no_current_backup/system
2024-12-08 12:55:52.681 INFO (MainThread) [supervisor.resolution.check] System checks complete
2024-12-08 12:55:52.681 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
2024-12-08 12:55:53.052 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-12-08 12:55:53.053 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2024-12-08 12:55:53.053 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
2024-12-08 12:56:15.747 WARNING (MainThread) [supervisor.homeassistant.websocket] Connection is closed
```

### Anything in the Host logs that might be useful for us?

```txt
2024-12-08 11:56:15.052 homeassistant kernel: kauditd_printk_skb: 105 callbacks suppressed
2024-12-08 11:56:15.053 homeassistant kernel: audit: type=1701 audit(1733658975.046:464): auid=4294967295 uid=0 gid=0 ses=4294967295 subj=unconfined pid=15008 comm="python3" exe="/usr/local/bin/python3.13" sig=11 res=1
2024-12-08 11:56:15.075 homeassistant systemd[1]: Created slice Slice /system/systemd-coredump.
2024-12-08 11:56:15.080 homeassistant kernel: audit: type=1334 audit(1733658975.074:465): prog-id=112 op=LOAD
2024-12-08 11:56:15.081 homeassistant kernel: audit: type=1334 audit(1733658975.074:466): prog-id=113 op=LOAD
2024-12-08 11:56:15.081 homeassistant kernel: audit: type=1334 audit(1733658975.074:467): prog-id=114 op=LOAD
2024-12-08 11:56:15.103 homeassistant systemd[1]: Started Process Core Dump (PID 49395/UID 0).
2024-12-08 11:56:15.567 homeassistant systemd-coredump[49396]: Process 15008 (python3) of user 0 terminated abnormally without generating a coredump.
2024-12-08 11:56:15.572 homeassistant systemd[1]: systemd-coredump@0-49395-0.service: Deactivated successfully.
2024-12-08 11:56:15.724 homeassistant kernel: audit: type=1334 audit(1733658975.718:468): prog-id=114 op=UNLOAD
2024-12-08 11:56:15.725 homeassistant kernel: audit: type=1334 audit(1733658975.718:469): prog-id=113 op=UNLOAD
2024-12-08 11:56:15.725 homeassistant kernel: audit: type=1334 audit(1733658975.718:470): prog-id=112 op=UNLOAD
2024-12-08 11:56:15.747 homeassistant bluetoothd[541]: Adv Monitor app :1.86 disconnected from D-Bus
2024-12-08 11:56:18.884 homeassistant systemd[1]: docker-fd6929bbe5ef3ead6e775cc2e7e71041a640a240f5def68028011779cdc9d0d5.scope: Deactivated successfully.
2024-12-08 11:56:18.885 homeassistant systemd[1]: docker-fd6929bbe5ef3ead6e775cc2e7e71041a640a240f5def68028011779cdc9d0d5.scope: Consumed 28min 1.874s CPU time.
2024-12-08 11:56:18.921 homeassistant dockerd[608]: time="2024-12-08T11:56:18.921158005Z" level=info msg="ignoring event" container=fd6929bbe5ef3ead6e775cc2e7e71041a640a240f5def68028011779cdc9d0d5 module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
2024-12-08 11:56:18.990 homeassistant systemd[1]: var-lib-docker-overlay2-4d72aef3d77da33041c08c19bba60d864e4de14eb258b86b045e4cf18ac0b149-merged.mount: Deactivated successfully.
2024-12-08 11:56:18.990 homeassistant systemd[1]: mnt-data-docker-overlay2-4d72aef3d77da33041c08c19bba60d864e4de14eb258b86b045e4cf18ac0b149-merged.mount: Deactivated successfully.
2024-12-08 11:58:06.216 homeassistant kernel: audit: type=1334 audit(1733659086.211:471): prog-id=101 op=UNLOAD
```

### System information

## System Information

version | core-2024.12.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.0
os_name | Linux
os_version | 6.6.51-haos-raspi
arch | aarch64
timezone | Europe/Zurich
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.1
Stage | running
Available Repositories | 1479
Downloaded Repositories | 20

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.0
-- | --
update_channel | stable
supervisor_version | supervisor-2024.11.4
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 228.5 GB
disk_used | 25.7 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | yellow
supervisor_api | ok
version_api | ok
installed_addons | Matter Server (6.6.1), UniFi Network Application (3.5.0), Grafana (10.2.2), File editor (5.8.0), Terminal & SSH (9.15.0), Cloudflared (5.2.2), Home Assistant Google Drive Backup (0.112.1), Mosquitto broker (6.4.1), Z-Wave JS (0.9.0), Silicon Labs Multiprotocol (2.4.5), AdGuard Home (5.2.0), Studio Code Server (5.17.3), WireGuard (0.10.2), ESPHome Device Compiler (2024.11.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 8
-- | --
resources | 7
views | 18
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 4. Dezember 2024 um 20:17
-- | --
current_recorder_run | 8. Dezember 2024 um 13:59
estimated_db_size | 1495.58 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Additional information

Core Log:
[11:56:15] INFO: Home Assistant Core finish process exit code 256
[11:56:15] INFO: Home Assistant Core finish process received signal 11
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

ge8020, 2024-12-08:
i experience the same freezes on a generic x86-64 platform. After reboot everything works for a few hours, then freezes again.
rolled back to 13.2 for now.

DerAutomatiker, 2024-12-08:
Do I have to revert Hass to 2024.11.* also or is it possible only to rollback to HAOS 13.2 and stay at Hass 2024.12.1?

ge8020, 2024-12-08:
I am on OS 13.2 and 2024.12.1 now for about 20 hours without this freeze. 14.0 never ran longer than 10 hours. I will post an update, if it crashes again with this constellation.

DerAutomatiker, 2024-12-08:
Thanks. Also reverted back now.
But the Bug isn't already known? or have I overseen it?

falcon8, 2024-12-08:
I have the same issue here, in my case the x86-64 stops all together. Only a hard power reset wil bring it alive and then it runs for 24-30 hours before it repeats it self’s.
I hope there is a fix soon otherwise I also might to revert back to the former OS version

DerAutomatiker, 2024-12-09:
Reverting only HA Core didn't work. Reverting only HA Is didn't work too.... Now I reverted both late last night and for the last 8hrs it works, let's see.

ge8020, 2024-12-09:
> Reverting only HA Core didn't work. Reverting only HA OS didn't work too.... Now I reverted both late last night and for the last 8hrs it works, let's see.

Sorry to hear that. Mine is still running. OS 13.2 and 2024.12.1

jonpaterson, 2024-12-10:
The problem is definately with HA OS.  I have the issue within hours on both ProxMox and bare metal with OS 14.  I also get the bluetooth errors mentioned on another raised issue (unclear if they are related).  you can roll back with the command 'ha os update --version 13.2 '

Jpsy, 2024-12-10:
Can confirm that. HA 2024.12 with OS 13.2 is fine. 
The freezes are seemingly not related to the underlying hardware. I had them on RPi 5 with 8Gb RAM.
I guess that several instabilities described in different issue reports for 14.0 all boil down to one common root cause, i.e. a memory leak. Another symptom that I experienced every X hours on 14.0 was that my HA reported that configuration.yaml could not be found anymore.

falcon8, 2024-12-10:
> I have the same issue here, in my case the x86-64 stops all together. Only a hard power reset wil bring it alive and then it runs for 24-30 hours before it repeats it self’s. I hope there is a fix soon otherwise I also might to revert back to the former OS version

I have reverted back to os 13.2 and kept the 2024.12 build. So far no issues anymore. GIF it another 24 hours and I am sure it was the os14 causing it

sairon, 2024-12-10:
@DerAutomatiker The problem are these two lines that you provided in the additional information:
```
[11:56:15] INFO: Home Assistant Core finish process exit code 256
[11:56:15] INFO: Home Assistant Core finish process received signal 11
```

This means the main python process of HA Core crashed with segmentation fault. While we can never rule out it can be related to the OS upgrade, it is not that likely - that's also why the downgrade attempt didn't resolve the issue. The trigger for the segmentation fault can be something external which runs some invalid code causing the segfault. Focus on the Core logs rather than on Host and Supervisor, and also try running Core in the safe mode. If the problem persists, report the issue in the Core repository, and ping me there if any suspicion the OS is really somehow involved is confirmed.

---

For the others, please open another issue with all the requested details to track down your problems - the OP's issues is actually not about freezes but Core crashes. Also, please don't cherry-pick the logs, the context is often as much important as the line you think is the problem.

closed 2024-12-10 by sairon