home-assistant/operating-system #3725: Bluetooth HCI reset failed

Terikos, 2024-12-08:
### Describe the issue you are experiencing

Since upgrade core to version 14.0 on the console of my Intel NUC I have repeating message:
Bluetooth: hci: sending initial HCI reset failed (-19).
[image]

My NUC doesn’t have build Bluetooth. I tried to disable service but systemd file system is read only, so it is not possible.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.63-haos

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

X86 based hardware

### Steps to reproduce the issue

After start HAOS

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-12-08 17:51:28.800 homeassistant kernel: usb 1-5.1: New USB device found, idVendor=8087, idProduct=0a2a, bcdDevice= 0.01
2024-12-08 17:51:28.825 homeassistant kernel: usb 1-5.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
2024-12-08 17:51:28.830 homeassistant kernel: sdhci-pci 0000:00:12.0: SDHCI controller found [8086:2296] (rev 21)
2024-12-08 17:51:28.831 homeassistant kernel: sdhci-pci 0000:00:12.0: SDHCI controller found [8086:2296] (rev 21)
2024-12-08 17:51:28.875 homeassistant systemd[1]: Reached target Bluetooth Support.
2024-12-08 17:51:30.582 homeassistant kernel: usb 1-5.1: USB disconnect, device number 91
2024-12-08 17:51:30.590 homeassistant kernel: Bluetooth: hci0: sending initial HCI reset failed (-19)
2024-12-08 17:51:30.598 homeassistant systemd[1]: Stopped target Bluetooth Support.
2024-12-08 17:51:30.764 homeassistant kernel: usb 1-5.1: new full-speed USB device number 92 using xhci_hcd
2024-12-08 17:51:30.846 homeassistant kernel: usb 1-5.1: New USB device found, idVendor=8087, idProduct=0a2a, bcdDevice= 0.01
2024-12-08 17:51:30.865 homeassistant kernel: usb 1-5.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
2024-12-08 17:51:30.871 homeassistant kernel: sdhci-pci 0000:00:12.0: SDHCI controller found [8086:2296] (rev 21)
2024-12-08 17:51:30.919 homeassistant systemd[1]: Reached target Bluetooth Support.
2024-12-08 17:51:32.630 homeassistant kernel: usb 1-5.1: USB disconnect, device number 92
2024-12-08 17:51:32.641 homeassistant kernel: Bluetooth: hci0: sending initial HCI reset failed (-19)
2024-12-08 17:51:32.675 homeassistant systemd[1]: Stopped target Bluetooth Support.
2024-12-08 17:51:32.816 homeassistant kernel: usb 1-5.1: new full-speed USB device number 93 using xhci_hcd
2024-12-08 17:51:32.898 homeassistant kernel: usb 1-5.1: New USB device found, idVendor=8087, idProduct=0a2a, bcdDevice= 0.01
2024-12-08 17:51:32.923 homeassistant kernel: usb 1-5.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
2024-12-08 17:51:32.924 homeassistant kernel: sdhci-pci 0000:00:12.0: SDHCI controller found [8086:2296] (rev 21)
2024-12-08 17:51:32.925 homeassistant kernel: sdhci-pci 0000:00:12.0: SDHCI controller found [8086:2296] (rev 21)
2024-12-08 17:51:32.971 homeassistant systemd[1]: Reached target Bluetooth Support.
2024-12-08 17:51:34.678 homeassistant kernel: usb 1-5.1: USB disconnect, device number 93
2024-12-08 17:51:34.687 homeassistant kernel: Bluetooth: hci0: sending initial HCI reset failed (-19)
2024-12-08 17:51:34.695 homeassistant systemd[1]: Stopped target Bluetooth Support.
2024-12-08 17:51:34.862 homeassistant kernel: usb 1-5.1: new full-speed USB device number 94 using xhci_hcd
2024-12-08 17:51:34.944 homeassistant kernel: usb 1-5.1: New USB device found, idVendor=8087, idProduct=0a2a, bcdDevice= 0.01
2024-12-08 17:51:34.969 homeassistant kernel: usb 1-5.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
2024-12-08 17:51:34.976 homeassistant kernel: sdhci-pci 0000:00:12.0: SDHCI controller found [8086:2296] (rev 21)
2024-12-08 17:51:34.976 homeassistant kernel: sdhci-pci 0000:00:12.0: SDHCI controller found [8086:2296] (rev 21)
2024-12-08 17:51:35.020 homeassistant systemd[1]: Reached target Bluetooth Support.
2024-12-08 17:51:36.727 homeassistant kernel: usb 1-5.1: USB disconnect, device number 94
2024-12-08 17:51:36.735 homeassistant kernel: Bluetooth: hci0: sending initial HCI reset failed (-19)
2024-12-08 17:51:36.755 homeassistant systemd[1]: Stopped target Bluetooth Support.
2024-12-08 17:51:36.908 homeassistant kernel: usb 1-5.1: new full-speed USB device number 95 using xhci_hcd
2024-12-08 17:51:36.988 homeassistant kernel: usb 1-5.1: New USB device found, idVendor=8087, idProduct=0a2a, bcdDevice= 0.01
2024-12-08 17:51:37.009 homeassistant kernel: usb 1-5.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
2024-12-08 17:51:37.018 homeassistant kernel: sdhci-pci 0000:00:12.0: SDHCI controller found [8086:2296] (rev 21)
2024-12-08 17:51:37.018 homeassistant kernel: sdhci-pci 0000:00:12.0: SDHCI controller found [8086:2296] (rev 21)
[... 169 lines cut ...]
```

### Anything in the Host logs that might be useful for us?

```txt
N/a
```

### System information

Version	core-2024.12.1
Installation type	Home Assistant OS
Development	false
Supervisor	true
Docker	true
User	root
Virtual environment	false
Python version	3.13.0
Operating system family	Linux
Operating system version	6.6.63-haos
CPU architecture	x86_64

### Additional information

N/a

jonpaterson, 2024-12-10:
same issue on beelink MiniPC.  Service bluetoothd gets hung for several minutes on bare metal and ProxMox forwarded internal BT controller.

sairon, 2024-12-10:
@Terikos You say the NUC (which model is it exactly?) doesn't have Bluetooth but the kernel very much thinks so. Can you post the logs of the boot, i.e. the part of the host logs which start with something like "Linux version 6.6.63-haos" and the following ~10 seconds of logs.

Alternatively, if you don't want to use the integrated Bluetooth, there's usually option for disabling it somewhere in the BIOS menu.

---

@jonpaterson How come is this the same issue? Do you see the same (meaning exactly the same) messages? Proxmox OR bare metal install? At least please share more details about the details and symptoms.

musapinar, 2024-12-10:
Same here.
```Bluetooth: hci0: sending initial HCI reset failed (-32)```

Oracle VM Virtual Box, on ThinkPad T450s. Bluetooth is the one integrated to the laptop. No usb dongle whatsoever.

jonpaterson, 2024-12-10:
@sairon I saw the EXACT same issue on both hardware and Proxmox.  The serivice took around 5 mins to kill (likely the systemd timeout).  Here is the dmesg items that might be interesting on the hardware

[    2.424914] Bluetooth: hci0: Firmware timestamp 2023.48 buildtype 1 build 75324
[    2.439986] Bluetooth: hci0: Found device firmware: intel/ibt-0040-1050.sfi
[    2.440000] Bluetooth: hci0: Boot Address: 0x100800
[    2.440001] Bluetooth: hci0: Firmware Version: 60-48.23
[    2.440002] Bluetooth: hci0: Firmware already loaded

Terikos, 2024-12-10:
This is old NUC hardware without Bluetooth.

Intel BOXNUC5CPYH, N3050, DDR3L-1600, SATA3, HDMI, SDXC slot, USB 3.0, BOX

 I’ve already checked in BIOS and there is no any option for Bluetooth :(

alfwro13, 2024-12-30:
I had the same issue on proxmox on my NUC, I disabled Bluetooth in BIOS and I switched to using ESPHome as bluetooth proxy and everything is fine.

closed 2025-04-07 by github-actions[bot]

JaySea77, 2025-04-16:
Ever got this fixed or did you also give up and go to esphome?

BramPuts, 2025-04-18:
Same here, i dont know when exactly it has started.  i think the showed up after the upgrade to the new Kernel 
I have also the same NUC;  NUC5CPYH 

Home assistant is installed Generic x32-x64

in the system specs it should have BT 4.2 and i know that Bluetooth had worked because earlier this year my Bluetooth adapter did found a philips sonicare toothbrush in the neighbourhood 

it is giving me the error messages every second and it wont stop.   my console is also full with the message;
Bluetooth: hci0: sending initial HCI reset failed (-19) 
and when i try to type , it is almost impossible because evere second a new error message  -19 is given 

Supervisor log:
2025-04-18 15:48:52.491 WARNING (MainThread) [supervisor.hardware.monitor] Ignore device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5.1 / failes to initialize by udev

Host log:
2025-04-18 13:49:49.888 homeassistant kernel: usb 1-5.1: new full-speed USB device number 118 using xhci_hcd
2025-04-18 13:49:49.964 homeassistant kernel: usb 1-5.1: New USB device found, idVendor=8087, idProduct=0a2a, bcdDevice= 0.01
2025-04-18 13:49:49.980 homeassistant kernel: usb 1-5.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
2025-04-18 13:49:50.041 homeassistant systemd[1]: Reached target Bluetooth Support.
2025-04-18 13:49:51.754 homeassistant kernel: usb 1-5.1: USB disconnect, device number 118
2025-04-18 13:49:51.765 homeassistant kernel: Bluetooth: hci0: sending initial HCI reset failed (-19)
2025-04-18 13:49:51.782 homeassistant systemd[1]: Stopped target Bluetooth Support.

The NUC has a  Intel AC 3165 card with wireless(wifi) and Bluetooth.  As far i can read in a Linux kernel article the firmware/driver has been discontenioud / is not supported anymore ?  in the newest kernel 

 https://wireless.docs.kernel.org/en/latest/en/users/drivers/iwlwifi.html  

7260, 3160, 7265, 7265D, 3165 and 3168 support
Those devices will not be supported by the newest firmware versions: the last firmware that was released for 3160, 7260 and 7265 is -17.ucode. Bug fixes will be ported to -17.ucode. 7265D, 3165 and 3168’s latest firmware version is -29.ucode. In order to determine if your 7265 device is a ‘D’ version, you can check the dmesg output:

because i dont use bluetooth i want to turn it off somehow, but i dont know where to do it.  i the BIOS there is no option. 
is there a possiblity to turn it off in HA ? 
or is there a possible way to load a other driver ?  the 7265D

BramPuts, 2025-04-21:
NVM , i solved this issue by turning off the nuc, disconnect the power for 5 minutes  and plugin the power again and boot the system.  a reboot isnt enough

JaySea77, 2025-04-22:
I 'solved' it by moving to a Bluetooth proxy.

denix0, 2025-09-26:
I can confirm that cold boot (i.e. completely shutting down, waiting a bit and powering back up) does indeed resolve this issue.

BramPuts, 2025-09-26:
My problemen has been solved this way. Never had these errors anymore since... 
Why not give it a try?