home-assistant/operating-system #3726: DNS resolution failure

furynick, 2024-12-08:
### Describe the issue you are experiencing

Latest OS version brake name resolution within supervisor container

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 14.0

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

OVA deployed on ESXi 8.0

### Steps to reproduce the issue

1. Get OVA from github
2. Deploy OVA
3. Expand supervisor logs during first install
4. Download errors due to host not found occurs

Same procedure with 13.2 OVA do not produce any error in installation logs on the same ESXi, the problem is not related to router/firewall/proxy/anything else outside HAOS.

Issue may be the same with Generic x86 as I encounter this problem after upgrading my HA box but haven't tested to reinstall 13.2 on it.

### Anything in the Supervisor logs that might be useful for us?

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/udev.sh
[19:04:16] INFO: Using udev information from host
cont-init: info: /etc/cont-init.d/udev.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun supervisor (no readiness notification)
services-up: info: copying legacy longrun watchdog (no readiness notification)
[19:04:16] INFO: Starting local supervisor watchdog...
s6-rc: info: service legacy-services successfully started
2024-12-08 19:04:21.612 INFO (MainThread) [__main__] Initializing Supervisor setup
2024-12-08 19:04:21.637 INFO (MainThread) [supervisor.docker.network] Can't find Supervisor network, creating a new network
2024-12-08 19:04:21.967 INFO (MainThread) [supervisor.bootstrap] Setting up coresys for machine: qemux86-64
2024-12-08 19:04:21.977 INFO (MainThread) [supervisor.docker.supervisor] Attaching to Supervisor ghcr.io/home-assistant/amd64-hassio-supervisor with version 2024.11.4
2024-12-08 19:04:21.977 INFO (MainThread) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
2024-12-08 19:04:22.168 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state initialize
2024-12-08 19:04:22.172 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-12-08 19:04:22.173 INFO (MainThread) [__main__] Setting up Supervisor
2024-12-08 19:04:22.305 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
2024-12-08 19:04:22.536 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
2024-12-08 19:04:22.541 INFO (MainThread) [supervisor.dbus.manager] Connected to system D-Bus.
2024-12-08 19:04:22.542 INFO (MainThread) [supervisor.dbus.agent] Load dbus interface io.hass.os
2024-12-08 19:04:22.543 INFO (MainThread) [supervisor.dbus.hostname] Load dbus interface org.freedesktop.hostname1
2024-12-08 19:04:22.543 INFO (MainThread) [supervisor.dbus.logind] Load dbus interface org.freedesktop.login1
2024-12-08 19:04:22.544 INFO (MainThread) [supervisor.dbus.network] Load dbus interface org.freedesktop.NetworkManager
2024-12-08 19:04:22.544 INFO (MainThread) [supervisor.dbus.rauc] Load dbus interface de.pengutronix.rauc
2024-12-08 19:04:22.544 INFO (MainThread) [supervisor.dbus.resolved] Load dbus interface org.freedesktop.resolve1
2024-12-08 19:04:22.544 INFO (MainThread) [supervisor.dbus.systemd] Load dbus interface org.freedesktop.systemd1
2024-12-08 19:04:22.545 INFO (MainThread) [supervisor.dbus.timedate] Load dbus interface org.freedesktop.timedate1
2024-12-08 19:04:22.770 INFO (MainThread) [supervisor.host.services] Updating service information
2024-12-08 19:04:22.776 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
2024-12-08 19:04:22.780 WARNING (SyncWorker_0) [supervisor.host.sound] Can't update PulseAudio data: Failed to connect to pulseaudio server
[... 55 lines cut ...]
  cmdline: git clone -v --recursive --depth=1 --shallow-submodules -- https://github.com/hassio-addons/repository /data/addons/git/a0d7b954
  stderr: 'Cloning into '/data/addons/git/a0d7b954'...
fatal: unable to access 'https://github.com/hassio-addons/repository/': Could not resolve host: github.com
'.
2024-12-08 20:04:29.742 ERROR (MainThread) [supervisor.store] Can't retrieve data from https://github.com/hassio-addons/repository due to 
2024-12-08 20:04:29.742 INFO (MainThread) [supervisor.resolution.module] Create new suggestion execute_remove - store / a0d7b954
2024-12-08 20:04:29.742 INFO (MainThread) [supervisor.resolution.module] Create new issue fatal_error - store / a0d7b954
2024-12-08 20:04:29.742 ERROR (MainThread) [supervisor.store.git] Can't clone https://github.com/home-assistant/addons repository: Cmd('git') failed due to: exit code(128)
  cmdline: git clone -v --recursive --depth=1 --shallow-submodules -- https://github.com/home-assistant/addons /data/addons/core
  stderr: 'Cloning into '/data/addons/core'...
fatal: unable to access 'https://github.com/home-assistant/addons/': Could not resolve host: github.com
'.
2024-12-08 20:04:29.742 ERROR (MainThread) [supervisor.store] Can't retrieve data from core due to 
2024-12-08 20:04:29.742 INFO (MainThread) [supervisor.resolution.module] Create new suggestion execute_remove - store / core
2024-12-08 20:04:29.742 INFO (MainThread) [supervisor.resolution.module] Create new issue fatal_error - store / core
2024-12-08 20:04:29.747 INFO (MainThread) [supervisor.store] Loading add-ons from store: 0 all - 0 new - 0 remove
2024-12-08 20:04:29.747 INFO (MainThread) [supervisor.addons.manager] Found 0 installed add-ons
```

### Anything in the Host logs that might be useful for us?

```txt
2024-12-08 19:03:01.624 homeassistant kernel: Linux version 6.6.63-haos (builder@ab2803c4973f) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -g1d7407c6) 12.4.0, GNU ld (GNU Binutils) 2.40) #1 SMP PREEMPT_DYNAMIC Tue Dec  3 19:17:54 UTC 2024
2024-12-08 19:03:01.624 homeassistant kernel: Command line: BOOT_IMAGE=(hd0,gpt2)/bzImage root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootwait zram.enabled=1 zram.num_devices=3 systemd.machine_id= fsck.repair=yes systemd.condition-first-boot=true console=ttyS0 console=tty0 rauc.slot=A
```

sairon, 2024-12-10:
Is it only matter of the Supervisor container or are other things affected? Can you resolve github.com (and some other address, e.g. home-assistant.io) when you try to ping it - both in another container (i.e. in the console of the SSH add-on) and in the OS itself (after typing `login` in the VM console)? Anything suspicious in the [DNS logs](https://my.home-assistant.io/create-link/?redirect=logs&provider=dns)?

furynick, 2024-12-10:
Seems strange, I deployed a new instance of OVA (I downgraded to 13.2 to be able to install ESPhome) and found errors in supervisor logs but after some time the curl query is successful in supervisor.
Name resolution is working with nslookup in every containers.
The error is still present in homeassistant container :
[image]

Unless I use a ping before curl :
[image]

Here is an extract of some lines of dns log :
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/corefile.sh
cont-init: info: /etc/cont-init.d/corefile.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun coredns (no readiness notification)
s6-rc: info: service legacy-services successfully started
[WARNING] plugin/hosts: File does not exist: /config/hosts
.:53
.:5553
CoreDNS-1.8.7
linux/amd64, go1.22.2, a9adfd5-dirty
[INFO] 127.0.0.1:46542 - 34412 "HINFO IN 1850618784595787974.5365468787441417443. udp 68 true 2048" REFUSED qr,aa,rd 57 0.00067584s
[INFO] 127.0.0.1:50898 - 28900 "HINFO IN 1850618784595787974.5365468787441417443. udp 57 false 512" REFUSED qr,aa,rd 57 0.037653659s
[INFO] 172.30.32.2:58306 - 29865 "A IN github.com. udp 51 false 1232" FORMERR qr,rd 28 0.000982579s
[INFO] 172.30.32.2:58306 - 36957 "AAAA IN github.com. udp 51 false 1232" FORMERR qr,rd 28 0.001706591s
[INFO] 172.30.32.2:47237 - 59116 "A IN github.com. udp 51 false 1232" FORMERR qr,rd 28 0.000904346s
[INFO] 172.30.32.2:47237 - 63516 "AAAA IN github.com. udp 51 false 1232" FORMERR qr,rd 28 0.000946174s
[INFO] 172.30.32.2:33334 - 29938 "A IN github.com. udp 51 false 1232" FORMERR qr,rd 28 0.000516288s
[INFO] 172.30.32.2:33334 - 6734 "AAAA IN github.com. udp 51 false 1232" FORMERR qr,rd 28 0.000811305s
[INFO] 172.30.32.2:33955 - 2220 "A IN github.com. udp 51 false 1232" FORMERR qr,rd 28 0.000765866s
[INFO] 172.30.32.2:33955 - 62433 "AAAA IN github.com. udp 51 false 1232" FORMERR qr,rd 28 0.000694173s
[INFO] 172.30.32.2:39753 - 32127 "A IN github.com. udp 51 false 1232" FORMERR qr,rd 28 0.002257864s
[INFO] 127.0.0.1:49984 - 16282 "NS IN . udp 17 false 512" REFUSED qr,aa,rd 17 0.000232007s
[INFO] 127.0.0.1:52969 - 6533 "NS IN . udp 17 false 512" REFUSED qr,aa,rd 17 0.000139881s
[INFO] 127.0.0.1:51363 - 15375 "NS IN . udp 17 false 512" REFUSED qr,aa,rd 17 0.00043593s
[INFO] 127.0.0.1:36352 - 36629 "NS IN . udp 17 false 512" REFUSED qr,aa,rd 17 0.000272156s
[INFO] 127.0.0.1:50991 - 45374 "NS IN . udp 17 false 512" REFUSED qr,aa,rd 17 0.000182978s`

There's also this error with i/o timeout on localhost :
`[INFO] 172.30.32.1:53095 - 26959 "PTR IN 7.0.168.192.in-addr.arpa. udp 42 false 512" - - 0 6.006508282s
[ERROR] plugin/errors: 2 7.0.168.192.in-addr.arpa. PTR: read udp 127.0.0.1:45751->127.0.0.1:5553: i/o timeout`

Sudo-Rob, 2024-12-20:
I'm seeing unexpected entries in my DNS log.  To wit:

```
[INFO] 127.0.0.1:58321 - 6464 "NS IN . udp 17 false 512" REFUSED qr,aa,rd 17 0.000281048s
[INFO] 127.0.0.1:54751 - 6238 "A IN api.weather.gov. udp 44 true 2048" - - 0 5.005394174s
[ERROR] plugin/errors: 2 api.weather.gov. A: dial tcp 1.0.0.1:853: i/o timeout
[INFO] 127.0.0.1:40169 - 43443 "AAAA IN api.weather.gov. udp 44 true 2048" - - 0 5.011741762s
[ERROR] plugin/errors: 2 api.weather.gov. AAAA: dial tcp 1.1.1.1:853: i/o timeout
```

Why am I seeing Cloudflare requests when DNS is configured for a local server?

````
➜  ~ ha dns info
fallback: true
host: 172.30.32.3
llmnr: true
locals:
- dns://192.168.1.1
mdns: true
servers:
- dns://9.9.9.9
- dns://149.112.112.112
update_available: false
version: 2024.12.0
version_latest: 2024.12.0
````
Could this be related?

furynick, 2024-12-21:
I think this is related to hassio_dns.
Here's an extract of dns logs when running ping & curl, only requests from curl (udp 51) are failing if they not occur right after a successful request, requests from ping (udp 28) is working all the time.
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/corefile.sh
cont-init: info: /etc/cont-init.d/corefile.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun coredns (no readiness notification)
s6-rc: info: service legacy-services successfully started
[DEBUG] plugin/hosts: Parsed hosts file into 44 entries
.:53
.:5553
CoreDNS-1.11.4
linux/amd64, go1.23.3, 6e11ebd-dirty
[INFO] 127.0.0.1:35428 - 12049 "HINFO IN 5448401962731780789.584384620373096407. udp 56 false 512" REFUSED qr,aa,rd 56 0.000246631s
[INFO] 127.0.0.1:42308 - 57550 "HINFO IN 5448401962731780789.584384620373096407. udp 56 false 512" REFUSED qr,aa,rd 56 0.02310589s
[INFO] 172.30.32.2:47497 - 9478 "A IN github.com. udp 51 false 1232" FORMERR qr,rd 28 0.002513326s
[INFO] 172.30.32.2:47497 - 57539 "AAAA IN github.com. udp 51 false 1232" FORMERR qr,rd 28 0.002255442s
[INFO] 172.30.32.2:33551 - 8283 "A IN github.com. udp 28 false 512" NOERROR qr,rd,ra 54 0.01846202s
[INFO] 172.30.32.2:33551 - 8767 "AAAA IN github.com. udp 28 false 512" NOERROR qr,rd,ra 125 0.019607489s
[INFO] 172.30.32.2:35955 - 60146 "AAAA IN github.com. udp 51 false 1232" NOERROR qr,aa,rd,ra 125 0.000201505s
[INFO] 172.30.32.2:35955 - 49360 "A IN github.com. udp 51 false 1232" FORMERR qr,rd 28 0.001746618s
[INFO] 172.30.32.2:36715 - 13656 "AAAA IN github.com. udp 28 false 512" NOERROR qr,aa,rd,ra 125 0.000195258s
[INFO] 172.30.32.2:36715 - 12984 "A IN github.com. udp 28 false 512" NOERROR qr,rd,ra 54 0.017885897s
[INFO] 172.30.32.2:53506 - 19179 "A IN github.com. udp 51 false 1232" NOERROR qr,aa,rd,ra 54 0.000202172s
[INFO] 172.30.32.2:53506 - 24686 "AAAA IN github.com. udp 51 false 1232" NOERROR qr,aa,rd,ra 125 0.000117374s
```

closed 2025-03-30 by github-actions[bot]