home-assistant/operating-system #3740: HAOS 14: rebooting freeze

crug80, 2024-12-12:
### Describe the issue you are experiencing

Tested the scenario 2 times in different days: 

Ask for a system reboot from "developer tools" -> reboot the system

The system shutdown, but it remains off with the fan on.. 

I have to poweroff completely by removing the power, then it boots again. 

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

14

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

## System Information

version | core-2024.12.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.0
os_name | Linux
os_version | 6.6.51-haos-raspi
arch | aarch64
timezone | Europe/Rome
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.1
Stage | running
Available Repositories | 1486
Downloaded Repositories | 9

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.0
-- | --
[... 172 lines cut ...]
2024-12-12 18:32:06.958 jarvis systemd-time-wait-sync[634]: adjtime state 5 status 40 time Thu 2024-12-12 18:22:17.269030 UTC
2024-12-12 18:32:06.958 jarvis systemd-time-wait-sync[634]: adjtime state 0 status 2000 time Thu 2024-12-12 18:32:06.955008 UTC
2024-12-12 18:32:06.958 jarvis systemd-resolved[544]: Clock change detected. Flushing caches.
2024-12-12 18:32:06.958 jarvis systemd[1]: Started Network Time Synchronization.
2024-12-12 18:32:06.958 jarvis systemd[1]: Reached target System Time Set.
2024-12-12 18:32:06.958 jarvis systemd-resolved[544]: Clock change detected. Flushing caches.
2024-12-12 18:32:06.958 jarvis systemd-timesyncd[635]: Contacted time server 192.168.80.1:123 (192.168.80.1).
2024-12-12 18:32:06.958 jarvis systemd-timesyncd[635]: Initial clock synchronization to Thu 2024-12-12 18:32:06.954833 UTC.
2024-12-12 18:32:06.958 jarvis systemd[1]: Finished Wait Until Kernel Time Synchronized.
2024-12-12 18:32:06.958 jarvis systemd[1]: Reached target System Time Synchronized.
2024-12-12 18:32:06.959 jarvis systemd[1]: Started Discard unused filesystem blocks once a week.
2024-12-12 18:32:06.959 jarvis systemd[1]: Started Remove Bluetooth cache entries.
2024-12-12 18:32:06.960 jarvis systemd[1]: Reached target Timer Units.
2024-12-12 18:32:06.979 jarvis systemd[1]: Starting HassOS AppArmor...
2024-12-12 18:32:06.992 jarvis systemd[1]: Finished HassOS AppArmor.
2024-12-12 18:32:06.994 jarvis systemd[1]: Starting Docker Application Container Engine...
2024-12-12 18:32:06.995 jarvis kernel: audit: type=1400 audit(1734028326.986:12): apparmor="STATUS" operation="profile_load" profile="unconfined" name="hassio-supervisor" pid=644 comm="apparmor_parser"
2024-12-12 18:32:06.995 jarvis kernel: audit: type=1400 audit(1734028326.986:12): apparmor="STATUS" operation="profile_load" profile="unconfined" name="hassio-supervisor///usr/bin/gdbus" pid=644 comm="apparmor_parser"
2024-12-12 18:32:06.995 jarvis kernel: audit: type=1400 audit(1734028326.986:12): apparmor="STATUS" operation="profile_load" profile="unconfined" name="hassio-supervisor///usr/bin/git" pid=644 comm="apparmor_parser"
2024-12-12 18:32:06.995 jarvis kernel: audit: type=1300 audit(1734028326.986:12): arch=c00000b7 syscall=64 success=yes exit=39339 a0=6 a1=559c16de00 a2=99ab a3=1 items=0 ppid=643 pid=644 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="apparmor_parser" exe="/usr/sbin/apparmor_parser" subj=unconfined key=(null)
2024-12-12 18:32:06.995 jarvis kernel: audit: type=1327 audit(1734028326.986:12): proctitle=61707061726D6F725F706172736572002D72002D57002D4C002F6D6E742F646174612F73757065727669736F722F61707061726D6F722F6361636865002F6D6E742F646174612F73757065727669736F722F61707061726D6F722F68617373696F2D73757065727669736F72
2024-12-12 18:32:07.184 jarvis dockerd[649]: time="2024-12-12T18:32:07.184033162Z" level=info msg="Starting up"
2024-12-12 18:32:07.184 jarvis dockerd[649]: time="2024-12-12T18:32:07.184092347Z" level=warning msg="Running experimental build"
2024-12-12 18:32:07.243 jarvis kernel: audit: type=1400 audit(1734028327.238:13): apparmor="STATUS" operation="profile_load" profile="unconfined" name="docker-default" pid=671 comm="apparmor_parser"
```

sairon, 2024-12-17:
If you can reproduce it on every reboot, can you attach a HDMI display and check the output? What is the blinking pattern of the on-board LED what that happens?

crug80, 2024-12-17:
Not quickly. I need a micro hdmi cable to attach the monitor..  I'll try in the next days

sairon, 2024-12-20:
Although the original topic is slightly unrelated, see here: https://github.com/home-assistant/operating-system/issues/3720#issuecomment-2557210157

On the troublesome Pi with the old firmware I saw the same behavior, where it got stuck after a reboot and had to be power-cycled manually.

crug80, 2024-12-21:
> Although the original topic is slightly unrelated, see here: [#3720 (comment)](https://github.com/home-assistant/operating-system/issues/3720#issuecomment-2557210157)
> 
> On the troublesome Pi with the old firmware I saw the same behavior, where it got stuck after a reboot and had to be power-cycled manually.

I installed the nvme 2 months ago and I've updated to the latest firmware before to do it... So it should already update with a recent firmware. On monday I should receive the cable, so as first thing I will attach a monitor to see what happens

crug80, 2024-12-23:
Finally I got the cable... The leds remain fixed and this is the video log.
[image]

As you can see the problem is on NVME. Do you think that is it an HAOS matter?

sairon, 2025-01-02:
The boot gets stuck early in the bootloader, which points to some hardware/incompatibility issue. It's either the NVMe HAT, the SSD, or the combination of both. What drive and HAT are you using? I'm afraid it's not fixable on our side though.

crug80, 2025-01-02:
The strange thing is that it appeared only after the OS upgrade, I tested reboot several times before going in production. 
The hat is the Geekworm X1001 PCIe to M.2 NVMe SSD Shield with SanDisk SSD Plus 250GB, M.2 2280 PCIe Gen3. 
Anyway, the system is stable, so I can avoid to restart it..  :)

Pheoxy, 2025-01-03:
While I can't get any logs until I get to the residence housing it, I'm having a similar issue with a Raspberry Pi 4 with a USB SSD. Update was clunky to 14.0 and required a reboot but then seemed to work and 14.1 seemed to do the same but now it's not coming back up after another reboot recently during a scheduled restart and restarting again didn't seem to fix it.

I'll get logs if I can when I get there and also check the boot sequence on a screen, but seems to be the same issue so far.

Was a driver module removed or changed  with the kernel update?

linux4life798, 2025-01-28:
This issue has been occurring reliably for a while now. It may not help much, but there has been a lot of conversation about this issue in the home-assistant/core issue tracker https://github.com/home-assistant/core/issues/129714.

Th3OnlyN00b, 2025-01-29:
Worth noting [this issue](https://community.home-assistant.io/t/haos-bricked-with-14-1-update-proxmox-operation-mkdir-permission-denied/814709) appears to be related, as the symptoms are identical. I am able to reliably get logs, if that would be useful. In my case I'm attempting to update from 13.8 -> 14.2. Unsure if the issue is identical as the linked issue is on a Proxmark, but I'm on a pi5 getting the exact issue in the link.

sairon, 2025-01-29:
> This does not occur on cold start (hard power-off and power-on). The workaround is use a smart switch to kill power remotely using another smart home platform.

Bingo, I finally managed to reproduce it after several retries. The Pi, running from an SD card, became stuck in the bootloader, but the logs there were not particularly helpful (I could only see it was trying different boot modes as you correctly stated). Thankfully, the serial console was more chatty:

```
359.41 Boot mode: SD (01) order f24
359.58 SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
359.69 SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
359.80 OCR c0ff8000 [2]
CID: 009f544953443634476145500bcf017a
CSD: 400e00325b590001d1ef7f800a400000
359.87 SD: bus-width: 4 spec: 2 SCR: 0x02858083 0x00000000
359.95 SD HOST: 200000000 CTL0: 0x00800f04 BUS: 50000000 Hz actual: 50000000 HZ div: 4 (2) status: 0x1fff0000 delay: 2
359.04 MBR: 0x00000001,122142719 type: 0xee
359.07 MBR: 0x00000000,       0 type: 0x00
359.11 MBR: 0x00000000,       0 type: 0x00
359.15 MBR: 0x00000000,       0 type: 0x00
359.18 Trying partition: 0
359.21 GPT bf3cf6ad30b3d04096d4e82941e03d9f 000000001 00747bfff num-partitions 128 entry-size 128 pentry 2 first 0
359.31 GPT partition: 0 c12a7328-f81f-11d2-ba4b-00a0c93ec93b 000000800..0000207ff
359.38 Matched GUID bootable-part-idx 0 want partition 0
359.44 type: 16 lba: 2048 'mkfs.fat' '  V       ^ ' clusters 32695 (4)
359.50 rsc 4 fat-sectors 128 root dir cluster 1 sectors 32 entries 512
359.56 FAT16 clusters 32695
359.60 [sdcard] autoboot.txt not found
359.62 Select partition rsts 0 C(boot_partition) 0 EEPROM config 0 result 0
359.69 Trying partition: 0
359.71 GPT bf3cf6ad30b3d04096d4e82941e03d9f 000000001 00747bfff num-partitions 128 entry-size 128 pentry 2 first 0
359.82 GPT partition: 0 c12a7328-f81f-11d2-ba4b-00a0c93ec93b 000000800..0000207ff
359.88 Matched GUID bootable-part-idx 0 want partition 0
359.94 type: 16 lba: 2048 'mkfs.fat' '  V       ^ ' clusters 32695 (4)
359.00 rsc 4 fat-sectors 128 root dir cluster 1 sectors 32 entries 512
359.46 FAT16 clusters 32695
359.47 [sdcard] tryboot.txt not found
359.49 Retry SD 1
```

For some reason, the bootloader attempts to load the tryboot config instead of the default one, although that should only be attempted after an upgrade. Seems there is something wrong with setting the reboot parameter for it, this shouldn't happen. It explains why RPi OS and cold boot are working correctly, this is probably some userspace bug. It doesn't explain all the issues that were reported but it's a serious lead, maybe the wrong behavior could manifest differently in other cases (i.e. after the upgrade instead of host reboot).

closed 2025-01-30 by sairon

ausfas, 2025-02-16:
For me HAOS has been fine with these settings.
https://community.home-assistant.io/t/reboot-error-nvme-not-detected/781485/21?u=ausfas

Will wait for the proper HAOS fix/release to test it.

Gronaldo1, 2025-02-17:
How to know, which version will contain the fix?
Testing 25.2.4 still has same issue, for me.

user86000, 2025-02-17:
haos 14.3 i think, it is an os change not the core

user86000, 2025-02-17:
in console:
ha supervisor options --channel dev
ha restart
update supervisor and os to dev-Version in gui
in console back to:
ha supervisor options --channel stable

3 reboots on 2 different pi 5 without bootloop

Pheoxy, 2025-03-17:
> Will this change apply to all sorts of media when booting?
> I moved to Raspberry nvme card/ssd beginning of Nov 2024.
> The reason to migrate was that i used/tested with some Samsung SSD T7 that now and then failed to boot.
> Power cycling and they normally booted normally. 
> I then just assumed that the Samsung T7's was not really working with the RPI's

I have the same Samsung SSD with the Raspberry Pi 4 with the same problem. I'm hoping it's not the hardware to be honest.

uffevedenbrant, 2025-03-19:
15 is out now and i tested on 2 RPI5 yesterday.. About 10 reboots each and no failure!

linux4life798, 2025-03-19:
> 15 is out now and i tested on 2 RPI5 yesterday.. About 10 reboots each and no failure!

Same here. I have been testing across 3 RPi 5 instances, one booting from USB mass storage, and two from NVMe. So far, they all keep rebooting correctly. I'm very happy.

---

To reduce +1s, feel free to react with 🚀 if you are also finding that this patch/update (HA OS 15) fixed your issue.

PR #3842 (2025-01-30): Fix RPi 5 getting stuck in bootloader after some reboots
Probably since home-assistant/supervisor#5276 introduced in Supervisor 2024.9.0, RAUC bootloader handler for tryboot can set the tryboot flag also when the tryboot file is not present, causing the Pi to become stuck in bootloader, trying to load the tryboot file.

This happens when the device is already in the tryboot state, in that case the tryboot files and flag are created by set-primary and in turn the files are removed in set-state, while the flag is persisted, causing the bootloader to attempt loading non-existing file.

To avoid u [... cut]
commit ffca0b30 (2025-01-30): Fix RPi 5 getting stuck in bootloader after some reboots (#3842)

Probably since home-assistant/supervisor#5276 introduced in Supervisor
2024.9.0, RAUC bootloader handler for tryboot can set the tryboot flag also
when the tryboot file is not present, causing the Pi to become stuck in
bootloader, trying to load the tryboot file.

This happens when the device is already in the tryboot state, in that case the
tryboot files and flag are created by set-primary and in turn the files are
removed in set-state, while the flag is persisted, causing the bootloader to
attempt loading non-existing file.

T [... cut]