home-assistant/operating-system #3743: OS 14.0 Reboots caused by python seg faults.

tescophil, 2024-12-15:
### Describe the issue you are experiencing

HA Reboots due to python segmentation faults within a few minutes of booting

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

14.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Pi 5 8Gb, 256Gb NVME

### Steps to reproduce the issue

Upgrade to OS 14.0 results in constant boot cycle

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing unusual in the supervisor logs,
```

### Anything in the Host logs that might be useful for us?

```txt
This is the home-assistant.log.fault

Fatal Python error: Segmentation fault

Thread 0x0000007f6d144be8 (most recent call first):
  File "/usr/local/lib/python3.13/threading.py", line 363 in wait
  File "/usr/local/lib/python3.13/queue.py", line 213 in get
  File "/usr/local/lib/python3.13/site-packages/google/api_core/bidi.py", line 107 in __iter__
  File "/usr/local/lib/python3.13/site-packages/grpc/_channel.py", line 276 in consume_request_iterator
  File "/usr/local/lib/python3.13/threading.py", line 992 in run
  File "/usr/local/lib/python3.13/threading.py", line 1041 in _bootstrap_inner
  File "/usr/local/lib/python3.13/threading.py", line 1012 in _bootstrap

Thread 0x0000007f6eda8be8 (most recent call first):
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 89 in _worker
  File "/usr/src/homeassistant/homeassistant/components/recorder/executor.py", line 22 in _worker_with_shutdown_hook
  File "/usr/local/lib/python3.13/threading.py", line 992 in run
  File "/usr/local/lib/python3.13/threading.py", line 1041 in _bootstrap_inner
  File "/usr/local/lib/python3.13/threading.py", line 1012 in _bootstrap

Thread 0x0000007f6eec6be8 (most recent call first):
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 89 in _worker
  File "/usr/src/homeassistant/homeassistant/components/recorder/executor.py", line 22 in _worker_with_shutdown_hook
  File "/usr/local/lib/python3.13/threading.py", line 992 in run
  File "/usr/local/lib/python3.13/threading.py", line 1041 in _bootstrap_inner
  File "/usr/local/lib/python3.13/threading.py", line 1012 in _bootstrap

Thread 0x0000007f6efe4be8 (most recent call first):
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 89 in _worker
  File "/usr/src/homeassistant/homeassistant/components/recorder/executor.py", line 22 in _worker_with_shutdown_hook
  File "/usr/local/lib/python3.13/threading.py", line 992 in run
  File "/usr/local/lib/python3.13/threading.py", line 1041 in _bootstrap_inner
  File "/usr/local/lib/python3.13/threading.py", line 1012 in _bootstrap

Thread 0x0000007f7634cbe8 (most recent call first):
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 89 in _worker
  File "/usr/local/lib/python3.13/threading.py", line 992 in run
[... 828 lines cut ...]
resources | 6
views | 10
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 13 December 2024 at 04:06
-- | --
current_recorder_run | 15 December 2024 at 20:20
estimated_db_size | 144.62 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

<details><summary>Sonoff</summary>

version | 3.8.1 (ffa7e22)
-- | --
cloud_online | 11 / 14
local_online | 10 / 10

</details>

sairon, 2024-12-17:
Does the issue go away if you downgrade OS by switching the boot slots (`ha os boot-slot other`) or with  `ha os update --version 13.2`?

tescophil, 2024-12-17:
Hi, no..., tried versions 12.x and 13.x to no avail.

I've come to the conclusion that the Pi5 sucks, so ditched it, bought a $100 N100 based PC. 10 mins to setup and restore and it's running perfectly.

closed 2024-12-17 by sairon