home-assistant/operating-system #3756: HAOS 14.1 unable to boot HASO VM on proxmox

yousaf465, 2024-12-19:
### Describe the issue you are experiencing

Tried updating to Home Assistant OS 14.1 but stuck at boot.
[image]

[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

14

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Beelink EQ12 Intel N100 16GBram 512GB SSD
Proxmox 8.3 running HAOS VM

### Steps to reproduce the issue

1.Update to HAOS 14.1

### Anything in the Supervisor logs that might be useful for us?

```txt
nothing
```

### Anything in the Host logs that might be useful for us?

```txt
nothinh
```

### System information

## System Information

version | core-2024.12.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.0
os_name | Linux
os_version | 6.6.63-haos
arch | x86_64
timezone | Asia/Karachi
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4976
Installed Version | 2.0.1
Stage | running
Available Repositories | 1560
Downloaded Repositories | 48

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 15

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.0
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.0
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 88.5 GB
disk_used | 41.6 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | ADS-B Multi-Portal Feeder (1.32.1), AdGuard Home (5.2.1), Advanced SSH & Web Terminal (19.0.0), Castsponsorskip (0.8.0), Cloudcommander (18.4.1), EMQX (0.7.2), ESPHome Device Builder (2024.12.1), ESPHome Device Builder (beta) (2024.12.1), File editor (5.8.0), Filebrowser (2.23.0_14), Frigate (0.14.1), Glances (0.21.1), Home Assistant Google Drive Backup (0.112.1), Jellyfin Server (1.4.3), Jellyseerr (2.0.1), MQTT Explorer (browser-1.0.3), MariaDB (2.7.1), Matter Server (6.6.1), Mosquitto broker (6.4.1), Nginx Proxy Manager (1.0.1), Postgres 15 (15.7-4), Prowlarr (0.16.0), RTSPtoWeb - WebRTC (1.4.0), Radarr (0.14.0), Samba NAS-β (12.3.2-nas.beta84), Sonarr (0.3.0), Studio Code Server (5.17.3), Tailscale (0.24.0), Tailscale (1.78.1.0), TasmoAdmin (0.31.2), qBittorrent (5.0.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 6
-- | --
resources | 26
views | 34
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 8 December 2024 at 22:55
-- | --
current_recorder_run | 19 December 2024 at 20:13
estimated_db_size | 4006.87 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

<details><summary>Sonoff</summary>

version | 3.8.1 (ffa7e22)
-- | --
cloud_online | 10 / 19
local_online | 6 / 6

</details>

agners, 2024-12-19:
I've just updated two HAOS installations on Proxmox 8.3 to 14.1, so this seems to be a specific issue to your environment.

Looks like writing the update did not work? Did you wait for the reboot to happen or was there a problem after updating?

Does the other boot slot work when you manually select it?

yousaf465, 2024-12-19:
I was waiting for the boot to happen, for like 10mins and nothing happened.

When I looked at the console it was stuck at boot. 
Choose Slot A and boot back to HAOS 14. 

I tried updating again and it is still the same. tried three times. 
there seems to be plenty of storage available. 
[image]
[image]

steuerlexi, 2024-12-20:
I have the same issue also raised here #3713 . Interestingly, I have ESXi as hypervisor, and after the upgrade, it does not boot into Slot B, where the 14.x is located. Therefore, there must be something wrong with the GRUB bootloader configuration on the updated slot, which would be my.

freeDom-, 2024-12-20:
I have the exact same system (only 32 GB RAM) - I can try the update later and tell you if I encountered the same problem. Did you try to backup to the state before the update and reinstall it?

Seems that your file system was damaged during the installation process. Could be related to the disk that you are using and the way that you pass it to your VM.

gfn256, 2024-12-20:
> I have the same issue also raised here #3713

Yours was an update from 13.2 to 14.0.  It would appear that the OP successfully updated (previously) his to the 14.0 version.

Have any of you guys tried a fresh VM install on 14.1? This way we will know if it is data corruption or incompatibility.

steuerlexi, 2024-12-20:
> Have any of you guys tried a fresh VM install on 14.1? This way we will know if it is data corruption or incompatibility.

No, not yet.  But I guess that is the only thing left to fix this.

yousaf465, 2024-12-20:
> we will know if it is data corruption or incompatibility.

I can fire up another VM. 
How did you install it initially? I did use the community script.

steuerlexi, 2024-12-20:
So, I tried to set up a brand new VM with HAOS 14.1 the OVA version for esxi. The first time it booted but it showed some irc issue. Then i set up another one and it´s hanging in GRUB bootloader and doesnothing. I asume it has something to do with the updated linux kernel in HAOS 14.x

yousaf465, 2024-12-20:
increased Disc space but same issue
[image]

steuerlexi, 2024-12-20:
I decided to stay at 13.2 as i can not figure out where the issue is. I hope this is maybe fixed in the future.

freeDom-, 2024-12-20:
I did not encounter any issues with the upgrade, upgrading on Proxmox 8.3.0 from HA OS 14.0 to 14.1.
Using Beelink EQ12 Intel N100 32GBram 512GB SSD. It does seem to be a problem with the individual setup rather than a general problem though...

sairon, 2024-12-20:
@yousaf465 Why is there another drive (scsi1) in your config? What does `blkid` say (typed into VM console after switching to OS console using `login`) when you boot to the working slot?

yousaf465, 2024-12-20:
> @yousaf465 Why is there another drive (scsi1) in your config? What does `blkid` say (typed into VM console after switching to OS console using `login`) when you boot to the working slot?

Initially, I created the VM using the script, and it was 32GB, so I thought I had to add another drive to expand the storage. I am proxmox noob.

yousaf465, 2024-12-20:
> I did not encounter any issues with the upgrade, upgrading on Proxmox 8.3.0 from HA OS 14.0 to 14.1. Using Beelink EQ12 Intel N100 32GBram 512GB SSD. It does seem to be a problem with the individual setup rather than a general problem though...
> 
> I did the upgrade twice, once on a test VM I cloned from my productive VM and once using my productive VM. Both upgrades did not have any errors

It worked for me on my testing VM too, but it did not on my production VM.
my testing VM config 
[image]

sairon, 2024-12-20:
> let me check that blkid command

There you go - you have multiple partitions with the same label - such setup can not work (at least correctly). IIUC you are using a USB drive for the data - in that case an easy fix would be creating a new instance (you don't need to allocate a lot of space - the data partition would not be used in that case) and adopt the data disk to that. However, I'll still recommend doing a backup first, in case anything goes south.

yousaf465, 2024-12-20:
> There you go - you have multiple partitions with the same label - such setup can not work (at least correctly). IIUC you are using a USB drive for the data - in that case an easy fix would be [...]

I don't want to use USB drive as data drive. 
Just need to keep the IP for Nginx reverse proxy and pass through both my USB adapters plane tracker and sonoff Zigbee.

yousaf465, 2024-12-20:
any recommended setting for setup up HAOS in Proxmox, should I use 14GB ram or 8 is more than enough? what type of machine to choose q35 or other? SCSI controller to choose? etc

sairon, 2024-12-20:
> I don't want to use USB drive as data drive.

Oh, mea culpa, it's actually the other way around (I misunderstood context of the other screenshots). In that case the `scsi1` drive is the data disk and the USB drive contains a complete system with partitions that clash with the partitions on the other QEMU drive. However, the correct partitions are currently on the attached USB drive (as the sdc device contains erofs partitions). At this point the best option for you would be really to start from scratch and restore a backup - while it's fixable, the risk of failure is high if you're not 100% sure what you're doing.

Anyway, this is not an OS bug. Please use more appropriate channels for general discussion and support :pray:

closed 2024-12-20 by sairon

rwez, 2024-12-20:
I did have some issues with this upgrade. I had two upgrades, one for homeassistant core 2024.12.4->5 and one for ha o/s 14.0-14.1. When I tried to upgrade ha o/s from 14.0 to 14.1 it would try to upgrade, boot, but not load ha core. Tried 3 times. Only when I upgraded ha core to 2024.12.5 first, *then* ha o/s would upgrade, boot and load ha core. 
Using a Rpi 4 with a standard card, no additional partitions. Just a headsup for those encountering this, first upgrade core, then o/s.

rwez, 2024-12-20:
> any recommended setting for setup up HAOS in Proxmox, should I use 14GB ram or 8 is more than enough? what type of machine to choose q35 or other? SCSI controller to choose? etc

8 Gb is enough, but more doesn't hurt either if you can spare it.

yousaf465, 2024-12-21:
@sairon removed USB SSD 
[image]

honzis1364, 2024-12-21:
> I did have some issues with this upgrade. I had two upgrades, one for homeassistant core 2024.12.4->5 and one for ha o/s 14.0-14.1. When I tried to upgrade ha o/s from 14.0 to 14.1 it would try to [...]

Same issue on RPI5, only it did not help to upgrade to the latest core. Still freeze and I can't boot. The only thing to do is to reconnect the power delivery and it will boot into 14.0