home-assistant/operating-system #3759: Continues core dump process

jhalldn, 2024-12-19:
### Describe the issue you are experiencing

After upgrade the host log are repeating the same issue

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

14.1

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Zwave and zigbee dongles. 

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
2024-12-19 21:23:09.368 homeassistant kernel: audit: type=1334 audit(1734643389.364:27996): prog-id=11729 op=UNLOAD
2024-12-19 21:23:09.368 homeassistant kernel: audit: type=1334 audit(1734643389.364:27997): prog-id=11728 op=UNLOAD
2024-12-19 21:23:09.368 homeassistant kernel: audit: type=1334 audit(1734643389.364:27998): prog-id=11727 op=UNLOAD
2024-12-19 21:23:10.032 homeassistant kernel: audit: type=1701 audit(1734643390.028:27999): auid=4294967295 uid=0 gid=0 ses=4294967295 subj=docker-default pid=136726 comm="python3" exe="/usr/bin/python3.7" sig=6 res=1
2024-12-19 21:23:10.036 homeassistant kernel: audit: type=1334 audit(1734643390.032:28000): prog-id=11730 op=LOAD
2024-12-19 21:23:10.036 homeassistant kernel: audit: type=1334 audit(1734643390.032:28001): prog-id=11731 op=LOAD
2024-12-19 21:23:10.050 homeassistant systemd[1]: Started Process Core Dump (PID 136748/UID 0).
2024-12-19 21:23:10.249 homeassistant systemd-coredump[136749]: Process 136726 (python3) of user 0 terminated abnormally without generating a coredump.
2024-12-19 21:23:10.250 homeassistant systemd[1]: systemd-coredump@3878-136748-0.service: Deactivated successfully.
2024-12-19 21:23:11.042 homeassistant systemd[1]: Started Process Core Dump (PID 136798/UID 0).
2024-12-19 21:23:11.242 homeassistant systemd-coredump[136799]: Process 136776 (python3) of user 0 terminated abnormally without generating a coredump.
2024-12-19 21:23:11.244 homeassistant systemd[1]: systemd-coredump@3879-136798-0.service: Deactivated successfully.
2024-12-19 21:23:12.065 homeassistant systemd[1]: Started Process Core Dump (PID 136831/UID 0).
2024-12-19 21:23:12.235 homeassistant systemd-coredump[136832]: Process 136809 (python3) of user 0 terminated abnormally without generating a coredump.
2024-12-19 21:23:12.238 homeassistant systemd[1]: systemd-coredump@3880-136831-0.service: Deactivated successfully.
2024-12-19 21:23:13.062 homeassistant systemd[1]: Started Process Core Dump (PID 136866/UID 0).
2024-12-19 21:23:13.261 homeassistant systemd-coredump[136867]: Process 136844 (python3) of user 0 terminated abnormally without generating a coredump.
2024-12-19 21:23:13.264 homeassistant systemd[1]: systemd-coredump@3881-136866-0.service: Deactivated successfully.
2024-12-19 21:23:14.036 homeassistant kernel: kauditd_printk_skb: 25 callbacks suppressed
2024-12-19 21:23:14.036 homeassistant kernel: audit: type=1701 audit(1734643394.032:28027): auid=4294967295 uid=0 gid=0 ses=4294967295 subj=docker-default pid=136878 comm="python3" exe="/usr/bin/python3.7" sig=6 res=1
```

### System information

## System Information

version | core-2024.12.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.0
os_name | Linux
os_version | 6.6.62-haos-raspi
arch | aarch64
timezone | Europe/Stockholm
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 16 oktober 2025 02:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
cloud_ice_servers_enabled | true
remote_server | eu-central-1-23.ui.nabu.casa
certificate_status | ready
instance_id | 76b46863ec2e412c897588eeb50fdbe5
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.0
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 469.4 GB
disk_used | 9.8 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi5-64
supervisor_api | ok
version_api | ok
installed_addons | deCONZ (7.0.0), Samba share (12.3.2), phpMyAdmin (0.11.0), Terminal & SSH (9.16.0), File editor (5.8.0), Google Assistant SDK (2.5.0), Z-Wave JS (0.9.0), Matter Server (6.6.1), MariaDB (2.7.1), chrony (5.0.0), Mosquitto broker (6.4.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 0
views | 11
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 7 december 2024 06:53
-- | --
current_recorder_run | 19 december 2024 21:18
estimated_db_size | 306.64 MiB
database_engine | mysql
database_version | 10.11.6

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

DHowett, 2024-12-21:
I'm seeing similar behavior on a CM4 (Home Assistant Yellow) after the upgrade from 14.0 to 14.1.

Rolling back solved the issue.

Sergmasharov, 2025-01-09:
I cant update too. For a long time =)
[#3631 ](https://github.com/home-assistant/operating-system/issues/3631)

mtrimarchi, 2025-03-06:
Hi guys, I'm having the same issue on a Intel NUC (HAOS, not virtualized).

This problem seems to happen while opening the web interface and the mobile app.

This is the host log:

```
2025-03-06 19:31:07.250 homeassistant kernel: audit: type=1701 audit(1741289467.248:380): auid=4294967295 uid=0 gid=0 ses=4294967295 subj=unconfined pid=5590 comm="python3" exe="/usr/local/bin/python3.13" sig=11 res=1
2025-03-06 19:31:07.258 homeassistant kernel: audit: type=1334 audit(1741289467.257:381): prog-id=97 op=LOAD
2025-03-06 19:31:07.259 homeassistant kernel: audit: type=1334 audit(1741289467.257:382): prog-id=98 op=LOAD
2025-03-06 19:31:07.259 homeassistant kernel: audit: type=1334 audit(1741289467.257:383): prog-id=99 op=LOAD
2025-03-06 19:31:07.274 homeassistant systemd[1]: Started Process Core Dump (PID 7881/UID 0).
2025-03-06 19:31:07.428 homeassistant systemd-coredump[7882]: Process 5590 (python3) of user 0 terminated abnormally without generating a coredump.
2025-03-06 19:31:07.428 homeassistant systemd[1]: systemd-coredump@1-7881-0.service: Deactivated successfully.
2025-03-06 19:31:07.479 homeassistant kernel: audit: type=1334 audit(1741289467.478:384): prog-id=99 op=UNLOAD
2025-03-06 19:31:07.480 homeassistant kernel: audit: type=1334 audit(1741289467.478:385): prog-id=98 op=UNLOAD
2025-03-06 19:31:07.480 homeassistant kernel: audit: type=1334 audit(1741289467.478:386): prog-id=97 op=UNLOAD
2025-03-06 19:31:10.526 homeassistant systemd[1]: docker-56266e790c11e9cbc3fab78c93b0e7023c27c1c7663930e1a5f519bb913e6108.scope: Deactivated successfully.
2025-03-06 19:31:10.527 homeassistant systemd[1]: docker-56266e790c11e9cbc3fab78c93b0e7023c27c1c7663930e1a5f519bb913e6108.scope: Consumed 41.984s CPU time.
2025-03-06 19:31:10.542 homeassistant dockerd[537]: time="2025-03-06T19:31:10.541290281Z" level=info msg="ignoring event" container=56266e790c11e9cbc3fab78c93b0e7023c27c1c7663930e1a5f519bb913e6108 module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
2025-03-06 19:31:10.597 homeassistant systemd[1]: var-lib-docker-overlay2-a918048c223d3cf8281f00538cb311fb6674a2970591c25c5b27e839b897a3e8-merged.mount: Deactivated successfully.
2025-03-06 19:31:10.598 homeassistant systemd[1]: mnt-data-docker-overlay2-a918048c223d3cf8281f00538cb311fb6674a2970591c25c5b27e839b897a3e8-merged.mount: Deactivated successfully.
2025-03-06 19:31:10.708 homeassistant systemd[1]: Started libcontainer container 56266e790c11e9cbc3fab78c93b0e7023c27c1c7663930e1a5f519bb913e6108.
2025-03-06 19:31:10.717 homeassistant kernel: audit: type=1334 audit(1741289470.716:387): prog-id=100 op=LOAD
2025-03-06 19:31:10.718 homeassistant kernel: audit: type=1300 audit(1741289470.716:387): arch=c000003e syscall=321 success=yes exit=15 a0=5 a1=c00016d998 a2=78 a3=0 items=0 ppid=7963 pid=7972 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="runc" exe="/usr/bin/runc" subj=unconfined key=(null)
2025-03-06 19:31:10.718 homeassistant kernel: audit: type=1327 audit(1741289470.716:387): proctitle=72756E63002D2D726F6F74002F7661722F72756E2F646F636B65722F72756E74696D652D72756E632F6D6F6279002D2D6C6F67002F72756E2F636F6E7461696E6572642F696F2E636F6E7461696E6572642E72756E74696D652E76322E7461736B2F6D6F62792F35363236366537393063313165396362633366616237386339
2025-03-06 19:32:51.929 homeassistant kernel: kauditd_printk_skb: 12 callbacks suppressed
2025-03-06 19:32:51.929 homeassistant kernel: audit: type=1334 audit(1741289571.928:392): prog-id=92 op=UNLOAD
```

CristianGonzalezFernandez, 2025-04-12:
Hi guys, 

I am having the same issue.

[image]

I am running HAOS inside a VM with VirtualBox. Don't know what else could I check, but it is a real problem because home assistant is down when you need it and there is no way to restart it :(

closed 2025-07-19 by github-actions[bot]