home-assistant/operating-system #3763: Failed update 14.0->14.1 on Rpi5

serl, 2024-12-23:
### Describe the issue you are experiencing

I have HAOS installed on a Rpi5 on the SD card.

I tried to update HAOS from the HA interface, and I wasn't able to connect to it after around 10 minutes. Tried to turn it off manually by a single click on the power button, and after 5 minutes I kept the power button pressed until it became red, then clicked once again. It booted successfully in the 14.0 slot. Here's approximately the same thing as `ha os info` would say it:

```
➜  ~ ha os info
board: rpi5-64
boot: B
boot_slots:
  A:
    state: inactive
    status: bad
    version: "14.1"
  B:
    state: booted
    status: good
    version: "14.0"
data_disk: SD128-0x00113076
update_available: true
version: "14.0"
version_latest: "14.1"
```

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

14.1

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Standard Rpi5 installation on SD Card. I have a Sonoff dongle for Zigbee.

### Steps to reproduce the issue

1. Clicking update to install Home Assistant OS 14.1 on the Home Assistant interface

### Anything in the Supervisor logs that might be useful for us?

```txt
I guess that the timestamps here are a little bit off since it goes back in time(?). I include from the update download to the successful boot after the hard reset. So this should include the poweroff, the failed boot and the successful boot. I'm not sure on how to split the three of them.

[32m2024-12-22 10:34:50.733 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://os-artifacts.home-assistant.io/14.1/haos_rpi5-64-14.1.raucb[0m
[32m2024-12-22 10:35:17.847 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-14.1.raucb[0m
[32m2024-12-22 10:35:49.707 INFO (MainThread) [supervisor.os.manager] Install of Home Assistant Operating System 14.1 success[0m
[32m2024-12-22 10:35:49.710 INFO (MainThread) [supervisor.host.control] Initialize host reboot using logind[0m
[32m2024-12-22 10:35:49.710 INFO (MainThread) [supervisor.addons.manager] Phase 'application' stopping 0 add-ons[0m
[32m2024-12-22 10:35:49.714 INFO (SyncWorker_5) [supervisor.docker.manager] Stopping homeassistant application[0m
[32m2024-12-22 10:35:49.791 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API for a0d7b954_vscode closed[0m
[32m2024-12-22 10:36:13.178 INFO (MainThread) [supervisor.addons.manager] Phase 'services' stopping 3 add-ons[0m
[32m2024-12-22 10:36:13.187 INFO (SyncWorker_1) [supervisor.docker.manager] Stopping addon_a0d7b954_ssh application[0m
[32m2024-12-22 10:36:16.612 INFO (SyncWorker_1) [supervisor.docker.manager] Cleaning addon_a0d7b954_ssh application[0m
[32m2024-12-22 10:36:16.643 INFO (SyncWorker_4) [supervisor.docker.manager] Stopping addon_5c53de3b_esphome application[0m
[32m2024-12-22 10:36:21.116 INFO (SyncWorker_4) [supervisor.docker.manager] Cleaning addon_5c53de3b_esphome application[0m
[32m2024-12-22 10:36:21.224 INFO (SyncWorker_7) [supervisor.docker.manager] Stopping addon_a0d7b954_vscode application[0m
[32m2024-12-22 10:36:26.134 INFO (SyncWorker_7) [supervisor.docker.manager] Cleaning addon_a0d7b954_vscode application[0m
[32m2024-12-22 10:36:26.173 INFO (MainThread) [supervisor.addons.manager] Phase 'system' stopping 1 add-ons[0m
[32m2024-12-22 10:36:26.177 INFO (SyncWorker_5) [supervisor.docker.manager] Stopping addon_core_mosquitto application[0m
[32m2024-12-22 10:36:30.054 INFO (SyncWorker_5) [supervisor.docker.manager] Cleaning addon_core_mosquitto application[0m
[32m2024-12-22 10:36:30.077 INFO (MainThread) [supervisor.addons.manager] Phase 'initialize' stopping 0 add-ons[0m
[32m2024-12-22 10:36:30.077 INFO (MainThread) [supervisor.plugins.cli] Stopping cli plugin[0m
[32m2024-12-22 10:36:30.080 INFO (SyncWorker_3) [supervisor.docker.manager] Stopping hassio_cli application[0m
[... 378 lines cut ...]
<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 0
views | 6
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | December 12, 2024 at 18:07
-- | --
current_recorder_run | December 22, 2024 at 18:00
estimated_db_size | 532.77 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

### Additional information

Just to be sure: if I try again, I can always hard reboot and come back to 14.0, correct? 
I'm not very familiar with the inner workings of HAOS (and frankly, that's because it always worked like a charm 🚀)

marlonqpa, 2024-12-23:
Hi
I had the same. 
According to this https://github.com/home-assistant/operating-system/issues/3740#issuecomment-2557375242 comment  ugrading EEPROM helped me.

serl, 2024-12-23:
(Honestly I thought that the EEPROM thing was for NVME, and since I have an SD card didn't apply. I guess I was wrong 😁)

Quick question before I fetch a SD card to try this firmware thing, @marlonqpa before all the upgrading, you were on 14.0 or 13.x? Because for both #3740 and #3720 they talk about 14.0, which is working fine in my setup.

marlonqpa, 2024-12-23:
I tried to upgrade from 14.0 to 14.1 few times last week. With no success.
Today I downloaded latest Raspbery Pi OS, upgradred all it's components with apt upgrade and finally update EEPROM by rpi-eeprom-update -a,
Then switched back to HA and this time OS 14.1 was installed at once.

Auka84, 2024-12-24:
I had the same

I just reboot HA and then I tried again to update and success

kaovilai, 2024-12-25:
Same. I mentioned this [here](https://github.com/home-assistant/operating-system/discussions/3753#discussioncomment-11641272) previously.

serl, 2024-12-25:
From @Auka84:
> I just reboot HA and then I tried again to update and success

Actually, same. I rebooted from Developer Tools > Restart > Advanced options > Reboot system.
Then tried to update and it worked.
[Seems like we're not alone](https://github.com/home-assistant/operating-system/discussions/3753#discussioncomment-11662784).

From @Xelnah:
> The update 14.0 to 14.1 worked just now, so maybe the eeprom update did work after some power cycles.

Yep so I didn't try to update the EEPROM... The thing we have in common is the rebooting 🤔

From @kaovilai:
> Remotely checking the AT&T Nokia Router/Gateway(BGW320-505) via Smart Home Manager app reveals the device is no longer ethernet connected.
(from [here](https://github.com/home-assistant/operating-system/discussions/3753#discussioncomment-11641272))

This is something I actually checked when it crashed, and everything looked fine. I have a Fritz!Box (not sure if I was looking to a cache entry or something).

---

Given that I (and everyone else if I read correctly) managed to update to 14.1, I'm closing this

closed 2024-12-25 by serl

kaovilai, 2024-12-25:
I did not.l resolve. I'm away overseas and my ha is still disconnected.

profesor-jb, 2025-01-29:
Hello, 

I had same problem a month ago, and after find the reboot “for free” commented in this issue discussion, I was able to update 14.0 to 14.1. Now I  can’t update 14.1 to 14.2 after try same workaround a couple of times. Rpi4 64 with SSD card. When I try to update to 14.2 system restarts but I am not able to connect with it. I have to force a hard reset (green button in my Rpi5) in order to restart and recover control of HAOS.

boot_slot a bad, 14.2 and boot_slot b(booted) good, 14.1. Don’t know why this happens, or how to update without a complex operation. 

I do suggest to reopen this issue discussion.

serl, 2025-01-30:
Since the issue is about updating to 14.1 and your issue is about updating to 14.2, I'd suggest to open a new one..?

Anyways, my current situation is that I rolled back to 13.2, since the system didn't survive a reboot at some point. I didn't reopen at that point, thinking that I was covered already by #3740 or #3720. I see that they are closed now, but I don't see this PR #3842 (which is announced to fix #3740) in the changelog for 14.2.
TLDR: my current plan is waiting for the next release and try again

profesor-jb, 2025-01-31:
> Hello,
>
> I had same problem a month ago, and after find the reboot “for free” commented in this issue discussion, I was able to update 14.0 to 14.1. Now I can’t update 14.1 to 14.2 after try same [...]

Sorry, Rpi 5, not 4