home-assistant/operating-system #3766: Booting image on cm5 pi stalls

AZSD1-6, 2024-12-28:
### Describe the issue you are experiencing

New CM5 4GB, 32GB emmc Pi on Pi IO 5 board (no peripherals) when loading HA image (both from the imager and custom load) pi boots and hangs after -

finds boot device then stalls after this line. (did leave for various hours)
mmc1: new ultra high speed ddr50 sdio card at address 0001. (no other data shown after this line, no error etc justs stops)

Regular Pi software loads without issue, HA image loads fine on Pi5, trialed various power supplies including 27W pi block, trialed various HDMI monitors, trialed various USB cables including one that is certified.

Trialed wiping eMMC and reloading, trialed updating EEProm and reloading with no change. Also trialed rc image and prior version of image and it is present.

Can not find any other notation of this error so thought I would post here as it does not seem to be a hardware issue rather a configuration issue. I know the cm5 is fairly new.

thanks

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

14.1

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

New CM5 4GB, 32GB emmc Pi on new Pi IO 5 board

### Steps to reproduce the issue

1. set pins to no emmc boot, boot pi
2. ripboot to set to mass storage
3.use Pi imager to select pi5 HAOSs 14.1
4. choose emmc 
5. flash OS 
6. boot pi with or without peripherals plugged including hdmi monitor (tired various hdmi cords, monitors and power supplies all reproduce issue
7. same as above using custome OS load in pi imager produces same issue 

Issue is not present in regular pi software load which works as expected. 
...

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

AZSD1-6, 2024-12-30:
Should add loads fine from NVMe. So trialed HAOS and only loads on USB and NVMe, but will not load from emmc. Also, regular raspberry os will load from all three.

hukm, 2025-01-01:
Having exactly the same issue

slankes, 2025-01-03:
Sorry for spamming this issue. I'm seeing the same. Unit works fine from NVME, but not from EMMC.

@AZSD1-6 / @hukm I am also seeing that connecting things to the usb-ports fails for me. Can you confirm that when booted from NVME in HAOS anything connected to the usb ports (I'm using the Compute Module 5 IO Board) doesn't work?

hukm, 2025-01-03:
@slankes my Zigbee dongle works just fine so I am not sure these issues are connected

AZSD1-6, 2025-01-04:
I have tried usb storage and ZBT-1 zigbee and both work with HAOS loaded on nvme

PR #3782 (2025-01-09): Add all BCM2712 device trees to RPi 5 image
RPi 5 images container only device tree for Pi 5 Model B. Add the other remaining BCM2712 device trees to enable running on CM5 and other variants supported upstream.

Fixes #3766

## Summary by CodeRabbit

- **New Features**
	- Expanded Linux kernel device tree support for multiple Raspberry Pi 5 and Compute Module 5 variants
	- Added configuration for different Raspberry Pi 5 board models and compute module configurations
commit c7e21c0c (2025-01-09): Add all BCM2712 device trees to RPi 5 image (#3782)

RPi 5 images container only device tree for Pi 5 Model B. Add the other
remaining BCM2712 device trees to enable running on CM5 and other variants
supported upstream.

Fixes #3766
closed 2025-01-09 by sairon