home-assistant/operating-system #3768: HAOS reverse DNS takes 7.5s before it fails, making lots of things painfully slow

Ten0, 2024-12-29:
### Describe the issue you are experiencing

Logging in to the HAOS host (`https://developers.home-assistant.io/docs/operating-system/debugging/`), I can see that:
1. Depending on the network I'm coming from, it takes forever to login. (This is because the SSH server makes reverse DNS lookups)
2. `iptables -L` is extremely slow for all the `Chain DOCKER` section (same delay of 7.5s as above for every line of that section). Running it without the reverse DNS lookups is fast: `iptables -L -n`
3. This is the `systemd-resolved` part that is slow. Indeed `resolvectl query 172.30.33.1` takes 7.5s to return.
 
`/etc/nssiwtch.conf` (root path for hostname resolution) delegates to systemd-resolved via the `resolve` statement here:
```apacheconf
hosts:          resolve [!UNAVAIL=return] files myhostname dns
```
Removing `resolve [!UNAVAIL=return]` from there, this removes usage of `systemd-resolved`, and `iptables` becomes fast again. (had to `mount --bind` the file to test this because otherwise fs is read-only)

`/etc/systemd/resolved.conf` contains:
```ini
[Resolve]
#DNS=
#FallbackDNS=1.1.1.1 8.8.8.8 1.0.0.1 8.8.4.4 2606:4700:4700::1111 2001:4860:4860::8888 2606:4700:4700::1001 2001:4860:4860::8844
#Domains=
DNSSEC=no
DNSOverTLS=no
#MulticastDNS=yes
#LLMNR=yes
#Cache=yes
DNSStubListener=no
#ReadEtcHosts=yes
#ResolveUnicastSingleLabel=no
```

Adding to this configuration file (bind mount again) then `systemctl restart systemd-resolved`:
```ini
MulticastDNS=no
LLMNR=no
```
solves the issue, it's very fast again.
Specifically, LLMNR seems to be responsible for 0.5s of the delay (I still get a painful 0.5s delay per `iptables` line), while `mDNS` seems to be responsible for 7s of it.

It is notable that the startup log of systemd-resolved gives:
```
homeassistant systemd-resolved[1478909]: mDNS-IPv4: There appears to be another mDNS responder running, or previously systemd-resolved crashed with some outstanding transfers.
homeassistant systemd-resolved[1478909]: mDNS-IPv6: There appears to be another mDNS responder running, or previously systemd-resolved crashed with some outstanding transfers.
```

### What operating system image do you use?

generic-aarch64 (Generic UEFI capable aarch64 systems)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 13.2.dev20241104

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

I'm running on a rockpi-4a-plus via https://github.com/citruz/haos-rockpi

### Steps to reproduce the issue

see above

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
2024-12-29 15:44:32.301 homeassistant systemd[1]: Stopping Network Name Resolution...
2024-12-29 15:44:32.312 homeassistant systemd[1]: systemd-resolved.service: Deactivated successfully.
2024-12-29 15:44:32.313 homeassistant systemd[1]: Stopped Network Name Resolution.
2024-12-29 15:44:32.331 homeassistant kernel: audit: type=1334 audit(1735487072.317:7235): prog-id=2108 op=LOAD
2024-12-29 15:44:32.376 homeassistant systemd[1]: Starting Network Name Resolution...
2024-12-29 15:44:32.381 homeassistant kernel: audit: type=1334 audit(1735487072.367:7236): prog-id=2107 op=UNLOAD
2024-12-29 15:44:32.730 homeassistant systemd-resolved[1478909]: Positive Trust Anchors:
2024-12-29 15:44:32.731 homeassistant systemd-resolved[1478909]: . IN DS 20326 8 2 e06d44b80b8f1d39a95c0b0d7c65d08458e880409bbc683457104237c7f8ec8d
2024-12-29 15:44:32.731 homeassistant systemd-resolved[1478909]: Negative trust anchors: home.arpa 10.in-addr.arpa 16.172.in-addr.arpa 17.172.in-addr.arpa 18.172.in-addr.arpa 19.172.in-addr.arpa 20.172.in-addr.arpa 21.172.in-addr.arpa 22.172.in-addr.arpa 23.172.in-addr.arpa 24.172.in-addr.arpa 25.172.in-addr.arpa 26.172.in-addr.arpa 27.172.in-addr.arpa 28.172.in-addr.arpa 29.172.in-addr.arpa 30.172.in-addr.arpa 31.172.in-addr.arpa 170.0.0.192.in-addr.arpa 171.0.0.192.in-addr.arpa 168.192.in-addr.arpa d.f.ip6.arpa ipv4only.arpa resolver.arpa corp home internal intranet lan local private test
2024-12-29 15:44:32.736 homeassistant systemd-resolved[1478909]: Using system hostname 'homeassistant'.
2024-12-29 15:44:32.737 homeassistant systemd-resolved[1478909]: mDNS-IPv4: There appears to be another mDNS responder running, or previously systemd-resolved crashed with some outstanding transfers.
2024-12-29 15:44:32.738 homeassistant systemd-resolved[1478909]: mDNS-IPv6: There appears to be another mDNS responder running, or previously systemd-resolved crashed with some outstanding transfers.
2024-12-29 15:44:32.743 homeassistant systemd[1]: Started Network Name Resolution.
```

### System information

## System Information

version | core-2024.12.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.0
os_name | Linux
os_version | 6.6.54-haos
arch | aarch64
timezone | Europe/Paris
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.2.dev20241104
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.0
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 13.5 GB
disk_used | 5.5 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rockpi-4a-plus
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (19.0.0), File editor (5.8.0), Home Assistant Google Drive Backup (0.112.1), Mosquitto broker (6.4.1), Zigbee2MQTT (1.42.0-2), AppDaemon (0.16.7), WireGuard (0.10.2), Duck DNS (1.18.0), NGINX Home Assistant SSL proxy (3.11.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 0
views | 1
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | December 13, 2024 at 15:28
-- | --
current_recorder_run | December 28, 2024 at 03:21
estimated_db_size | 24.19 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

Ten0, 2025-04-01:
Still here.
To reproduce:
```
ssh -p 22222 haos
iptables -L
```
super slow when looking at the DOCKER chain.

sbraz, 2025-05-25:
Hi,
> Depending on the network I'm coming from, it takes forever to login

I noticed the same. My tests (looking at packet captures) show that PTR queries are **only sent if the client is in the same subnet as the server**.
In my case, the DNS server quickly replies to the PTR query with NXDomain, but no reply is sent to the subsequent mDNS PTR query, hence the timeout.
Example tcpdump output for `192.168.0.1` (also acting as DNS server) SSH'ing into HAOS at `192.168.0.3`:
```
20:22:40.682241 vboxnet0 In  IP 192.168.0.3.59641 > 192.168.0.1.53: 62065+ [1au] PTR? 1.0.168.192.in-addr.arpa. (54)
20:22:40.682327 vboxnet0 Out IP 192.168.0.1.53 > 192.168.0.3.59641: 62065 NXDomain 0/0/1 (54)
20:22:40.682337 vboxnet0 Out IP 192.168.0.3.5353 > 224.0.0.251.5353: 0 PTR (QM)? 1.0.168.192.in-addr.arpa. (43)
20:22:40.682340 vboxnet0 M   IP 192.168.0.3.5353 > 224.0.0.251.5353: 0 PTR (QM)? 1.0.168.192.in-addr.arpa. (43)
20:22:41.736830 vboxnet0 Out IP 192.168.0.3.5353 > 224.0.0.251.5353: 0 PTR (QM)? 1.0.168.192.in-addr.arpa. (43)
20:22:41.736835 vboxnet0 M   IP 192.168.0.3.5353 > 224.0.0.251.5353: 0 PTR (QM)? 1.0.168.192.in-addr.arpa. (43)
20:22:43.991266 vboxnet0 Out IP 192.168.0.3.5353 > 224.0.0.251.5353: 0 PTR (QM)? 1.0.168.192.in-addr.arpa. (43)
20:22:43.991270 vboxnet0 M   IP 192.168.0.3.5353 > 224.0.0.251.5353: 0 PTR (QM)? 1.0.168.192.in-addr.arpa. (43)
```

I assume that is why disabling mDNS for `systemd-resolved` fixes the issue.
I ended up creating an entry in `/etc/hosts` as a workaround. 

> This might be because the SSH server (dropbear?) makes reverse DNS lookups

That is likely the case, yes. I created #4076 to make HAOS rebuild Dropbear without the feature enabled.

agners, 2025-05-26:
So there are actually two separate things here, both seemingly related to reverse lookups:

1. Dropbear SSH server being slow on login
2. iptables command beeing slow

For 1, that seems to be a Dropbear default behavior. At least in my setup where the DHCP and DNS server are the same system and make sure there is a reverse lookup answered, things seem to be fast. 

For 2,  that is reproducible also here, `iptables -L` is slow. 

> It seems that we might be able to ~fix this by disabling llmnr and mdns on that interface [like this](https://unix.stackexchange.com/a/652582/237088). 

Hm, yeah so the question really becomes if we want/need mDNS/LLMNR on these interfaces. Technically, it really might be useful to discover services in add-ons. I am not sure though if things really work as intended 🤔 

> That being said, there may be a better way to solve this that may be achieved by understanding why these are both so slow on this interface, when they normally are not.

Is it really slower on this interface? I think the difference is that on other interfaces there is typically a name server which response in due time, which probably makes system-resoolved not wait for 7s until any delayed mDNS responses comes in? 🤔

Ten0, 2025-08-25:
Still there

Ten0, 2025-11-24:
Still there

Ten0, 2026-02-24:
Still there