home-assistant/operating-system #3769: Wifi no longer working on RPi 3B+ following upgrade to 14.1

arh-au, 2024-12-30:
### Describe the issue you are experiencing

Issue introduced in [Home Assistant OS 14.1 on  Rasperry Pi 3B+, Wifi no longer works and I need to utilise cabled LAN.

During startup, multiple fail messages are issued by brcmf_set_channel to following channel numbers: 0x100e, 0xd022, 0xd026, 0xd02a, 0xd02e, 0xd078, 0xd07c, 0xd080, 0xd090, 0x1000e, 0xd022, 0xd026, 0xd02a, 0xd02e, 0xd078, 0xd07c, 0xd080, 0xd090

brcmfac set channespecl fail (reason: -52)

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

6.6.62-haos-raspi

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Raspberry Pi 3B+ 
Home network is mesh Wifi utilising Fritzbox 7490 + Fritzbox 3000 Wifi points. All other equipment remains operational, and all Fritzbox firmware is up-to-date (nothing changed recently).
Timezone Australia/Brisbane (GMT+10)

### Steps to reproduce the issue

1. Upgrade Home Assistant Core to 14.1
2. Wifi no longer works (responds briefly during raspberry Pi boot to ping and then drops out)
3. Note that HA shows Wifi point has an IP address, however it is not reachable except when rasperry PI is also given a wired LAN connection

Note for below logs - between 15:57 and 15:59 - only used Wifi, at 16:00 plugged in ethernet so I could extract logs below.

### Anything in the Supervisor logs that might be useful for us?

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/udev.sh
[05:58:02] INFO: Using udev information from host
cont-init: info: /etc/cont-init.d/udev.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun supervisor (no readiness notification)
services-up: info: copying legacy longrun watchdog (no readiness notification)
s6-rc: info: service legacy-services successfully started
[05:58:02] INFO: Starting local supervisor watchdog...
2024-12-30 05:58:11.455 INFO (MainThread) [__main__] Initializing Supervisor setup
2024-12-30 05:58:11.752 INFO (MainThread) [supervisor.utils.sentry] Initializing Supervisor Sentry
2024-12-30 15:58:11.785 INFO (MainThread) [supervisor.bootstrap] Setting up coresys for machine: raspberrypi3-64
2024-12-30 15:58:11.810 INFO (MainThread) [supervisor.docker.supervisor] Attaching to Supervisor ghcr.io/home-assistant/aarch64-hassio-supervisor with version 2024.12.0
2024-12-30 15:58:11.878 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state initialize
2024-12-30 15:58:11.887 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-12-30 15:58:11.890 INFO (MainThread) [__main__] Setting up Supervisor
2024-12-30 15:58:22.303 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
2024-12-30 15:58:22.554 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
2024-12-30 15:58:22.564 INFO (MainThread) [supervisor.dbus.manager] Connected to system D-Bus.
2024-12-30 15:58:22.566 INFO (MainThread) [supervisor.dbus.agent] Load dbus interface io.hass.os
2024-12-30 15:58:22.570 INFO (MainThread) [supervisor.dbus.hostname] Load dbus interface org.freedesktop.hostname1
2024-12-30 15:58:22.572 INFO (MainThread) [supervisor.dbus.logind] Load dbus interface org.freedesktop.login1
2024-12-30 15:58:22.576 INFO (MainThread) [supervisor.dbus.network] Load dbus interface org.freedesktop.NetworkManager
2024-12-30 15:58:22.583 INFO (MainThread) [supervisor.dbus.rauc] Load dbus interface de.pengutronix.rauc
2024-12-30 15:58:22.587 INFO (MainThread) [supervisor.dbus.resolved] Load dbus interface org.freedesktop.resolve1
2024-12-30 15:58:22.589 INFO (MainThread) [supervisor.dbus.systemd] Load dbus interface org.freedesktop.systemd1
2024-12-30 15:58:22.590 INFO (MainThread) [supervisor.dbus.timedate] Load dbus interface org.freedesktop.timedate1
2024-12-30 15:58:23.731 INFO (MainThread) [supervisor.host.services] Updating service information
[... 242 lines cut ...]
Disk total | 28.5 GB
Disk used | 5.8 GB
Healthy | true
Supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization |  
Board | rpi3-64
Supervisor API | ok
Version API | ok
Installed add-ons | Advanced SSH & Web Terminal (19.0.0)

Dashboards
Dashboards | 3
Resources | 0
Views | 1
Mode | storage

Recorder
Oldest run start time | 17 December 2024 at 23:07
Current run start time | 30 December 2024 at 15:21
Estimated database size (MiB) | 273.19 MiB
Database engine | sqlite
Database version | 3.45.3

maksyms, 2025-01-23:
I'm experiencing the same issue. However, I observed it on 14.0 as well.

arh-au, 2025-01-23:
I have found connecting to wired LAN and rebooting allows ping to wired and wireless IP address, then after disconnecting from wired LAN whilst running then the wireless works again.

maksyms, 2025-01-23:
@arh-au - understood. It is not dissimilar to running `nmcli conn up "Supervisor wlan0"`, as connecting/disconnecting these causes NetworkManager to restart.

Fundamentally, there is an issue with either kernel or some drivers in HAOS 14.0 onwards, by the looks of it, which causes the hardware to malfunction. For now, I just downgraded to 13.2 - there is no downside in it, given the way HA works.

arh-au, 2025-04-24:
Have recently updated to HAOS 15.2 using wired Ethernet and will investigate if the issue with Wifi still exists.

arh-au, 2025-04-25:
OK, have disconnected wired ethernet, confirmed working with Wifi for last day, also performed reboot of OS and connected via Wifi.

closed 2025-04-25 by arh-au