home-assistant/operating-system #3770: Bluetooth not detected on Atheros AR5B22 card (VID: 04ca, PID: 3007) in Home Assistant OS after recent updates

sharklatan, 2025-01-01:
### Describe the issue you are experiencing

I am using a Gigabyte NUC (model GB-BXi7H-4500) with a mini PCIe Atheros AR5B22 card that includes Wi-Fi and Bluetooth 4.0. While the Wi-Fi functionality works perfectly, Bluetooth is not functioning.

When I check the Bluetooth integration in the Home Assistant OS (HAOS) web interface, it shows the message:
	•	No unconfigured Bluetooth adapters found. There are 0 ignored adapters.

[image]

When I run the lsusb command, the VID and PID are detected, showing the following output:
	•	VID: 04ca
	•	PID: 3007

[image]

However, when interacting with Bluetooth using the bluetoothctl command, I receive the error:
	•	Error: Unable to open mgmt_socket.

I have confirmed that the Bluetooth device is working, as it functioned correctly when I had Windows installed on the system just a few days ago. It is also detected and functional when booting a Live USB with Ubuntu.

The exact model of the Wi-Fi + Bluetooth card is Atheros AR5B22.

[image]

At some point when trying to activate Bluetooth, I did see “Lite-On” (full brand name: Lite-On Technology Corp.) as the Bluetooth device name, which indicates that the Bluetooth adapter was detected. Therefore, I suspect that Bluetooth support might have broken after a recent update

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.66-haos

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Hardware details:
	•	Device model: Gigabyte NUC
	•	Model: GB-BXi7H-4500
	•	Processor: Intel Core i7-4500U
	•	RAM: 12GB
	•	Storage: SSD 240GB
	•	Graphics: Integrated Intel HD Graphics 4400
	•	Wi-Fi + Bluetooth card: Atheros AR5B22 (Mini PCIe)
	•	Bluetooth version: Bluetooth 4.0
	•	Wi-Fi: Working correctly (Wi-Fi functionality works fine)
	•	Bluetooth: Not functioning correctly in Home Assistant OS
	•	USB devices:
	•	No additional USB devices attached.
	•	Operating System: Home Assistant OS
	•	Version: (please provide the specific version you’re using)

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2025-01-01 01:26:48.447 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
2025-01-01 01:26:48.447 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
2025-01-01 01:26:48.448 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
2025-01-01 01:26:48.448 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
2025-01-01 01:26:48.715 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
2025-01-01 01:26:48.715 INFO (MainThread) [supervisor.resolution.checks.base] Run check for no_current_backup/system
2025-01-01 01:26:48.716 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
2025-01-01 01:26:48.716 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
2025-01-01 01:26:48.717 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
2025-01-01 01:26:48.727 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
2025-01-01 01:26:48.728 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
2025-01-01 01:26:48.728 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
2025-01-01 01:26:48.729 INFO (MainThread) [supervisor.resolution.check] System checks complete
2025-01-01 01:26:48.729 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
2025-01-01 01:26:48.937 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2025-01-01 01:26:48.938 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2025-01-01 01:26:48.938 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
```

### Anything in the Host logs that might be useful for us?

```txt
2025-01-01 08:42:46.532 homeassistant dockerd[498]: time="2025-01-01T08:42:46.531836587Z" level=error msg="[resolver] failed to query external DNS server" client-addr="udp:x.30.232.2:43155" dns-server="udp:x.140.242.21:53" error="read udp x.30.232.2:43155->x.140.242.21:53: i/o timeout" question=";o427061.ingest.sentry.io.\tIN\t A"
```

### System information

## System Information

version | core-2024.12.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.0
os_name | Linux
os_version | 6.6.66-haos
arch | x86_64
timezone | America/Mexico_City
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.1
Stage | running
Available Repositories | 1492
Downloaded Repositories | 43

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.0
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 109.3 GB
disk_used | 23.2 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | AirCast (4.2.3), File editor (5.8.0), RPC Shutdown (2.5), Network UPS Tools (0.14.0), Nginx Proxy Manager (1.0.1), Let's Encrypt (5.2.10), Studio Code Server (5.18.0), VLC (0.3.0), Log Viewer (0.17.1), MariaDB (2.7.1), Castsponsorskip (0.8.1), Epic Games Free (debian-2024-12-09), Free Games Claimer (NoVNC not working) (1.6-6), Mosquitto broker (6.4.1), Home-Assistant-Matter-Hub (3.0.0-alpha.58), Matter Server (7.0.0), Terminal & SSH (9.16.0), Cloudflared (5.2.5), ESPHome Device Builder (2024.12.2), Samba share (12.3.2), Webtop (4.16-r0-ls95)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 28
views | 4
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 20 de diciembre de 2024, 18:58
-- | --
current_recorder_run | 31 de diciembre de 2024, 19:27
estimated_db_size | 187.49 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

tony-gutierrez, 2025-01-07:
Same adapter, and same situation...there was a liteon device showing a problem.
bluetoothctl shows no controllers.

chrishome7, 2025-02-18:
This is also Being reported https://github.com/home-assistant/operating-system/issues/3802

and over at core https://github.com/home-assistant/core/issues/135526