home-assistant/operating-system #3783: HA Blue: excessive CPU usage by systemd-udevd, over-current warning in host log

JensJott, 2025-01-09:
### Describe the issue you are experiencing

[image]

After reboot, everything seems fine. CPU usages is low (<10%). After some hours, CPU usage starts to climb on one core until reaching 100% - and stays there.
BT is working without problems but there are constantly warnings in the host log.

Has my USB stick gone bad? What commands can I use to track this down?

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

14.1

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

USB: bluetooth-stick ("Raspberry Pi Foundation CSR8510 A10")

### Steps to reproduce the issue

?

### Anything in the Supervisor logs that might be useful for us?

```txt
2025-01-09 19:08:21.778 ERROR (SyncWorker_1) [supervisor.docker.manager] Can't read stats from addon_core_mariadb: Expecting value: line 1 column 1 (char 0)
2025-01-09 19:08:24.765 WARNING (MainThread) [supervisor.services.modules.mysql] Can't remove not exists services

probably unrelated
```

### Anything in the Host logs that might be useful for us?

```txt
2025-01-09 19:51:45.661 homeassistant kernel: usb 1-1-port2: over-current condition
2025-01-09 19:51:45.753 homeassistant kernel: usb 2-1-port2: over-current condition
2025-01-09 19:51:45.917 homeassistant kernel: usb 1-1-port2: over-current condition
2025-01-09 19:51:45.973 homeassistant kernel: usb 2-1-port2: over-current condition
```

### System information

## System Information

version | core-2025.1.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.6.66-haos
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.2
Stage | running
Available Repositories | 1498
Downloaded Repositories | 11

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.3
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 113.9 GB
disk_used | 40.6 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), Samba share (12.3.2), Terminal & SSH (9.16.0), TasmoAdmin (0.31.2), Glances (0.21.1), Mosquitto broker (6.4.1), MariaDB (2.7.2), phpMyAdmin (0.11.0), SQLite Web (4.3.0), Nginx Proxy Manager (1.0.1), Samba Backup (5.2.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 8
views | 17
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 9. Januar 2024 um 18:10
-- | --
current_recorder_run | 8. Januar 2025 um 18:55
estimated_db_size | 20146.75 MiB
database_engine | mysql
database_version | 10.11.6

</details>

sairon, 2025-01-16:
The overcurrent warnings point to a faulty peripheral - if the USB Bluetooth adapter is the only one connected, then it might indeed _have gone bad_. I imagine a misbehaving USB device might trigger the udev daemon, which is responsible for managing device events.

JensJott, 2025-01-22:
Thanks for your reply.
I will ordner a new one. Do you think the CPU usage is related?

JensJott, 2025-02-12:
The new stick did not fix the problem. 
I ordered this one: https://www.amazon.de/dp/B0CRN8WNQK
HA Blue detects it as "Realtek Bluetooth 5.4 Radio".

2025-02-12 18:43:56.465 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 18:43:56.661 homeassistant kernel: usb 1-1-port2: over-current condition
2025-02-12 18:43:56.685 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 18:43:57.549 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 18:43:57.685 homeassistant kernel: usb 1-1-port2: over-current condition
2025-02-12 18:43:57.765 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 18:43:57.941 homeassistant kernel: usb 1-1-port2: over-current condition
2025-02-12 18:43:57.981 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 18:43:59.061 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 18:43:59.221 homeassistant kernel: usb 1-1-port2: over-current condition
2025-02-12 18:43:59.281 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 18:44:00.151 homeassistant bluetoothd[428]: src/device.c:device_new() Unable to register device interface for 00:00:00:00:00:00
2025-02-12 18:44:00.152 homeassistant bluetoothd[428]: Unable to create object for found device 00:00:00:00:00:00
2025-02-12 18:44:00.361 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 18:44:00.501 homeassistant kernel: usb 1-1-port2: over-current condition
2025-02-12 18:44:00.577 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 18:44:00.757 homeassistant kernel: usb 1-1-port2: over-current condition
2025-02-12 18:44:00.793 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 18:44:01.877 homeassistant kernel: usb 2-1-port2: over-current condition
...
2025-02-12 19:32:33.921 homeassistant kernel: usb 1-1-port2: over-current condition
2025-02-12 19:32:33.997 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 19:32:34.173 homeassistant kernel: usb 1-1-port2: over-current condition
2025-02-12 19:32:34.213 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 19:32:35.201 homeassistant kernel: usb 1-1-port2: over-current condition
2025-02-12 19:32:35.293 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 19:32:35.457 homeassistant kernel: usb 1-1-port2: over-current condition
2025-02-12 19:32:35.509 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 19:32:36.547 homeassistant bluetoothd[428]: src/device.c:device_new() Unable to register device interface for 00:00:00:00:00:00
2025-02-12 19:32:36.548 homeassistant bluetoothd[428]: Unable to create object for found device 00:00:00:00:00:00
2025-02-12 19:32:36.589 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 19:32:36.737 homeassistant kernel: usb 1-1-port2: over-current condition
2025-02-12 19:32:36.809 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 19:32:36.993 homeassistant kernel: usb 1-1-port2: over-current condition
2025-02-12 19:32:37.025 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 19:32:38.058 homeassistant bluetoothd[428]: src/device.c:device_new() Unable to register device interface for 00:00:00:00:00:00
2025-02-12 19:32:38.059 homeassistant bluetoothd[428]: Unable to create object for found device 00:00:00:00:00:00
2025-02-12 19:32:38.113 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 19:32:38.273 homeassistant kernel: usb 1-1-port2: over-current condition
2025-02-12 19:32:38.329 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 19:32:38.371 homeassistant bluetoothd[428]: src/device.c:device_new() Unable to register device interface for 00:00:00:00:00:00
2025-02-12 19:32:38.371 homeassistant bluetoothd[428]: Unable to create object for found device 00:00:00:00:00:00
2025-02-12 19:32:38.819 homeassistant bluetoothd[428]: src/device.c:device_new() Unable to register device interface for 00:00:00:00:00:00
2025-02-12 19:32:38.819 homeassistant bluetoothd[428]: Unable to create object for found device 00:00:00:00:00:00
2025-02-12 19:32:39.316 homeassistant bluetoothd[428]: src/device.c:device_new() Unable to register device interface for 00:00:00:00:00:00
2025-02-12 19:32:39.316 homeassistant bluetoothd[428]: Unable to create object for found device 00:00:00:00:00:00
2025-02-12 19:32:39.413 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 19:32:39.553 homeassistant kernel: usb 1-1-port2: over-current condition
2025-02-12 19:32:39.629 homeassistant kernel: usb 2-1-port2: over-current condition
2025-02-12 19:32:39.809 homeassistant kernel: usb 1-1-port2: over-current condition
2025-02-12 19:32:39.849 homeassistant kernel: usb 2-1-port2: over-current condition

Uptime is now ~1hr and systemd-udevd is already the most cpu intensive task again.

JensJott, 2025-04-01:
Hello, please help me, I'm under the water?!

closed 2025-07-09 by github-actions[bot]