home-assistant/operating-system #3794: Windows Surface Notebook - built-in keyboard not working in HAOS

VidarParry, 2025-01-15:
### Describe the issue you are experiencing

Just installed HAOS 14.1 for the first time on a Windows Surface Notebook. All working fine EXCEPT the in-built notebook keyboard doesn’t work - nothing appears on the command line when typing.

Notebook in-built keyboard works in BIOS and previously worked when it was a Windows machine.

Plugging in a USB keyboard works OK.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

14.1 with core v2025.1.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Windows Surface Notebook

### Steps to reproduce the issue

1. Boot Internal Storage after install of HAOS
2. Arrive at screen showing CLI, pressing any key or combination of key does not display anything or cause any cursor move or change.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing in logs
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing I could see that is relevant.
```

### System information

## System Information

version | core-2025.1.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.6.66-haos
arch | x86_64
timezone | Australia/Brisbane
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | pending
GitHub Web | pending
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.2
Stage | running
Available Repositories | 1558
Downloaded Repositories | 2

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | pending
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.3
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 116.7 GB
disk_used | 4.5 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Get HACS (1.3.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 0
views | 0
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | January 14, 2025 at 2:24 AM
-- | --
current_recorder_run | January 14, 2025 at 2:35 PM
estimated_db_size | 2.60 MiB
database_engine | sqlite
database_version | 3.47.1

</details>

sairon, 2025-01-17:
Unfortunately there is no native support for the Surface hardware in Linux. There are some [patchsets](https://github.com/linux-surface/linux-surface) and utilities floating around that aim to add the support for Linux, but given how esoteric it is, it's not something that has a chance to be included in HAOS.

closed 2025-01-17 by sairon

VidarParry, 2025-01-17:
Thank you Jan.
All the surface notebook facilities (including keyboard, mouse, touch screen) work in Ubuntu. Can those drivers be used in HAOS?
Cheers... Steve

sairon, 2025-01-17:
As I indicated, we're bit more conservative with applying patches downstream to add hardware support. However, I see that there has been some development in the Linux and the 6.12 kernel has support for more Surface hardware. Eventually the upcoming HAOS 15 may fix it, but if it's not supported in the vanilla kernel, it's unlikely we would be porting any drivers to HAOS.