home-assistant/operating-system #3797: Restoring backup of system with SSL enabled can result in hang/failure

Flupsy, 2025-01-16:
### Describe the issue you are experiencing

I installed a fresh copy of HAOS, and then restored the backup of my self-installed system.  The restore never completed, appearing to hang forever, and the logs revealed that the SSL certificate and private key files couldn't be found so the http service couldn't start.  The user is left with a nonfunctional system and no obvious way to proceed as there's now no http service.

Manually editing the config file and taking out the SSL directives allowed the system to start properly.

I'm not sure if it affects things but the cert and key on the source system were in a nonstandard place.

The backup was made on 2025.1.2, and restored on 2025.1.2.

In my view the process should either

- include the cert and key in the backup, and restore it somewhere appropriate, modifying the config file;
- test for presence of the cert and key, and refuse to run if they're not present; or
- remove the SSL config, warning the user, thus at least letting the restore complete cleanly.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

2025.1.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

TrueNAS SCALE VM using the qcow2 image converted to raw and written to a virtual disk.

### Steps to reproduce the issue

1. Make a backup of a running system using SSL (possibly where key and cert are in nonstandard places)
2. Install the VM image of HAOS and start it
3. Restore the backup on the new HAOS system
4. The restore process hangs forever

### Anything in the Supervisor logs that might be useful for us?

```txt
nothing relevant
```

### Anything in the Host logs that might be useful for us?

```txt
2025-01-15 13:49:01.497 ERROR (MainThread) [homeassistant.config] Invalid config for 'http' at configuration.yaml, line 13: not a file for dictionary value 'http->ssl_certificate', got '/etc/ssl/[domain redacted]/fullchain.pem', please check the docs at https://www.home-assistant.io/integrations/http
Invalid config for 'http' at configuration.yaml, line 14: not a file for dictionary value 'http->ssl_key', got '/etc/ssl/[domain redacted]/privkey.pem', please check the docs at https://www.home-assistant.io/integrations/http
2025-01-15 13:49:01.497 ERROR (MainThread) [homeassistant.setup] Setup failed for 'http': Invalid config.
2025-01-15 13:49:01.498 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of 'auth'. Setup failed for dependencies: ['http']
2025-01-15 13:49:01.498 ERROR (MainThread) [homeassistant.setup] Setup failed for 'auth': Could not setup dependencies: http
2025-01-15 13:49:01.498 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of 'config'. Setup failed for dependencies: ['http']
2025-01-15 13:49:01.498 ERROR (MainThread) [homeassistant.setup] Setup failed for 'config': Could not setup dependencies: http
2025-01-15 13:49:01.498 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of 'diagnostics'. Setup failed for dependencies: ['http']
2025-01-15 13:49:01.498 ERROR (MainThread) [homeassistant.setup] Setup failed for 'diagnostics': Could not setup dependencies: http
2025-01-15 13:49:01.498 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of 'file_upload'. Setup failed for dependencies: ['http']
2025-01-15 13:49:01.498 ERROR (MainThread) [homeassistant.setup] Setup failed for 'file_upload': Could not setup dependencies: http
2025-01-15 13:49:01.498 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of 'image_upload'. Setup failed for dependencies: ['http']
2025-01-15 13:49:01.498 ERROR (MainThread) [homeassistant.setup] Setup failed for 'image_upload': Could not setup dependencies: http
2025-01-15 13:49:01.498 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of 'repairs'. Setup failed for dependencies: ['http']
2025-01-15 13:49:01.499 ERROR (MainThread) [homeassistant.setup] Setup failed for 'repairs': Could not setup dependencies: http
2025-01-15 13:49:01.499 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of 'websocket_api'. Setup failed for dependencies: ['http']
2025-01-15 13:49:01.499 ERROR (MainThread) [homeassistant.setup] Setup failed for 'websocket_api': Could not setup dependencies: http
2025-01-15 13:49:01.499 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of 'api'. Setup failed for dependencies: ['http']
2025-01-15 13:49:01.499 ERROR (MainThread) [homeassistant.setup] Setup failed for 'api': Could not setup dependencies: http
...etc...
```

### System information

This is the new system's config once I'd fixed the problem by editing the config file.

## System Information

version | core-2025.1.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.6.66-haos
arch | x86_64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.3
Stage | running
Available Repositories | 1500
Downloaded Repositories | 2

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.3
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 31.8 GB
disk_used | 6.5 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Whisper (2.4.0), Piper (1.5.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 0
views | 2
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 29 December 2024 at 15:03
-- | --
current_recorder_run | 15 January 2025 at 14:28
estimated_db_size | 150.22 MiB
database_engine | sqlite
database_version | 3.47.1

</details>

Flupsy, 2025-01-20:
> Hello,
> 
> I just face this same issue when restoring a backup on the same OS version with SSL set up. Can you point me on how to edit the file so the system can boot? Thank you

I got a console on the VM, and edited `/mnt/data/supervisor/homeassistant/configuration.yaml`, removing the `ssl_certificate` and `ssl_key` directives.

sairon, 2025-01-20:
There was/is a bug that affects backing up SSL data, see https://github.com/home-assistant/core/issues/135634 and https://github.com/home-assistant/core/pull/136080. But that is related to the "standard" SSL folder used by Home Assistant, there is no way it could back up random non-standard folders. Any other changes in behavior of backups should be discussed as feature requests or bug reports of the Home Assistant Core, the Operating System has no role in the backups.

closed 2025-01-20 by github-actions[bot]