home-assistant/operating-system #3798: Trying to flash image

jmon2, 2025-01-16:
### Describe the issue you are experiencing

When installing latest version on Rasberry pi 4, gets hung up on latest version update. 2025.1.2 not found.

[observer_2025-01-15T18-02-39.212Z.log.txt](https://github.com/user-attachments/files/18438283/observer_2025-01-15T18-02-39.212Z.log.txt)

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

2025.1.2

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Mini sd card flashed with raspberry pi imager.

### Steps to reproduce the issue

1. Install of image from sd card.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Can't  get to system.
```

### Anything in the Host logs that might be useful for us?

```txt
Can't get to system. Not installing completely.
```

mclardass, 2025-01-16:
Seems to be an on-going problem that is only affecting _some_ users but not _enough_ users to get the devs attention. 

See also [OS issue 3633](https://github.com/home-assistant/operating-system/issues/3633) and [core issue 135342](https://github.com/home-assistant/core/issues/135342), I'm stuck at 2024.12.4 and am tired of making changes on my side to correct something that is likely on the github side.

jmon2, 2025-01-16:
I agree.

Sent via the Samsung Galaxy Z Fold3 5G, an AT&T 5G smartphone

mclardass, 2025-01-17:
Found [this post](https://github.com/home-assistant/core/issues/107811#issuecomment-1922392578) which hasn't helped me but perhaps you'll have some success. I did a find on yaml and sorted by size, backed them up or commented out entries in my config, and still getting the same failure. Good luck

sairon, 2025-01-17:
It's not that we're ignoring the issue, the problem just can have multiple causes, in some cases (and most often) it's intermittent issue caused by network issues we can't do anything about. Search the Supervisor repository issues for the error, there is plenty of them and there has been always some solution found. But in this case it's something else - there is no space on the SD card to finish the installation:
```
2025-01-15 12:09:03.756 WARNING (MainThread) [supervisor.jobs] 'GitRepo.clone' blocked from execution, not enough free space (0.9GB) left on the device
```

closed 2025-01-17 by github-actions[bot]