home-assistant/operating-system #3802: Bluetooth Adapters not being detected / Failed to set up.

chrishome7, 2025-01-17:
### Describe the issue you are experiencing

I believe This issue is being reported by others over in Core. They claim its an OS issue and to report it here.  (https://github.com/home-assistant/core/issues/135526)

[image]

Full hardware reboots clear up warnings, but then they randomly return or randomly return on restarts. Even with without the warning, theres no detection of Bluetooth devices. This wasn't a problem to until recent updates I'm uncertain of witch update and assumed it was core. One other person reported the same issue and warnings that began only after recent updates. The only Logs I can see anything relevant is within the Core Logs.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.66-haos

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

x86-64 (Dell Wyse)

Bluetooth adapter
Intel 8087:07dc.

### Steps to reproduce the issue

1. Randomly happens or happens on restarts.

### Anything in the Supervisor logs that might be useful for us?

```txt
nothing
```

### Anything in the Host logs that might be useful for us?

```txt
2025-01-15 10:33:35.604 homeassistant kernel: Bluetooth: RFCOMM TTY layer initialized
2025-01-15 10:33:35.604 homeassistant kernel: Bluetooth: RFCOMM socket layer initialized
2025-01-15 10:33:35.604 homeassistant kernel: Bluetooth: RFCOMM ver 1.11
2025-01-15 10:33:35.609 homeassistant bluetoothd[397]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSink/sbc
2025-01-15 10:33:35.612 homeassistant bluetoothd[397]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/sbc
2025-01-15 10:33:35.612 homeassistant bluetoothd[397]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSink/sbc_xq_453
2025-01-15 10:33:35.613 homeassistant bluetoothd[397]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/sbc_xq_453
2025-01-15 10:33:35.613 homeassistant bluetoothd[397]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSink/sbc_xq_512
2025-01-15 10:33:35.619 homeassistant bluetoothd[397]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/sbc_xq_512
2025-01-15 10:33:35.620 homeassistant bluetoothd[397]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSink/sbc_xq_552
2025-01-15 10:33:35.620 homeassistant bluetoothd[397]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/sbc_xq_552
2025-01-15 10:33:37.371 homeassistant os-agent[166]: INFO: 2025/01/15 10:33:37 main.go:95: Diagnostics is now true

025-01-15 10:48:03.115 homeassistant systemd[1]: systemd-tmpfiles-clean.service: Deactivated successfully.
2025-01-15 10:48:03.118 homeassistant systemd[1]: Finished Cleanup of Temporary Directories.
2025-01-15 10:49:42.325 homeassistant bluetoothd[397]: Path /org/bleak/68/140225922052208 removed along with Adv Monitor app :1.44
2025-01-15 10:49:42.333 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -112
2025-01-15 10:49:42.333 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -112
2025-01-15 10:49:42.334 homeassistant kernel: Bluetooth: hci0: stop background scanning failed: -112
2025-01-15 10:49:44.970 homeassistant bluetoothd[397]: Path /org/bleak/68/140225975971504 reserved for Adv Monitor app :1.44
2025-01-15 10:51:42.331 homeassistant bluetoothd[397]: Path /org/bleak/68/140225975971504 removed along with Adv Monitor app :1.44
2025-01-15 10:51:44.955 homeassistant bluetoothd[397]: Path /org/bleak/68/140225922051760 reserved for Adv Monitor app :1.44
2025-01-15 10:53:42.335 homeassistant bluetoothd[397]: Path /org/bleak/68/140225922051760 removed along with Adv Monitor app :1.44
2025-01-15 10:53:42.342 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -112
2025-01-15 10:53:42.342 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -112
2025-01-15 10:53:42.343 homeassistant kernel: Bluetooth: hci0: stop background scanning failed: -112
2025-01-15 10:53:42.355 homeassistant kernel: Bluetooth: hci0: HCI reset during shutdown failed
2025-01-15 10:53:42.875 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x0c03
2025-01-15 10:53:42.879 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x1001
2025-01-15 10:53:44.956 homeassistant bluetoothd[397]: Path /org/bleak/68/140225922315248 reserved for Adv Monitor app :1.44
[... 580 lines cut ...]
  File "/usr/local/lib/python3.13/site-packages/bleak/backends/bluezdbus/scanner.py", line 178, in start
    self._stop = await manager.passive_scan(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<4 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/bleak/backends/bluezdbus/manager.py", line 502, in passive_scan
    self._check_adapter(adapter_path)
    ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/bleak/backends/bluezdbus/manager.py", line 195, in _check_adapter
    raise BleakError(f"adapter '{adapter_path.split('/')[-1]}' not found")
bleak.exc.BleakError: adapter 'hci0' not found

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "src/habluetooth/scanner.py", line 528, in habluetooth.scanner.HaScanner._async_restart_scanner
  File "src/habluetooth/scanner.py", line 286, in _async_start
  File "src/habluetooth/scanner.py", line 373, in _async_start_attempt
habluetooth.scanner.ScannerStartError: hci0 (7C:5C:F8:D1:2C:65): Failed to start Bluetooth: adapter 'hci0' not found; Try power cycling the Bluetooth hardware.
2025-01-15 08:26:41.859 WARNING (MainThread) [bluetooth_auto_recovery.recover] There are no BT controllers present in the system!
2025-01-15 08:26:41.862 ERROR (MainThread) [bluetooth_auto_recovery.recover] hciNone [7C:5C:F8:D1:2C:65] seems not to exist (anymore), check BT interface mac address in your settings; Available adapters: {} 
2025-01-15 08:26:41.869 WARNING (MainThread) [bluetooth_auto_recovery.recover] There are no BT controllers present in the system!
2025-01-15 08:26:41.872 ERROR (MainThread) [bluetooth_auto_recovery.recover] hciNone [7C:5C:F8:D1:2C:65] seems not to exist (anymore), check BT interface mac address in your settings; Available adapters: {} 
2025-01-15 08:26:41.877 ERROR (MainThread) [habluetooth.scanner] hci0 (7C:5C:F8:D1:2C:65): Failed to restart Bluetooth scanner

meetyourlaser, 2025-01-28:
After upgrading from 14.1 to 14.2, I, too, am missing my Bluetooth adapter. Nothing in core/supervisor/host logs. 

Running `bluetoothctl` the first message I get is _Waiting to connect to bluetoothd...Unable to open mgmt_socket_

Downgrading to 14.1/0 didn’t resolve the issue. Generic x86-64 installation on a Dell OptiPlex 3060.

chrishome7, 2025-02-06:
Hopefully someone takes a look at this since BLE is a very common utility!

meetyourlaser, 2025-02-06:
Hopefully! I thought there would be more issues reported.

I miss playing music over Bluetooth.

chrishome7, 2025-02-18:
This continues to be a growing problem that people are reporting over at core.   Can some please take a look at it?

[image]

migube, 2025-03-05:
not sure if related but on Home Assistant Blue / Hardkernel ODROID-N2/N2+ with Realtek BT chip 
I get 
2969231.610783] Bluetooth: hci0: command tx timeout
[2972565.668366] Bluetooth: hci0: HCI reset during shutdown failed
[2972567.710931] Bluetooth: hci0: Opcode 0x0c03 failed: -110
[2972570.679606] Bluetooth: hci0: urb 000000005674952e failed to resubmit (2)
[2972570.793385] usb 1-1.2: reset full-speed USB device number 18 using xhci-hcd
[2972570.924580] Bluetooth: hci1: RTL: examining hci_ver=0a hci_rev=dfc6 lmp_ver=0a lmp_subver=d922
[2972571.133518] Bluetooth: hci1: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761

so it jumps from hci0 to hic1 :(
(need to disconnect /reconnect to get it back...)
cr

aotearoait, 2025-03-25:
Add to this, now on my prox vm for home assisant i get 
Bluetooth: hci0: unexpected event for opcode 0x1001
Then crashes HA

SirSixx, 2025-04-25:
I think I'm getting the same on my Pi 5.

Bluetooth management socket connection lost: [Errno 22] Invalid argument
adapter 'hci0' not found
[Errno 16] Resource busy
...
<command_opcode=ReadControllerInformation, status=InvalidIndex>