home-assistant/operating-system #3807: dockerd is trying to kill homeassistant container

amaciuc, 2025-01-18:
### Describe the issue you are experiencing

HI team,

I am using HA as OS installed on an Intel NUC (4 Intel(R) Core(TM) i5-5250U CPU @ 1.60GHz, x86_64, 8 GB RAM). HA version is `2024.12.1`, 
- `ha os info` output:
```
# ha os info
board: generic-x86-64
boot: B
boot_slots:
  A:
    state: inactive
    status: good
    version: "13.2"
  B:
    state: booted
    status: good
    version: "14.0"
data_disk: SAMSUNG-MZ7TE128HMGR-000H1-S1BDNSAG873285
update_available: true
version: "14.0"
version_latest: "14.1"
```
- `ha host info` outut:

```
# ha host info
agent_version: 1.6.0
apparmor_version: 3.1.2
boot_timestamp: 1737190446492693
broadcast_llmnr: true
broadcast_mdns: true
chassis: embedded
cpe: cpe:2.3:o:home-assistant:haos:14.0:*:production:*:*:*:generic-x86-64:*
deployment: production
disk_free: 48
disk_life_time: null
disk_total: 116.7
disk_used: 63.9
dt_synchronized: true
dt_utc: "2025-01-18T15:18:22.705541+00:00"
features:
- reboot
- shutdown
- services
- network
- hostname
- timedate
- os_agent
- haos
- resolved
- journal
- disk
- mount
hostname: homeassistant
kernel: 6.6.63-haos
llmnr_hostname: homeassistant
operating_system: Home Assistant OS 14.0
startup_time: 0.802022
timezone: Etc/UTC
use_ntp: true
virtualization: ""
```

My worrying problem is that I caught when `dockerd` is trying to kill the `homeassistant` container. Here are the logs from host (`ha host logs`):
```
2025-01-18 08:42:12.584 homeassistant dockerd[592]: time="2025-01-18T08:42:12.584646978Z" level=info msg="Container failed to exit within 4m20s of signal 15 - using the force" container=3c6651e079843c7b1d1f67503a7754bd9391872a2eb58636fd61e65dc4494ad4
2025-01-18 08:42:22.615 homeassistant dockerd[592]: time="2025-01-18T08:42:22.615601898Z" level=warning msg="Container failed to exit within 10s of kill - trying direct SIGKILL" container=3c6651e079843c7b1d1f67503a7754bd9391872a2eb58636fd61e65dc4494ad4 error="context deadline exceeded"
2025-01-18 08:42:26.617 homeassistant dockerd[592]: time="2025-01-18T08:42:26.616605687Z" level=error msg="error killing container: context deadline exceeded" container=3c6651e079843c7b1d1f67503a7754bd9391872a2eb58636fd61e65dc4494ad4 error="tried to kill container, but did not receive an exit event"
[... 120 lines cut ...]
host_connectivity	true
supervisor_connectivity	true
ntp_synchronized	true
virtualization	
Board	generic-x86-64
Supervisor API	ok
Version API	ok
Installed add-ons	Studio Code Server (5.14.2), Mosquitto broker (6.4.0), Home Assistant Google Drive Backup (0.112.1), AdGuard Home (5.0.1), ESPHome Device Compiler (2024.11.3), Zigbee2MQTT (1.42.0-2), Cloudflared (5.2.7), Samba share (12.3.2), Portainer (2.19.1), Whisper (1.0.0), Piper (1.4.0), Squeezelite (0.0.17), openWakeWord (1.8.2), VLC (0.2.0), Logitech Media Server (8.3.1.57), Uptime Kuma (0.12.4), Network UPS Tools (0.13.0), Filebrowser (2.23.0_14), ESPresense Companion (1.0.3), Samba NAS (12.3.2-nas), go2rtc (1.9.7), Frigate (Full Access) (0.14.1), HDD Tools (1.1.0), Terminal & SSH (9.16.0), Redis Stack (7.4.0-v1), ClamAV (1.4.1_base), MariaDB (2.7.2), Nextcloud (30.0.4)
Dashboards
Dashboards	2
Resources	26
Views	14
Mode	storage
Recorder
Oldest run start time	September 5, 2023 at 8:36 AM
Current run start time	January 18, 2025 at 10:55 AM
Estimated database size (MiB)	34673.73 MiB
Database engine	sqlite
Database version	3.45.3
Sonoff
Version	3.5.4 (a4a8c5f)
Cloud devices online	3 / 7
Local devices online	2 / 2
Debug page	– [more info](https://suceavahome.com/api/sonoff/5a8d5dbb-93ca-493a-af44-9713381af32e)
```

sairon, 2025-01-22:
Can you post complete logs with the context when that happens, without cherry-picking, and including Supervisor logs? The log lines indicate that a graceful shutdown was requested through the Docker API (not obvious from the logs when) and the Core container is not responding to then, then it's not responding to a kill signal either. There should be something that requested the restart, and it can't be OOM killer, because that would kill a process without calling the API. Also, the question is why the container doesn't react to the shutdown signals, Core logs may be helpful here too.

amaciuc, 2025-01-29:
So, I came back with more logs. This issue appeared once again and now I can show you more logs

```
# date
Wed Jan 29 08:38:33 UTC 2025
# env TZ=Europe/Bucharest date
Wed Jan 29 10:38:02 EET 2025

```

I case you want to trace the timedate in logs

Logs of `homeassistant` container:

```
# docker logs --tail=200 homeassistant
2025-01-29 08:10:32.428 ERROR (MainThread) [homeassistant.components.webostv.notify] TV unreachable
2025-01-29 08:10:32.429 ERROR (MainThread) [homeassistant.components.webostv.notify] TV unreachable
2025-01-29 08:48:08.526 ERROR (MainThread) [homeassistant.components.webostv.notify] TV unreachable
2025-01-29 09:08:43.129 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'frigate' is overwriting the 'frontend_stream_type' property in the FrigateCameraWebRTC class, which is deprecated and will be removed in Home Assistant 2025.6,  at custom_components/frigate/camera.py, line 180: Camera.__init__(self). Please create a bug report at https://github.com/blakeblackshear/frigate-hass-integration/issues
2025-01-29 09:18:25.154 ERROR (MainThread) [homeassistant.components.generic.config_flow] Error getting camera image from Http://127.0.0.1:8554/living_baby_cam_orig: ConnectError
2025-01-29 09:18:39.434 ERROR (MainThread) [homeassistant.components.generic.config_flow] Error getting camera image from Http://127.0.0.1:8554/living_baby_cam_orig: ConnectError
2025-01-29 09:18:48.012 ERROR (MainThread) [homeassistant.components.generic.config_flow] Error getting camera image from http://127.0.0.1:8554/living_baby_cam_orig: ConnectError
2025-01-29 09:18:54.071 ERROR (MainThread) [homeassistant.components.generic.config_flow] Error getting camera image from http://127.0.0.1:8554/living_baby_cam_orig: ConnectError
2025-01-29 09:19:50.010 ERROR (MainThread) [homeassistant.components.generic.config_flow] Error getting camera image from Rtsp://127.0.0.1:8554/living_baby_cam_orig: UnsupportedProtocol
2025-01-29 09:20:03.632 ERROR (MainThread) [homeassistant.components.generic.config_flow] Error getting camera image from rtsp://127.0.0.1:8554/living_baby_cam_orig: UnsupportedProtocol
2025-01-29 09:20:06.449 ERROR (MainThread) [homeassistant.components.generic.config_flow] Error getting camera image from rtsp://127.0.0.1:8554/living_baby_cam_orig: UnsupportedProtocol
2025-01-29 09:21:26.515 ERROR (MainThread) [homeassistant.components.generic.config_flow] Error getting camera image from rtsp://127.0.0.1:8554/living_baby_cam_orig: UnsupportedProtocol
2025-01-29 09:21:53.088 ERROR (MainThread) [homeassistant.components.generic.config_flow] Error getting camera image from rtsp://192.168.114:8554/living_baby_cam_orig: UnsupportedProtocol
2025-01-29 09:21:57.870 ERROR (MainThread) [homeassistant.components.generic.config_flow] Error getting camera image from rtsp://192.168.114:8554/living_baby_cam_orig: UnsupportedProtocol
2025-01-29 09:25:29.839 ERROR (MainThread) [homeassistant.components.generic.config_flow] Error getting camera image from rtsp://127.0.0.1:8554/living_baby_cam: UnsupportedProtocol
2025-01-29 09:25:51.553 ERROR (MainThread) [homeassistant.components.generic.config_flow] Error getting camera image from rtsp://192.168.0.114:8554/living_baby_cam: UnsupportedProtocol
2025-01-29 09:25:55.415 ERROR (MainThread) [homeassistant.components.generic.config_flow] Error getting camera image from rtsp://192.168.0.114:8554/living_baby_cam: UnsupportedProtocol
s6-rc: info: service legacy-services: stopping
s6-svwait: fatal: timed out
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

Don't bother with logs from camera. I tried something

A large snapshot of `ha host logs`

```
# ha host logs -n 200
2025-01-29 07:17:01.344 homeassistant kernel: audit: type=1105 audit(1738135021.343:15129): pid=3136907 uid=0 auid=0 ses=1988 subj=docker-default msg='op=PAM:session_open grantors=pam_loginuid,pam_env,pam_env,pam_permit,pam_umask,pam_unix,pam_limits acct="root" exe="/usr/sbin/cron" hostname=? addr=? terminal=cron res=success'
2025-01-29 07:17:01.348 homeassistant kernel: audit: type=1104 audit(1738135021.347:15130): pid=3136907 uid=0 auid=0 ses=1988 subj=docker-default msg='op=PAM:setcred grantors=pam_permit acct="root" exe="/usr/sbin/cron" hostname=? addr=? terminal=cron res=success'
2025-01-29 07:17:01.348 homeassistant kernel: audit: type=1106 audit(1738135021.347:15131): pid=3136907 uid=0 auid=0 ses=1988 subj=docker-default msg='op=PAM:session_close grantors=pam_loginuid,pam_env,pam_env,pam_permit,pam_umask,pam_unix,pam_limits acct="root" exe="/usr/sbin/cron" hostname=? addr=? terminal=cron res=success'
2025-01-29 07:17:33.833 homeassistant bluetoothd[431]: Path /org/bleak/67/140156312270704 removed along with Adv Monitor app :1.63
2025-01-29 07:17:36.453 homeassistant bluetoothd[431]: Path /org/bleak/67/140156139260960 reserved for Adv Monitor app :1.63
2025-01-29 07:19:33.840 homeassistant bluetoothd[431]: Path /org/bleak/67/140156139260960 removed along with Adv Monitor app :1.63
2025-01-29 07:19:36.442 homeassistant bluetoothd[431]: Path /org/bleak/67/140156090531440 reserved for Adv Monitor app :1.63
[... cut ...]
3c6651e07984   ghcr.io/home-assistant/generic-x86-64-homeassistant:2024.12.1   "/init"                  7 weeks ago     Up 7 days                                                                                                                                              homeassistant
b441fc000c78   ghcr.io/home-assistant/amd64-hassio-observer:2023.06.0          "/usr/bin/observer"      19 months ago   Up 7 days               0.0.0.0:4357->80/tcp, [::]:4357->80/tcp                                                                                        hassio_observer

```

I think the `init` process will help to prevent this kind of situation

closed 2025-05-07 by github-actions[bot]