home-assistant/operating-system #3829: Violation of FHS by storing variable data under /usr/share?

ralfbergs, 2025-01-26:
### Describe the issue you are experiencing

It somehow feels wrong that frequently-modified data is stored under `/usr/share`, more specifically under `/usr/share/hassio/homeassistant`, such as various SQLite databases, logfiles.

To me that seems a clear violation of the [Filesystem Hierarchy Standard](https://en.wikipedia.org/wiki/Filesystem_Hierarchy_Standard) that says `/usr/share` is meant for "Architecture-independent (shared) data."

I believe most of the data should go under `/var/lib/homeassistant/` or `/var/log/homeassistant/`.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

not relevant

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

not relevant

### Steps to reproduce the issue

1. Install HASSIO.
2. Observe the folders used by HASSIO within the file system hierarchy.

### Anything in the Supervisor logs that might be useful for us?

```txt
not relevant
```

### Anything in the Host logs that might be useful for us?

```txt
not relevant
```

### System information

not relevant

sairon, 2025-01-27:
I'm not sure what exactly you mean - neither Home Assistant OS nor the Home Assistant Core seem to use that folder today. I'm closing this here but if you elaborate on the part _"Install HASSIO"_, I can point you to the proper place where to report this.

ralfbergs, 2025-01-27:
@sairon Hi Jan, thanks for your response.

It seems I should have been a bit more verbose, sorry for that.

I used the "[Home Assistant Supervised](https://www.home-assistant.io/installation/linux#install-home-assistant-supervised)" install under Ubuntu Server (`Ubuntu 24.04.1 LTS`). (Yes, I know this Debian-based OS is explicitly not supported, but it would have been exactly the same under Debian, so I believe my report *is* relevant.) This section directs me to use [this install script](https://github.com/home-assistant/supervised-installer). It even mentions explicitly under "[Configuration](https://github.com/home-assistant/supervised-installer#configuration)" that a path of `/usr/share/hassio` is the default for `$DATA_SHARE`.

Am I misunderstanding this big thing, or should this maybe changed to use a different path?

ralfbergs, 2025-01-27:
Many thanks, I will go there.

I was naively assuming that HAOS is simply a ready-made "supervised install"... :-)