home-assistant/operating-system #3832: wifi.powersave option reverts to default after host reboot

vturekhanov, 2025-01-27:
### Describe the issue you are experiencing

Exploring issue #3826 I tried to change powersave mode for my wireless adapter using nmcli tool and direct edit connection file using vi (/etc/NetworkManager/system-connections/Supervisor\ wlp1s0.nmconnection). Setting powersave=2 was successful in both cases but after host reboot I see powersave option reverts to its default value: powersave=1. So, I can't change powersave option.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

14.1

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

G3 Mini PC Alder Lake N100 Intel 12th DDR4 8GB RAM 256GB ROM WiFi BT
https://aliexpress.ru/item/1005006340106437.html

### Steps to reproduce the issue

1. Set powersave option to 2 in the wifi section of network connection in the connection configuration file (/etc/NetworkManager/system-connections/Supervisor\ wlp1s0.nmconnection in my case).
2. Reboot the host.
3. View connection configuration file.

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
n/a
```

### System information

## System Information

version | core-2025.1.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.6.66-haos
arch | x86_64
timezone | Asia/Almaty
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.3
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 234.0 GB
disk_used | 5.9 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Matter Server (7.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 0
views | 0
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | January 25, 2025 at 7:59 AM
-- | --
current_recorder_run | January 27, 2025 at 11:19 AM
estimated_db_size | 7.17 MiB
database_engine | sqlite
database_version | 3.47.1

</details>

vturekhanov, 2025-01-28:
Workaround: set powersave=2 in the connection file and make it immutable (chattr +i).
After rebooting the host got one warning concerning unable writing to the connection file: Operation not permitted.

dgdavid, 2025-01-28:
I'm experiencing the same issue. Thanks a lot for sharing the workaround. I'll check if it works for me too :)

vturekhanov, 2025-01-28:
So, HAOS tries to update connection settings file during host boot. In this case immutable attribute prevents it.

```
2025-01-28 13:43:56.289 homeassistant NetworkManager[438]: <warn>  [1738071836.2897] keyfile: commit: failure to write 539c2326-a82f-45a6-97a6-82fc242d30da ((null)) to "/etc/NetworkManager/system-connections/wlp1s0.nmconnection": error writing to file '/etc/NetworkManager/system-connections/wlp1s0.nmconnection': failed rename /etc/NetworkManager/system-connections/wlp1s0.nmconnection.HJL802 to /etc/NetworkManager/system-connections/wlp1s0.nmconnection: Operation not permitted
2025-01-28 13:43:56.290 homeassistant NetworkManager[438]: <info>  [1738071836.2897] audit: op="connection-update" uuid="539c2326-a82f-45a6-97a6-82fc242d30da" name="wlp1s0" args="802-11-wireless.powersave,connection.id,connection.timestamp" pid=1079 uid=0 result="fail" reason="failed to update connection: error writing to file '/etc/NetworkManager/system-connections/wlp1s0.nmconnection': failed rename /etc/NetworkManager/system-connections/wlp1s0.nmconnection.HJL802 to /etc/NetworkManager/system-connections/wlp1s0.nmconnection: Operation not permitted"
```

Why powersave option overwrites? How it corresponds with HAOS network configuration documentation? https://github.com/home-assistant/operating-system/blob/dev/Documentation/network.md#powersave

dgdavid, 2025-01-31:
Finally, I didn't apply the workaround but it has been working without experiencing the https://github.com/home-assistant/operating-system/issues/3826 issue for a couple of days already. The only change I did was to update to the newest https://github.com/home-assistant/operating-system/releases/tag/14.2 version released 4 days ago which upgrades the Linux kernel from 6.6.71 to 6.6.73. Maybe some of the drivers update on [6.6.72](https://lwn.net/Articles/1005383/) solved the issue, don't know.

closed 2025-05-09 by github-actions[bot]