home-assistant/operating-system #3837: SATA M2 drive not available after Odroid M1S upgrade to 14.2

marco44, 2025-01-27:
### Describe the issue you are experiencing

I've got a M2 SATA drive as the data volume on my Odroid M1S. After upgrading to 14.2, the device is not available. I can't see it in lsblk, and nothing can be started, because of 
```
containerd: mkdir /mnt/data/docker: read-only file system
```

### What operating system image do you use?

odroid-m1s (Hardkernel ODROID-M1S)

### What version of Home Assistant Operating System is installed?

14.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Toshiba NVMe SSD KBG30ZMV256G

### Steps to reproduce the issue

Upgrade to HAOS 14.2
Don't have access to your M2 drive anymore

### Anything in the Supervisor logs that might be useful for us?

```txt
I can't reach anything, as nothing starts
```

### Anything in the Host logs that might be useful for us?

```txt
containerd: mkdir /mnt/data/docker: read-only file system
```

### System information

Can't reach this

sairon, 2025-01-28:
Yes, I can confirm this. I have suspicion it's related to #3841, it broke between the same dev builds. As a workaround, revert to 14.1 (but as I see you already did). Sorry for the inconvenience.

sairon, 2025-01-28:
Broken by [this](https://github.com/gregkh/linux/commit/908806a2e789998f8897ebab83be10b4b058b6e2) stable Linux commit, there is no backport of the needed [device tree changes](https://lore.kernel.org/all/20241122073006.99309-1-amadeus@jmu.edu.cn/) and the fallback suggested in the thread wasn't applied either 😢

PR #3843 (2025-01-30): Revert kernel patch causing USB 3 and PCIe breakage on ODROID-M1/M1S
Revert the patch changing phy reset behavior, requiring also changes in the device tree that are missing in the stable backport. The issue was reported to the regressions mailing list and hopefully future patch release should contain a proper fix.

The patch is added to the patches-rockchip directory, potentially affecting Green as well, although the broken peripherals are not used there.

Fixes #3837, fixes #3841

## Summary by CodeRabbit

- **Revert**
	- Reverted a previous PHY reset patch for Rockchip naneng-combphy driver [... cut]
commit f56130af (2025-01-30): Revert kernel patch causing USB 3 and PCIe breakage on ODROID-M1/M1S (#3843)

Revert the patch changing phy reset behavior, requiring also changes in the
device tree that are missing in the stable backport. The issue was reported to
the regressions mailing list and hopefully future patch release should contain
a proper fix.

The patch is added to the patches-rockchip directory, potentially affecting
Green as well, although the broken peripherals are not used there.

Fixes #3837, fixes #3841
closed 2025-01-30 by sairon