home-assistant/operating-system #3838: System not start

kaidem4, 2025-01-28:
### Describe the issue you are experiencing

After update to last version, the system dont start anymore. black screen and not keyboard abailable. Only ping responses.
rpi 5 8Gb

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

14.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

rpi5

### Steps to reproduce the issue

1.automatic from ha web
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
i cant access the system to view
```

### Anything in the Host logs that might be useful for us?

```txt
The sistem not start. I cant view anything
```

bigskookum, 2025-01-28:
I also got this, upgrading to 14.2 from 14.1 on x86. I managed to boot back into 14.1 using this procedure:

https://github.com/home-assistant/operating-system/issues/2272#issuecomment-1600300443

The  /dev/sda8 partition was named haaos-data-dis

idcrook, 2025-01-28:
Doing HAOS 14.2 Update from 14.1 I also got an unresponsive system.  rpi5-64 as well.

idcrook, 2025-01-28:
HAOS seems to know it was "bad"

> _Update_: When plugged back in power to troubleshoot, the system booted into HAOS **14.1** -- and Settings update was offering **14.2** Update again.
> 
> FWIW, rpi5 4GB HAOS on m.2 NVMe SSD

```
ha > os info 
board: rpi5-64
boot: A
boot_slots:
  A:
state: booted status: good version: "14.1"
  B:
state: inactive
status: bad 
version: "14.2"
data_disk: INTEL-SSDPEKNUO10TZ-PHKA12630BBY1P0B
update_available: true 
version: "14.1"
version_latest: "14.2"
```

dengi76, 2025-01-28:
Hello! I want to say that I have the same thing on my virtual box!

VHC2020, 2025-01-28:
i get this afther update en try to restart in any way normal of save

Failed to perform the action homeassistant/restart. The system cannot restart because the configuration is not valid: Error loading /config/configuration.yaml: while scanning for the next token found character '%' that cannot start any token in "/config/sensors.yaml", line 8, column 12

i did not chnage any thing there just a update, is lucky my test server main server will not be updated so far

dengi76, 2025-01-28:
I managed to update on the fourth try!

larrybml, 2025-01-28:
How did you managed to update? I can't connect to HA RPi5 anymore.

dengi76, 2025-01-28:
Three times in a row we pressed update, and all three times, after rebooting, the old version remained. After the fourth attempt to update, HA rebooted and the new version 14.2 was installed

larrybml, 2025-01-28:
OK, thank you!

petepall, 2025-01-28:
I have a home assistant green and am no longer able to get it to boot after the 14.2 OS update.
I don't have a HDMI monitor available to check the issue how can I get this to work again?

larrybml, 2025-01-28:
Same here, after 3 failed tries to update, I gave up, still on 14.1 , HA OS installed on an RPi5.

claytonjcoleman, 2025-01-28:
Pushed the button on 14.2, failed to come up. I'm remote so could not see anything on a screen. IP was not offered from DHCP from the router so not pingable. Made me consider a cloud connected switch to reboot this. Asked somone at home to pull power and it came back up -my octopus rates card, restored and all back to 14.1.

DieterC1984, 2025-01-28:
> I have a home assistant green and am no longer able to get it to boot after the 14.2 OS update. I don't have a HDMI monitor available to check the issue how can I get this to work again?
> 
> Update: After 4 reboots of the system I have access again and system returned back to OS14.1

Same for me!

DieterC1984, 2025-01-28:
> Three times in a row we pressed update, and all three times, after rebooting, the old version remained. After the fourth attempt to update, HA rebooted and the new version 14.2 was installed

Same for me. RPI5

VHC2020, 2025-01-28:
I also rebooted Hard cut power off en on, came in a recovery mode every thing was back core update and os update, but had problems with the setting sensors.yaml i trun it of and reboot system was ok one version lower updates are there and i don't touch them, now fix my sensors.yaml who i did not touch when i was updaten pretty strange.

linux4life798, 2025-01-28:
For the people experiencing the Raspberry Pi not booting, after a system reboot (have to do a hard power-off and power-on to recover). It is likely #3740.

jhalldn, 2025-01-28:
Seems like there's many ppl having issues with the upgrade to 14.2 - also me too. I would like to see a new build or version that is possible to upgrade without any issues. I'm stay on 14.1   After a faulty attempt to do 14.2.

anilkusaksiz, 2025-01-28:
Upgraded to 14.2 today and it started crashing every few minutes

[image]

Won't recommend updating to 14.2 to anyone

I'm running HA on VirtualBox on Windows 11.

villasenor, 2025-01-29:
@sairon FYI I know you are busy, but thought I'd bump this to the top of your notifications because this looks like a breaking regression, and seems different than past issues because it is **happening on Home Assistant Green** and VMs (not just RPI). I haven't experienced this yet because I haven't done the update yet, but would it be worth rolling back the release until the root cause is determined?

spencermacca, 2025-01-29:
please fix hoe assistat

ManiakNL, 2025-01-29:
> please fix hoe assistat

I think that is another service buddy :)

spencermacca, 2025-01-29:
> > please fix hoe assistat
> 
> I think that is another service buddy :)

typo my bad

sairon, 2025-01-29:
> [@sairon](https://github.com/sairon) FYI I know you are busy, but thought I'd bump this to the top of your notifications because this looks like a breaking regression, and seems different than past [...]

@villasenor  While we can't rule out anything, I don't feel there is a strong indication about a breaking regression affecting more platforms. Of course, the kernel update could have introduced some issues for particular hypervisors or Green as well but many other systems are running fine and I am not able to reproduce anything similar on my end either (upgraded Green couple of times in the morning and now it's running for couple of hours straight). Basically, with ~1M installs, every update which results in a host reboot wakes up some gremlins that were waiting for their prime time - if there's a real issue, we need to figure out the pattern, and we need logs for that.

What I unfortunately can't deny is that there's an issue with Raspberry Pi 5, it's been reported in couple of threads already and some summary can be found in post by @linux4life798. However, that is likely an intermittent problem that couldn't have been introduced by 14.2, as there are simply no changes between 14.1 and 14.2 that could have caused it. So far the common pattern is that it happens on **RPi 5 running from NVMe on unofficial NVMe HATs**. But being intermittent, it affects different people than before. And also, I still need to be able to successfully reproduce it here, I don't have any of the problematic HATs (yet) but with the official M.2 HAT+ or when running from an SD card, there are no issues.

---

**For everyone, spamming issues with "me too" reports doesn't help anyone. If you want to comment, at least provide more details about your setup, i.e. is it running from an SD card or an NVMe? What HATs and peripherals are connected to it? Can you connect an HDMI display and check what's there? The more information, the better.**

bigskookum, 2025-01-29:
How can I produce logs? My update failed on x86 and rolled itself back to 14.1.

ManiakNL, 2025-01-29:
> [@villasenor](https://github.com/villasenor) While we can't rule out anything, I don't feel there is a strong indication about a breaking regression affecting more platforms. Of course, the kernel [...]

I have the following connected to my RPi-5

SSD connected using a SATA to  USB3 cable, the OS is installed on the SATA drive, the PI does not have a SD card inserted.
1 Zigbee Dongle (Conbee II)
1 Zigbee Dongle (SONOFF ZigBee 3.0 USB Dongle Plus)

spencermacca, 2025-01-29:
> **For everyone, spamming issues with "me too" reports doesn't help anyone. If you want to comment, at least provide more details about your setup, i.e. is it running from an SD card or an NVMe? What [...]

out of the box raspi5 with out of the box raspi power supply

TechSysEng, 2025-01-29:
I disagree with the manager that this is not a real issue. Same issues as other are stating here on a Pi5. I tried 3 times to update from 14.1 to 14.2. After the update it reloads and takes a good 30 minutes or more to come back up at which time I get SSL handshake errors. I then hard power cycle the Pi and then it will come back up, and still on 14.1. No custom integrations, all basic things.

Core 2025.1.4
Supervisor 2024.12.3
Operating System 14.1
Frontend 20250109.2

2025-01-27 21:08:22.818 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
File "/usr/local/lib/python3.12/site-packages/aiohttp/client_proto.py", line 92, in connection_lost
uncompleted = self._parser.feed_eof()
^^^^^^^^^^^^^^^^^^^^^^^
File "aiohttp/_http_parser.pyx", line 508, in aiohttp._http_parser.HttpParser.feed_eof
aiohttp.http_exceptions.TransferEncodingError: 400, message:
Not enough data for satisfy transfer length header.
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request
resp = await request_handler(request)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/aiohttp/web_app.py", line 569, in _handle
return await handler(request)
^^^^^^^^^^^^^^^^^^^^^^
aiohttp.client_exceptions.ClientPayloadError: Response payload is not completed: <TransferEncodingError: 400, message='Not enough data for satisfy transfer length header.'>
2025-01-27 21:08:22.821 ERROR (MainThread) [supervisor.api] Failed to get supervisor logs using advanced_logs API
Traceback (most recent call last):
File "/usr/local/lib/python3.12/site-packages/aiohttp/client_proto.py", line 92, in connection_lost
uncompleted = self._parser.feed_eof()
aiohttp.client_exceptions.ClientPayloadError: Response payload is not completed: <TransferEncodingError: 400, message='Not enough data for satisfy transfer length header.'>
s6-rc: info: service legacy-services: stopping
2025-01-27 21:08:23.236 INFO (MainThread) [supervisor.misc.scheduler] Shutting down scheduled tasks
2025-01-27 21:08:23.237 INFO (MainThread) [supervisor.docker.monitor] Stopped docker events monitor
2025-01-27 21:08:23.238 INFO (MainThread) [supervisor.api] Stopping API on 172.30.32.2
[02:08:23] INFO: Watchdog restart after closing
2025-01-27 21:08:23.247 INFO (MainThread) [supervisor.hardware.monitor] Stopped Supervisor hardware monitor
2025-01-27 21:08:23.250 INFO (MainThread) [supervisor.dbus.manager] Closed conection to system D-Bus.
2025-01-27 21:08:23.252 INFO (MainThread) [supervisor.core] Supervisor is down - 0
2025-01-27 21:08:23.253 INFO (MainThread) [main] Closing Supervisor
[02:08:23] WARNING: Halt Supervisor
[02:08:23] INFO: Supervisor restart after closing

mickhei, 2025-01-29:
Same issue here on Raspi Pi5 8GB with Raspberry Pi SSD Kit with 512GB. No SD card connected. Nothing unofficial from my point of view.

TechSysEng, 2025-01-29:
@sairon those logs below per your request. The second produced no output

[image]

mickhei, 2025-01-29:
[image]

[image]

sairon, 2025-01-29:
@TechSysEng @mickhei I should have emphasized the "right after the update" part, the times are somewhat too far in the past. Also make sure to properly add all flags and not clip anything from the output.

In the case when the second command didn't produce anything, what's the output of `ha host logs -v -b-1 -n 100`?

mickhei, 2025-01-29:
@sairon Sorry, I thought you meant from the point where the update was tried. That was on the 27.01.2025 in my case. No more data for the 27.01.

TechSysEng, 2025-01-29:
@sairon I did it 3 times the other night so those are all the logs I have. If I get brave and want to risk it being down for another night, maybe I'll try again to get fresh logs when i get home frm work and try it again. nothing was missing from my screenshot as far as the output goes from the first command. I do see in that screenshot there is a line referencing 14.2. If there is data there, (pardon my ignorance) regardless of from when and the timestamps, it still provides no value? As far as the first command goes.

The second command gives no output or error, just back to a new line

[image]

The output of ha host logs -v -b-1 -n 100 is probably going to be too old as it only goes back a day. 

[image]

[image]

[image]

TechSysEng, 2025-01-29:
@sairon I got home and went to update to break it for the cause. :) I updated and this 4th try worked. VERY bizarre. Regardless, below is the FULL output from your original 2 requested commands. Not sure if it helps anymore since it worked this time. This was very shortly after it updated, I pulled these. Maybe it still helps.

[core-ssh ~]$ ha host logs -v -n100 -t rauc
2025-01-28 01:44:43.116 homeassistant rauc[2288]: tryboot set-state B good
2025-01-28 01:44:43.124 homeassistant rauc[2291]: tryboot get-state A
2025-01-28 01:44:43.124 homeassistant rauc[2291]: returning bad
2025-01-28 01:44:43.126 homeassistant rauc[2292]: tryboot get-state B
2025-01-28 01:44:43.127 homeassistant rauc[2292]: returning good
2025-01-28 01:45:55.766 homeassistant rauc[4385]: tryboot get-state A
2025-01-28 01:45:55.766 homeassistant rauc[4385]: returning bad
2025-01-28 01:45:55.768 homeassistant rauc[4387]: tryboot get-state B
2025-01-28 01:45:55.768 homeassistant rauc[4387]: returning good
2025-01-28 02:06:35.205 homeassistant rauc[530]: input bundle: /mnt/data/supervisor/tmp/hassos-14.2.raucb
2025-01-28 02:06:35.269 homeassistant rauc[6930]: tryboot set-state A bad
2025-01-28 02:07:34.998 homeassistant rauc[7047]: tryboot set-primary A
2025-01-28 12:23:55.304 homeassistant rauc[1816]: tryboot get-state A
2025-01-28 12:23:55.304 homeassistant rauc[1816]: returning bad
2025-01-28 12:23:55.306 homeassistant rauc[1818]: tryboot get-state B
2025-01-28 12:23:55.307 homeassistant rauc[1818]: returning good
2025-01-28 12:23:57.970 homeassistant rauc[2272]: tryboot set-primary B
2025-01-28 12:23:58.014 homeassistant rauc[2285]: tryboot set-state B good
2025-01-28 12:23:58.022 homeassistant rauc[2288]: tryboot get-state A
2025-01-28 12:23:58.022 homeassistant rauc[2288]: returning bad
2025-01-28 12:23:58.024 homeassistant rauc[2289]: tryboot get-state B
2025-01-28 12:23:58.024 homeassistant rauc[2289]: returning good
2025-01-28 12:25:12.322 homeassistant rauc[4380]: tryboot get-state A
2025-01-28 12:25:12.322 homeassistant rauc[4380]: returning bad
2025-01-28 12:25:12.323 homeassistant rauc[4382]: tryboot get-state B
2025-01-28 12:25:12.323 homeassistant rauc[4382]: returning good
2025-01-28 14:31:52.284 homeassistant rauc[14975]: tryboot get-state A
2025-01-28 14:31:52.284 homeassistant rauc[14975]: returning bad
2025-01-28 14:31:52.286 homeassistant rauc[14977]: tryboot get-state B
2025-01-28 14:31:52.286 homeassistant rauc[14977]: returning good
2025-01-28 16:38:32.811 homeassistant rauc[25092]: tryboot get-state A
2025-01-28 16:38:32.811 homeassistant rauc[25092]: returning bad
2025-01-28 16:38:32.813 homeassistant rauc[25094]: tryboot get-state B
2025-01-28 16:38:32.813 homeassistant rauc[25094]: returning good
2025-01-28 18:45:13.264 homeassistant rauc[34936]: tryboot get-state A
2025-01-28 18:45:13.264 homeassistant rauc[34936]: returning bad
2025-01-28 18:45:13.266 homeassistant rauc[34938]: tryboot get-state B
2025-01-28 18:45:13.266 homeassistant rauc[34938]: returning good
2025-01-28 20:51:53.706 homeassistant rauc[44172]: tryboot get-state A
2025-01-28 20:51:53.706 homeassistant rauc[44172]: returning bad
2025-01-28 20:51:53.708 homeassistant rauc[44174]: tryboot get-state B
2025-01-28 20:51:53.708 homeassistant rauc[44174]: returning good
2025-01-28 22:58:34.249 homeassistant rauc[54131]: tryboot get-state A
2025-01-28 22:58:34.249 homeassistant rauc[54131]: returning bad
2025-01-28 22:58:34.251 homeassistant rauc[54133]: tryboot get-state B
2025-01-28 22:58:34.251 homeassistant rauc[54133]: returning good
2025-01-29 01:05:14.681 homeassistant rauc[64054]: tryboot get-state A
2025-01-29 01:05:14.681 homeassistant rauc[64054]: returning bad
2025-01-29 01:05:14.683 homeassistant rauc[64056]: tryboot get-state B
2025-01-29 01:05:14.683 homeassistant rauc[64056]: returning good
2025-01-29 03:11:55.135 homeassistant rauc[73304]: tryboot get-state A
2025-01-29 03:11:55.135 homeassistant rauc[73304]: returning bad
2025-01-29 03:11:55.138 homeassistant rauc[73306]: tryboot get-state B
2025-01-29 03:11:55.138 homeassistant rauc[73306]: returning good
2025-01-29 05:18:35.578 homeassistant rauc[83061]: tryboot get-state A
2025-01-29 05:18:35.578 homeassistant rauc[83061]: returning bad
2025-01-29 05:18:35.579 homeassistant rauc[83063]: tryboot get-state B
2025-01-29 05:18:35.580 homeassistant rauc[83063]: returning good
2025-01-29 07:25:16.021 homeassistant rauc[92841]: tryboot get-state A
2025-01-29 07:25:16.021 homeassistant rauc[92841]: returning bad
2025-01-29 07:25:16.023 homeassistant rauc[92843]: tryboot get-state B
2025-01-29 07:25:16.023 homeassistant rauc[92843]: returning good
2025-01-29 09:31:56.461 homeassistant rauc[102147]: tryboot get-state A
2025-01-29 09:31:56.461 homeassistant rauc[102147]: returning bad
2025-01-29 09:31:56.463 homeassistant rauc[102149]: tryboot get-state B
2025-01-29 09:31:56.463 homeassistant rauc[102149]: returning good
2025-01-29 11:38:36.911 homeassistant rauc[110671]: tryboot get-state A
[... 110 lines cut ...]
2025-01-29 19:31:31.733 homeassistant hassio_supervisor[626]:     line = await resp.content.readuntil(b"\n")
2025-01-29 19:31:31.733 homeassistant hassio_supervisor[626]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-01-29 19:31:31.733 homeassistant hassio_supervisor[626]:   File "/usr/local/lib/python3.12/site-packages/aiohttp/streams.py", line 386, in readuntil
2025-01-29 19:31:31.733 homeassistant hassio_supervisor[626]:     await self._wait("readuntil")
[... cut ...]
2025-01-29 19:31:32.869 homeassistant hassio_supervisor[626]: s6-rc: info: service legacy-cont-init successfully stopped
2025-01-29 19:31:32.869 homeassistant hassio_supervisor[626]: s6-rc: info: service fix-attrs: stopping
2025-01-29 19:31:32.870 homeassistant hassio_supervisor[626]: s6-rc: info: service fix-attrs successfully stopped
2025-01-29 19:31:32.870 homeassistant hassio_supervisor[626]: s6-rc: info: service s6rc-oneshot-runner: stopping
2025-01-29 19:31:32.875 homeassistant hassio_supervisor[626]: s6-rc: info: service s6rc-oneshot-runner successfully stopped

ktvanzwol, 2025-01-31:
@sairon attached output of both commands right after a failed upgrade. I had to power cycle the RPi5 to get it boot again.

[host.log](https://github.com/user-attachments/files/18621392/host.log)
[supervisor.log](https://github.com/user-attachments/files/18621393/supervisor.log)

erkr, 2025-01-31:
Just to chime in. I had to power cycle my fresh imaged (and a restored backup) RPi5-8Gb 3 times, the first time I rebooted the pi. That happened already with HAOS 14.1, not 14.2

iyanmv, 2025-02-02:
> For everyone, spamming issues with "me too" reports doesn't help anyone. If you want to comment, at least provide more details about your setup, i.e. is it running from an SD card or an NVMe? What [...]

RPi5, running on SD card only, no HATs, and two USBs attached (SONOFF Zigbee 3.0 USB Dongle Plus V2 and Aeotec Z-Stick 7). Upgrading to 14.2 makes HA unreachable. Scanning all the ports with nmap it seems nothing is listening on port 8123, so the OS must freeze/crash before it starts listening on that port. After power cycling the RPi5, HA is back in version 14.1 and everything works just fine.

nemzoles, 2025-02-02:
> > For everyone, spamming issues with "me too" reports doesn't help anyone. If you want to comment, at least provide more details about your setup, i.e. is it running from an SD card or an NVMe? What HATs and peripherals are connected to it? Can you connect an HDMI display and check what's there? The more information, the better.
> 
> RPi5, running on SD card only, no HATs, and two USBs attached (SONOFF Zigbee 3.0 USB Dongle Plus V2 and Aeotec Z-Stick 7). Upgrading to 14.2 makes HA unreachable. Scanning all the ports with nmap it seems nothing is listening on port 8123, so the OS must freeze/crash before it starts listening on that port. After power cycling the RPi5, HA is back in version 14.1 and everything works just fine.
> 
> The RPi5 is the 8GB model and I'm using the official RPi5 power supply, 27W USB-C.

I have the same setup same behaviour... onlöy difference, no zigbee dongle but ENOCEAN USB Dongle

idcrook, 2025-02-06:
> Anyone with RPi 5 update failing, can you please post the output of `ha host logs -v -n100 -t rauc` and `ha host logs -v -b-1 -t hassio_supervisor` obtained right after the update fails and boots back to the old version?

I had intended to report back with this logs after what I had assumed was going to be a failed update attempt. Have a rpi5 4GB with non-official NVMe HAT. 

I powered down the 14.1  HAOS that I been staying on after may earlier failed update attempt. Plugging in USB keyboard, mouse, and HDMI. powered up and using the CLI `ha > os update`. the text output said the update was successful, and system re-booted itself. When it came back up was operating HAOS 14.2 🤷

derBertW, 2025-02-06:
I had the base problem of the ticket with Raspi 5. After updating HomeAssistant Core to 2025.2.0 today I could also apply the update of the Operating System to 14.2. I don't know if this is just a coincidence or a causal relationship.

derBertW, 2025-02-06:
yes and yes. HACS is version 2.0.5.

tripplehelix, 2025-02-06:
Thanks, updated, restarted and all is good!

Mizokuiam, 2025-03-07:
Hi there @kaidem4! It seems like several users are hitting a black screen after updating HAOS to 14.2, particularly on RPi5.  The system remains powered on (responds to pings) but becomes inaccessible.

While a few users managed to recover by reverting to 14.1 (see bigskookum's comment with the data partition mounting method), or even by repeatedly attempting the update, these aren't ideal solutions.  idcrook's observation of the system booting into 14.1 after a power cycle with peripherals attached hints at a possible boot process issue. This might be related to display output initialization or some peripheral dependency during early boot.

If you're encountering this issue:

1. **Try the peripheral trick:** Power down your RPi5, attach a USB keyboard, mouse, and HDMI monitor, and then power it back on. See if it boots into 14.1 or even directly into 14.2.  If it boots into 14.1, try the 14.2 update again.

2. **Check your boot partition:** If the peripheral approach fails, try bigskookum's suggestion of reverting to 14.1 by manually mounting the data partition.  This requires some technical comfort with Linux commands.

3. **If you're comfortable on the command line:**  If you can access the system at all (even just a console), check the logs (`journalctl -xe`) for any error messages related to the boot process, display drivers, or kernel modules. Share those logs here.

4. **As a last resort:** A fresh install of HAOS 14.2 might be necessary, but this means restoring from a backup.

The repeated update success reported by dengi76 suggests a possible transient issue within the update process.  If you're less comfortable with manual recovery, attempting the update a few more times *might* work, but isn't guaranteed.  

It's important to distinguish the black screen issue from configuration errors like the one VHC2020 reported.  If you get a configuration error, that's a separate problem likely related to your `configuration.yaml` or included files, and needs to be addressed by checking your YAML syntax.

Let's gather more data on this to pinpoint the root cause. Please report back your experiences with these suggestions, including your hardware setup (RPi version, storage type - SD card, SSD, etc.), and any error messages you encounter.

Hope that resolves your issue!

tripplehelix, 2025-03-22:
It failed to start for me after the recent updating, but when I plugged in a display the device booted up without issue... weird.

derBertW, 2025-03-22:
Last time I could update HAOS after updating HA core. This time I updated core from 2025.3.3 to 2025.3.4. Still could not update HAOS. I might try the display though it really sounds weird.

derBertW, 2025-03-23:
After some reboots and retries HAOS 15.0 is now running. I don't know what made the difference. So, yes, it is weird.

tripplehelix, 2025-06-29:
OK, annoyingly, this most recent update seems to also have this issue.

apeltzer, 2025-07-17:
Yup, same here