home-assistant/operating-system #3839: Some USB device no longer detected

GraemeDBlue, 2025-01-28:
### Describe the issue you are experiencing

On Upgrading my USB Coral device for Frigate running on same  Generic x86-64 hardware, as an add on  is no longer detected and I see this in the Frigate logs on attempted start of the add on 

```
2025-01-28 09:16:34.345188064  [2025-01-28 09:16:34] frigate.detectors.plugins.edgetpu_tfl INFO    : Attempting to load TPU as usb
2025-01-28 09:16:34.359369421  [2025-01-28 09:16:34] frigate.app                    INFO    : Output process started: 389
2025-01-28 09:16:34.359388537  [2025-01-28 09:16:34] frigate.detectors.plugins.edgetpu_tfl ERROR   : No EdgeTPU was detected. If you do not have a Coral device yet, you must configure CPU detectors.
2025-01-28 09:16:34.359395088  Process detector:coral:
2025-01-28 09:16:34.363232986  Traceback (most recent call last):
2025-01-28 09:16:34.363244607    File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 160, in load_delegate
2025-01-28 09:16:34.363249051      delegate = Delegate(library, options)
2025-01-28 09:16:34.363254083    File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 119, in __init__
2025-01-28 09:16:34.363258256      raise ValueError(capture.message)
2025-01-28 09:16:34.367372468  ValueError
2025-01-28 09:16:34.367381385  
2025-01-28 09:16:34.367385517  During handling of the above exception, another exception occurred:
2025-01-28 09:16:34.367388177  
2025-01-28 09:16:34.367391201  Traceback (most recent call last):
2025-01-28 09:16:34.367394730    File "/usr/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
2025-01-28 09:16:34.367397407      self.run()
2025-01-28 09:16:34.367400772    File "/usr/lib/python3.9/multiprocessing/process.py", line 108, in run
2025-01-28 09:16:34.367405651      self._target(*self._args, **self._kwargs)
2025-01-28 09:16:34.367446094    File "/opt/frigate/frigate/object_detection.py", line 102, in run_detector
2025-01-28 09:16:34.367450590      object_detector = LocalObjectDetector(detector_config=detector_config)
2025-01-28 09:16:34.367454253    File "/opt/frigate/frigate/object_detection.py", line 53, in __init__
2025-01-28 09:16:34.367457664      self.detect_api = create_detector(detector_config)
2025-01-28 09:16:34.367516888    File "/opt/frigate/frigate/detectors/__init__.py", line 18, in create_detector
2025-01-28 09:16:34.367520742      return api(detector_config)
2025-01-28 09:16:34.367552065    File "/opt/frigate/frigate/detectors/plugins/edgetpu_tfl.py", line 41, in __init__
2025-01-28 09:16:34.367556626      edge_tpu_delegate = load_delegate("libedgetpu.so.1.0", device_config)
2025-01-28 09:16:34.367560593    File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 162, in load_delegate
2025-01-28 09:16:34.367563978      raise ValueError('Failed to load delegate from {}\n{}'.format(
2025-01-28 09:16:34.367567510  ValueError: Failed to load delegate from libedgetpu.so.1.0
```

Unplugging the Coral lets the add on boot again , so my guess is driver for Coral on latest OS update?

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.73-haos

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

- AMD Quad Core, 8GB DDR4, 32GB SSD, USB3
- SONOFF Zigbee 3.0 USB Dongle Plus Stick Zigbee TI CC2652P+CP2102 (N) Coordinator
- Google Coral USB Edge TPU ML Accelerator
- USB bluetooth device

### Steps to reproduce the issue

1. Reboot HAOS
2. Start the Frigate add on
3. Frigate Add on crashes 
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2025-01-28 09:17:52.639 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-14299' coro=<Addon.watchdog_container() done, defined at /usr/src/supervisor/supervisor/addons/addon.py:1506> exception=AddonsJobError('Rate limit exceeded, more than 10 calls in 0:30:00')>
Traceback (most recent call last):
[... 57 lines cut ...]
2025-01-28 10:30:05.133711204    File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 119, in __init__
2025-01-28 10:30:05.133714547      raise ValueError(capture.message)
2025-01-28 10:30:05.133717647  ValueError
2025-01-28 10:30:05.134388977  
2025-01-28 10:30:05.134400173  During handling of the above exception, another exception occurred:
2025-01-28 10:30:05.134403394  
2025-01-28 10:30:05.134683633  Traceback (most recent call last):
2025-01-28 10:30:05.135683163    File "/usr/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
2025-01-28 10:30:05.135692237      self.run()
2025-01-28 10:30:05.135696249    File "/usr/lib/python3.9/multiprocessing/process.py", line 108, in run
2025-01-28 10:30:05.135700172      self._target(*self._args, **self._kwargs)
2025-01-28 10:30:05.135704080    File "/opt/frigate/frigate/object_detection.py", line 102, in run_detector
2025-01-28 10:30:05.135707979      object_detector = LocalObjectDetector(detector_config=detector_config)
2025-01-28 10:30:05.135714182    File "/opt/frigate/frigate/object_detection.py", line 53, in __init__
2025-01-28 10:30:05.135770608      self.detect_api = create_detector(detector_config)
2025-01-28 10:30:05.135775178    File "/opt/frigate/frigate/detectors/__init__.py", line 18, in create_detector
2025-01-28 10:30:05.135778643      return api(detector_config)
2025-01-28 10:30:05.135782454    File "/opt/frigate/frigate/detectors/plugins/edgetpu_tfl.py", line 41, in __init__
2025-01-28 10:30:05.135845729      edge_tpu_delegate = load_delegate("libedgetpu.so.1.0", device_config)
2025-01-28 10:30:05.135850266    File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 162, in load_delegate
2025-01-28 10:30:05.135880420      raise ValueError('Failed to load delegate from {}\n{}'.format(
2025-01-28 10:30:05.135884468  ValueError: Failed to load delegate from libedgetpu.so.1.0
```

I have tried multiple USB ports on my device, USB2 and USB3 and none of them seem to work for the Coral anymore

jacknife71, 2025-01-28:
Same issue here.

Using ODROID M1.

2 of 4 USB-Ports are not working anymore.
Actually 3 USB-Ports used
Switching Sticks to working ports is a temporary workaround.

sairon, 2025-01-28:
@GraemeDBlue Does reverting to 14.1 (using the commands from the previous comment) fix the issue? Have you tried whether a cold boot (unplugging the PC for a while) doesn't fix it? If not, could you please post the complete host logs when it doesn't work?

jacknife71, 2025-01-28:
Downgrade to 14.1 works.
All USB Ports are online now.
Thanks for Support.
Regards
Jacknife

GraemeDBlue, 2025-01-28:
@sairon unfortunately reverting to 14.1 does not seem to solve the issue for me . Coral is just no longer detected

GraemeDBlue, 2025-01-28:
@sairon actually I lie, it does now work again on 14.1, but against all suggestions "Protection mode" needs to be ON for it to work, which previously it was set to OFF

GraemeDBlue, 2025-01-28:
@sairon bumped to 14.2 with above change and it still works, so unsure how that flag can have been changed to now be opposite to suggested setting for using a Coral

sairon, 2025-01-29:
Not sure what's going on there with Frigate, but for the device not appearing in the system, that seems to be a different problem - the device wasn't enumerated by the kernel (returning error 71 = protocol error), something that wouldn't be fixed by toggling the protection mode. Possible reasons are a bad USB cable or interference, bit less likely general problem with the hardware or a kernel regression.

GraemeDBlue, 2025-01-29:
@sairon yep makes no sense, but its still working  with the protection mode turned on, really makes no sense at all

closed 2025-05-07 by github-actions[bot]