home-assistant/operating-system #3841: USB 3 ports don't work on ODROID-M1 in HAOS 14.2

sairon, 2025-01-28:
> Same issue here.
> 
> Using ODROID M1.
> 
> 2 of 4 USB-Ports are not working anymore.
> Actually 3 USB-Ports used
> Switching Sticks to working ports is a temporary workaround. 
>  

 _Originally posted by @jacknife71 in [#3839](https://github.com/home-assistant/operating-system/issues/3839#issuecomment-2618796547)_

escoand, 2025-01-28:
For me the internal M2 SSD is not found anymore.

sairon, 2025-01-28:
@escoand See the linked issue(s), it's a known problem, a stable kernel regression introduced by a patch in Linux 6.6.69. We have pulled back the 14.2 for M1/M1S in the meantime.

sairon, 2025-01-29:
@zoltanf Right, the nature of the bug (explained in the other issue) results in NVMe and USB 3 peripherals broken both on M1 and M1S. I have updated the title to clarify that and added a notice to the release notes.

PR #3843 (2025-01-30): Revert kernel patch causing USB 3 and PCIe breakage on ODROID-M1/M1S
Revert the patch changing phy reset behavior, requiring also changes in the device tree that are missing in the stable backport. The issue was reported to the regressions mailing list and hopefully future patch release should contain a proper fix.

The patch is added to the patches-rockchip directory, potentially affecting Green as well, although the broken peripherals are not used there.

Fixes #3837, fixes #3841

## Summary by CodeRabbit

- **Revert**
	- Reverted a previous PHY reset patch for Rockchip naneng-combphy driver [... cut]
commit f56130af (2025-01-30): Revert kernel patch causing USB 3 and PCIe breakage on ODROID-M1/M1S (#3843)

Revert the patch changing phy reset behavior, requiring also changes in the
device tree that are missing in the stable backport. The issue was reported to
the regressions mailing list and hopefully future patch release should contain
a proper fix.

The patch is added to the patches-rockchip directory, potentially affecting
Green as well, although the broken peripherals are not used there.

Fixes #3837, fixes #3841
closed 2025-01-30 by sairon