home-assistant/operating-system #3844: Can not update from 14.1 to 14.2 Yellow with cm5 upgrade with ssd and mmc

edhahn4444, 2025-01-29:
### Describe the issue you are experiencing

Many reboots and trying to install the update via the web and ssh.  Never updates and says inactive and bad is os info.

ha os info

board: yellow
boot: A
boot_slots:
  A:
    state: booted
    status: good
    version: "14.1"
  B:
    state: inactive
    status: bad
    version: "14.2"
data_disk: TEAM-TM8FP6512G-TPBF2207070060200211
update_available: true
version: "14.1"
version_latest: "14.2"

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

6.6.62-haos-raspi

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Yellow with CM5 that has both an ssd installed and emmc on the rpi.  Currently booting off the ssd.

### Steps to reproduce the issue

1.try to install update
2.HAOS reboots like nothing happens
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing that i saw but attached from downloading update.

[32m2025-01-29 10:56:52.391 INFO (MainThread) [supervisor.host.manager] Host information reload completed[0m
[32m2025-01-29 10:59:45.086 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://os-artifacts.home-assistant.io/14.2/haos_yellow-14.2.raucb[0m
[32m2025-01-29 10:59:49.305 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-14.2.raucb[0m
[32m2025-01-29 10:59:52.950 INFO (MainThread) [supervisor.os.manager] Install of Home Assistant Operating System 14.2 success[0m
[32m2025-01-29 10:59:52.951 INFO (MainThread) [supervisor.host.control] Initialize host reboot using logind[0m
[32m2025-01-29 10:59:52.951 INFO (MainThread) [supervisor.addons.manager] Phase 'application' stopping 8 add-ons[0m
[32m2025-01-29 10:59:52.962 INFO (SyncWorker_6) [supervisor.docker.manager] Stopping addon_9074a9fa_cloudflared application[0m
[32m2025-01-29 11:00:03.175 INFO (SyncWorker_6) [supervisor.docker.manager] Cleaning addon_9074a9fa_cloudflared application[0m
[32m2025-01-29 11:00:03.192 INFO (SyncWorker_0) [supervisor.docker.manager] Stopping addon_core_configurator application[0m
[32m2025-01-29 11:00:06.413 INFO (SyncWorker_0) [supervisor.docker.manager] Cleaning addon_core_configurator application[0m
[32m2025-01-29 11:00:06.433 INFO (SyncWorker_7) [supervisor.docker.manager] Stopping addon_a0d7b954_bitwarden application[0m
[32m2025-01-29 11:00:09.775 INFO (SyncWorker_7) [supervisor.docker.manager] Cleaning addon_a0d7b954_bitwarden application[0m
[32m2025-01-29 11:00:09.791 INFO (SyncWorker_3) [supervisor.docker.manager] Stopping addon_core_piper application[0m
[32m2025-01-29 11:00:13.007 INFO (SyncWorker_3) [supervisor.docker.manager] Cleaning addon_core_piper application[0m
[32m2025-01-29 11:00:13.025 INFO (SyncWorker_6) [supervisor.docker.manager] Stopping addon_core_whisper application[0m
[32m2025-01-29 11:00:16.486 INFO (SyncWorker_6) [supervisor.docker.manager] Cleaning addon_core_whisper application[0m
[32m2025-01-29 11:00:16.501 INFO (SyncWorker_0) [supervisor.docker.manager] Stopping addon_core_openwakeword application[0m
[32m2025-01-29 11:00:19.756 INFO (SyncWorker_0) [supervisor.docker.manager] Cleaning addon_core_openwakeword application[0m
[32m2025-01-29 11:00:19.776 INFO (SyncWorker_7) [supervisor.docker.manager] Stopping addon_7094bb28_docker_wyze_bridge application[0m
[32m2025-01-29 11:00:21.980 INFO (SyncWorker_7) [supervisor.docker.manager] Cleaning addon_7094bb28_docker_wyze_bridge application[0m
[... 223 lines cut ...]
[32m2025-01-29 11:04:50.466 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete[0m
[32m2025-01-29 11:04:50.620 INFO (MainThread) [supervisor.host.services] Updating service information[0m
[32m2025-01-29 11:04:50.626 INFO (MainThread) [supervisor.host.network] Updating local network information[0m
[32m2025-01-29 11:04:50.733 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information[0m
[32m2025-01-29 11:04:50.740 INFO (MainThread) [supervisor.host.manager] Host information reload completed[0m
[32m2025-01-29 11:04:58.709 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_ssh[0m
[32m2025-01-29 11:04:58.713 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_ssh[0m
[32m2025-01-29 11:04:58.715 INFO (MainThread) [supervisor.api.middleware.security] /host/info access from core_ssh[0m
[32m2025-01-29 11:04:58.718 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from core_ssh[0m
[32m2025-01-29 11:05:04.090 INFO (MainThread) [supervisor.api.middleware.security] /os/info access from core_ssh[0m
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### System information

Nothing that I saw

### Additional information

None

sairon, 2025-01-29:
Can you please post the output of `ha host logs -v -n100 -t rauc` and `ha host logs -b-1 -t hassio_supervisor` obtained right after the update fails and boots back to the old version?

edhahn4444, 2025-01-29:
2025-01-29 15:32:13.730 homeassistant rauc[639]: input bundle: /mnt/data/supervisor/tmp/hassos-14.2.raucb
2025-01-29 15:32:13.853 homeassistant rauc[8024]: cp: can't stat '/mnt/boot/EFI/BOOT/grubenv': No such file or directory
2025-01-29 15:32:16.663 homeassistant rauc[8046]: cp: can't stat '/tmp/boot-tmp/grubenv': No such file or directory
2025-01-29 15:59:49.308 homeassistant rauc[525]: input bundle: /mnt/data/supervisor/tmp/hassos-14.2.raucb
2025-01-29 15:59:49.429 homeassistant rauc[8295]: cp: can't stat '/mnt/boot/EFI/BOOT/grubenv': No such file or directory
2025-01-29 15:59:49.709 homeassistant rauc[8298]: cp: can't stat '/tmp/boot-tmp/grubenv': No such file or directory
2025-01-29 16:22:42.413 homeassistant rauc[522]: input bundle: /mnt/data/supervisor/tmp/hassos-14.2.raucb
2025-01-29 16:22:42.520 homeassistant rauc[11624]: cp: can't stat '/mnt/boot/EFI/BOOT/grubenv': No such file or directory
2025-01-29 16:22:42.774 homeassistant rauc[11627]: cp: can't stat '/tmp/boot-tmp/grubenv': No such file or directory

line = await resp.content.readuntil(b"\n")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/streams.py", line 386, in readuntil
    await self._wait("readuntil")
  File "/usr/local/lib/python3.12/site-packages/aiohttp/streams.py", line 347, in _wait
    await waiter
aiohttp.client_exceptions.ClientPayloadError: Response payload is not completed: <TransferEncodingError: 400, message='Not enough data for satisfy transfer length header.'>
2025-01-29 11:24:35.363 ERROR (MainThread) [supervisor.api] Failed to get supervisor logs using advanced_logs API
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/client_proto.py", line 92, in connection_lost
    uncompleted = self._parser.feed_eof()
                  ^^^^^^^^^^^^^^^^^^^^^^^
  File "aiohttp/_http_parser.pyx", line 508, in aiohttp._http_parser.HttpParser.feed_eof
aiohttp.http_exceptions.TransferEncodingError: 400, message:
  Not enough data for satisfy transfer length header.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/api/__init__.py", line 403, in get_supervisor_logs
    return await self._api_host.advanced_logs_handler(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/host.py", line 257, in advanced_logs_handler
    async for cursor, line in journal_logs_reader(resp, log_formatter):
  File "/usr/src/supervisor/supervisor/utils/systemd_journal.py", line 80, in journal_logs_reader
    line = await resp.content.readuntil(b"\n")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/streams.py", line 386, in readuntil
    await self._wait("readuntil")
  File "/usr/local/lib/python3.12/site-packages/aiohttp/streams.py", line 347, in _wait
    await waiter
aiohttp.client_exceptions.ClientPayloadError: Response payload is not completed: <TransferEncodingError: 400, message='Not enough data for satisfy transfer length header.'>
2025-01-29 11:24:35.364 ERROR (MainThread) [supervisor.api] Failed to get supervisor logs using advanced_logs API
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/client_proto.py", line 92, in connection_lost
    uncompleted = self._parser.feed_eof()
                  ^^^^^^^^^^^^^^^^^^^^^^^
  File "aiohttp/_http_parser.pyx", line 508, in aiohttp._http_parser.HttpParser.feed_eof
aiohttp.http_exceptions.TransferEncodingError: 400, message:
  Not enough data for satisfy transfer length header.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/api/__init__.py", line 403, in get_supervisor_logs
    return await self._api_host.advanced_logs_handler(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/host.py", line 257, in advanced_logs_handler
    async for cursor, line in journal_logs_reader(resp, log_formatter):
  File "/usr/src/supervisor/supervisor/utils/systemd_journal.py", line 80, in journal_logs_reader
    line = await resp.content.readuntil(b"\n")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/streams.py", line 386, in readuntil
    await self._wait("readuntil")
  File "/usr/local/lib/python3.12/site-packages/aiohttp/streams.py", line 347, in _wait
    await waiter
aiohttp.client_exceptions.ClientPayloadError: Response payload is not completed: <TransferEncodingError: 400, message='Not enough data for satisfy transfer length header.'>
2025-01-29 11:24:35.365 ERROR (MainThread) [supervisor.api] Failed to get supervisor logs using advanced_logs API
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/client_proto.py", line 92, in connection_lost
[... 15 lines cut ...]
    line = await resp.content.readuntil(b"\n")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/streams.py", line 386, in readuntil
    await self._wait("readuntil")
  File "/usr/local/lib/python3.12/site-packages/aiohttp/streams.py", line 347, in _wait
    await waiter
aiohttp.client_exceptions.ClientPayloadError: Response payload is not completed: <TransferEncodingError: 400, message='Not enough data for satisfy transfer length header.'>
[... cut ...]
2025-01-29 11:24:35.818 INFO (MainThread) [__main__] Closing Supervisor
[16:24:36] WARNING: Halt Supervisor
[16:24:36] INFO: Supervisor restart after closing
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

edhahn4444, 2025-01-29:
Hold on the second command was a few min after a boot so reran it right after and had different reslutls

2025-01-29 15:32:13.730 homeassistant rauc[639]: input bundle: /mnt/data/supervisor/tmp/hassos-14.2.raucb
2025-01-29 15:32:13.853 homeassistant rauc[8024]: cp: can't stat '/mnt/boot/EFI/BOOT/grubenv': No such file or directory
2025-01-29 15:32:16.663 homeassistant rauc[8046]: cp: can't stat '/tmp/boot-tmp/grubenv': No such file or directory
2025-01-29 15:59:49.308 homeassistant rauc[525]: input bundle: /mnt/data/supervisor/tmp/hassos-14.2.raucb
2025-01-29 15:59:49.429 homeassistant rauc[8295]: cp: can't stat '/mnt/boot/EFI/BOOT/grubenv': No such file or directory
2025-01-29 15:59:49.709 homeassistant rauc[8298]: cp: can't stat '/tmp/boot-tmp/grubenv': No such file or directory
2025-01-29 16:22:42.413 homeassistant rauc[522]: input bundle: /mnt/data/supervisor/tmp/hassos-14.2.raucb
2025-01-29 16:22:42.520 homeassistant rauc[11624]: cp: can't stat '/mnt/boot/EFI/BOOT/grubenv': No such file or directory
2025-01-29 16:22:42.774 homeassistant rauc[11627]: cp: can't stat '/tmp/boot-tmp/grubenv': No such file or directory
2025-01-29 16:41:34.518 homeassistant rauc[523]: input bundle: /mnt/data/supervisor/tmp/hassos-14.2.raucb
2025-01-29 16:41:34.622 homeassistant rauc[10630]: cp: can't stat '/mnt/boot/EFI/BOOT/grubenv': No such file or directory
2025-01-29 16:41:34.884 homeassistant rauc[10633]: cp: can't stat '/tmp/boot-tmp/grubenv': No such file or directory
[core-ssh ~]$ ha host logs -b-1 -t hassio_supervisor
2025-01-29 11:28:05.217 INFO (MainThread) [supervisor.misc.tasks] All core tasks are scheduled
2025-01-29 11:28:05.218 INFO (MainThread) [supervisor.core] Supervisor is up and running
2025-01-29 11:28:05.218 INFO (MainThread) [supervisor.host.info] Updating local host information
2025-01-29 11:28:05.219 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
2025-01-29 11:28:05.220 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
2025-01-29 11:28:05.220 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
2025-01-29 11:28:05.220 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
2025-01-29 11:28:05.220 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
2025-01-29 11:28:05.220 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
2025-01-29 11:28:05.249 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
2025-01-29 11:28:05.425 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
2025-01-29 11:28:05.425 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
2025-01-29 11:28:05.430 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
2025-01-29 11:28:05.430 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
2025-01-29 11:28:05.449 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
2025-01-29 11:28:05.449 INFO (MainThread) [supervisor.resolution.checks.base] Run check for no_current_backup/system
2025-01-29 11:28:05.449 INFO (MainThread) [supervisor.resolution.check] System checks complete
2025-01-29 11:28:05.449 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
2025-01-29 11:28:05.577 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2025-01-29 11:28:05.577 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2025-01-29 11:28:05.577 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
2025-01-29 11:28:05.667 INFO (MainThread) [supervisor.host.services] Updating service information
2025-01-29 11:28:05.674 INFO (MainThread) [supervisor.host.network] Updating local network information
2025-01-29 11:28:05.776 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
2025-01-29 11:28:05.784 INFO (MainThread) [supervisor.host.manager] Host information reload completed
2025-01-29 11:38:02.319 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_ssh
2025-01-29 11:38:02.323 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_ssh
2025-01-29 11:38:02.325 INFO (MainThread) [supervisor.api.middleware.security] /host/info access from core_ssh
2025-01-29 11:38:02.327 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from core_ssh
2025-01-29 11:39:17.477 INFO (MainThread) [supervisor.api.middleware.security] /host/logs/identifiers/rauc access from core_ssh
2025-01-29 11:39:58.847 INFO (MainThread) [supervisor.api.middleware.security] /host/logs/boots/-1/identifiers/hassio_supervisor access from core_ssh
2025-01-29 11:41:30.207 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://os-artifacts.home-assistant.io/14.2/haos_yellow-14.2.raucb
2025-01-29 11:41:34.517 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-14.2.raucb
2025-01-29 11:41:38.122 INFO (MainThread) [supervisor.os.manager] Install of Home Assistant Operating System 14.2 success
2025-01-29 11:41:38.123 INFO (MainThread) [supervisor.host.control] Initialize host reboot using logind
[... cut ...]
[16:43:21] INFO: Watchdog restart after closing
[16:43:21] WARNING: Halt Supervisor
[16:43:21] INFO: Supervisor restart after closing
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

sairon, 2025-01-29:
Is the eMMC content empty? For example `blkid` could tell whether there isn't something in collision. Also getting logs directly from the Systemd journal could have more information (e.g. `journalctl -b-1 -n200 --no-pager`) but that both needs to be executed either through the developer SSH on port 22222 or via the USB-serial console on Yellow.

edhahn4444, 2025-01-29:
I don't think the emmc is empty.  It gave me an error right after the migration saying both were in use and did i want to mark the emmc as inactive.  This is out of memory so not exactly what it said.

How do I enable developer ssh?  Right now it says connection refused.

sairon, 2025-01-29:
> I don't think the emmc is empty. It gave me an error right after the migration saying both were in use and did i want to mark the emmc as inactive. This is out of memory so not exactly what it said.

If that means that HAOS was installed both to the internal eMMC and on the NVMe, then it's almost surely the cause of the problem.

> How do I enable developer ssh? Right now it says connection refused.

Here's how: https://developers.home-assistant.io/docs/operating-system/debugging/
However, I think that the [serial console](https://yellow.home-assistant.io/guides/use-serial-console-linux-macos/) is a bit more convenient to use, just make sure that the USB/UART jumper is in the correct position (i.e. UART).

pdonn, 2025-01-29:
I think I have the same problem.
Could you please give us a hint what is the most convinient way to delete the emmc?

Andware, 2025-01-29:
And I think I have the same problem too.
What is the best way to clear or format the eMMC?
Is it to use the rpiboot tool?

pdonn, 2025-01-30:
it is very easy to solve. Just use the rpiboot tool  and then the Raspberry Pi Imager to erase the eMMC.

sairon, 2025-01-30:
If you are **really sure** the eMMC can be wiped, either do what @pdonn suggests, or, after connecting to SSH or to the serial console (see my previous posts about that), run `echo 'label: gpt' | sfdisk --force /dev/mmcblk0` - that will create a new empty partition table on the eMMC, which will prevent access to the partitions there. Just in case, save the output of `sfdisk --dump /dev/mmcblk0`, that could be eventually used to restore the partition table to the correct state, it will be just a bit more complex.

edhahn4444, 2025-01-30:
I can confirm that wiping the emmc worked.

Steps I took,
1. Full backup and off of local storage
2. confirmed I am using the nvme not emmc
3. rpi boot following these instructions [https://yellow.home-assistant.io/guides/reinstall-os/#:~:text=Option%202%3A%20Reinstall%20Home%20Assistant%20OS%20using%20rpiboot]
4. rip imager but instead of selecting an image I used the erase option.
5. reset jumpers to original position
6. booted yellow back up
7. installed update
8. on the reboot it updated successfully!

mihaicristianpetrescu, 2025-01-30:
@edhahn4444 What did you use to validate that you are using the nvme and not emmc (step 2) ?

edhahn4444, 2025-01-30:
There is probably a better way but I,
Went to setting
System
Under storage I saw 432 GB free, way more than my emmc
Clicked storage and tried to move data disk
it said no suitable storage found so I made an assumption.

mihaicristianpetrescu, 2025-01-30:
Gotcha! I did the same thing initially when I did the setup for the Yellow + NVME.
Question is...do all people with Yellow + NVME have to go thorough this horrendous steps in order to make 14.2 OS update work? Makes no sense...I urge for fix on the OS level so we don't all have a broken up update and have to go through this hell...

Andware, 2025-01-30:
The problem is that we use the cm5 module. 
I already had the problem back when I tried to update from 14.0 to 14.1. I solved then this problem by installing a cm4 module and updating to a newer version and then switching back to the cm5 module. 

It's just not a long-term solution

mihaicristianpetrescu, 2025-01-30:
Oups...sorry, I've misread the title it seems. I have CM4 actually. So it shouldn't be a problem on CM4 I believe? It's strictly related to CM5 combined with Yellow? 🤔

Andware, 2025-01-30:
@pdonn 
Not correctly in my case.
I had installed HA on CM5 emmc,  bought later a ssd disk installed it and moved the data partition on SSD disk, Not more. Så that is an standart function in HA

mihaicristianpetrescu, 2025-01-30:
> I can confirm that wiping the emmc worked.
> 
> Steps I took,
> 
> 1. Full backup and off of local storage
> 2. confirmed I am using the nvme not emmc
> 3. rpi boot following these instructions https://yellow.home-assistant.io/guides/reinstall-os/#:~:text=Option%202%3A%20Reinstall%20Home%20Assistant%20OS%20using%20rpiboot
> 4. rpi imager but instead of selecting an image I used the erase option for only the emmc.
> 5. reset jumpers to original position
> 6. booted yellow back up
> 7. installed update
> 8. on the reboot it updated successfully!

I have also used the option to move the data to the NVME SSD, so I guess the emmc is still running the HA OS and not the SSD.
I've done the update of OS from 14.1 to 14.2 and HA web interface is not booting up even though the docker container is running as I'm able to access it via SSH root.

Not sure what to do now thought...from your input it seems like I might have to wipe the emmc, but that means that the SSD would only have the data and not the OS...so not sure again what to do next.

edhahn4444, 2025-01-30:
I would start by making sure you are accessing it by ip and port to make
sure it is not a dns thing.

I am by no means an expert but my next move would be to wipe everything and
reset up as if I had just installed the cm5 and restore from the backup.
It is weird that you can ssh in and see the docker instance running so
there is probably a better option.

mihaicristianpetrescu, 2025-01-30:
@edhahn4444 I actually did the wipe on the CM4 eMMC and now it doesn't boot anything up...so it seems that just DATA was migrated to the SSD, OS was left on eMMC, however after update to OS 14.2 it didn't boot up anymore but SSH root was working.
Now SSH root is not working anymore after wiping the eMMC so I assume I just have to start from scratch and reinstall HA on SSD directly somehow

edhahn4444, 2025-01-30:
I only started having the issues after upgrading to cm5.  I am not sure if
our issues were the same.  The advantage is you have a cm4 which is easier
to recover.  You can follow the set up instructions here to start over
https://yellow.home-assistant.io/

mihaicristianpetrescu, 2025-01-30:
Both yes and no...
I wiped the eMMC, didn't boot up, now I've reinstalled with using an external enclosure directly on the SSD the HA, put the SSD back into HA Yellow and have been waiting for the past 10 minutes to be able to connect to its IP address and restore from backup...
Hopefully this means that now HA is fully installed on the SSD and if I want to ever migrate to CM5 it will be as easy as swaping the CM4 with CM5 card.

mihaicristianpetrescu, 2025-01-31:
In the end the resolution was to wipe the eMMC, download HA OS 14.1, use balena etcher to flash it directly to the NVMe drive (via SSD enclosure), install it back and then boot it up + restore from backup.

CopyPasteHero, 2025-02-02:
> I can confirm that wiping the emmc worked.
>
> Steps I took,
> 1. Full backup and off of local storage
> 2. confirmed I am using the nvme not emmc
> 3. rpi boot following these instructions [...]

These steps worked for me. Thanks @edhahn4444 !

sairon, 2025-03-28:
The original issue was reported as resolved.

closed 2025-03-28 by sairon