home-assistant/operating-system #3849: HA green - Cannot connect to the network following the 14.2 upgrade

shaiger, 2025-01-31:
### Describe the issue you are experiencing

HA green with an external SSD.
Core version: 2025.1.3
I have local CLI access (keyboard + monitor). 
In the Banner plot there is no ipv4 IP, when I set one it gets erased upon reboot. 
I also get a very frequent  "Out of memory: Killed process #### (python3)" message, as in every few minutes. 
I am not a Linux expert unfortunately, but definitely capable of following instructions for providing more debug info.

### What operating system image do you use?

green (Home Assistant Green)

### What version of Home Assistant Operating System is installed?

14.2

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

USB: SkyConnect dongle, SSD drive (and the keyboard)

### Steps to reproduce the issue

1. I've upgraded to 14.2
2. The UI loaded but then at some point has stopped to respond 

### Anything in the Supervisor logs that might be useful for us?

```txt
I do not have UI access, and don't know ho to export the logs. I can take a photo of the screen if that would help.
There is something about rate limit exceeded for async io, but other than that my untrained eye can't spot much drama.
```

### Anything in the Host logs that might be useful for us?

```txt
I do not have UI access, and don't know ho to export the logs. I can take a photo of the screen if that would help.
The most prominent message I see:
[resolver] failed to query external DNS server
```

### System information

I do not have UI access, and don't know ho to export data. I can take a photo of the screen if that would help.

sairon, 2025-01-31:
It's not a known OS 14.2 issue and ethernet works fine for me, so there's something specific to your setup. Have you tried if reverting to the previous version (was it 14.1?) fixes the issue? You can revert easily to the previous version from the CLI with `ha os boot-slot other`. 

> In the Banner plot there is no ipv4 IP, when I set one it gets erased upon reboot.

When connected over USB, you can type `login` to enter the root shell and then run `journalctl -f -eu NetworkManager`. With that running, unplug the network cable for a while, plug it back and send a photo of the output. Press Ctrl+C to exit.

> I also get a very frequent "Out of memory: Killed process #### (python3)" message, as in every few minutes. I am not a Linux expert unfortunately, but definitely capable of following instructions for providing more debug info.

That's definitely not right. It _shouldn't_ be an OS regression though (that can be again confirmed by reverting the OS version). How often does it appear? You can watch output of `docker stats` to see what container uses most of the memory and if you see steady increase in some of them.

shaiger, 2025-01-31:
Thank you for your guidance!

**journalctl -f -eu NetworkManager**
[image]

docker stats:
It takes ~20 minutes to run out of memory. The homeassistant container is the hog. 
[image]

**ha os boot-slot other**
I switched to 14.0, and indeed the situation persists. But perhaps something already got irreversibly corrupted?

closed 2025-05-11 by github-actions[bot]