home-assistant/operating-system #3851: energy dashboard shows incorrect values.

Devil-67, 2025-01-31:
### Describe the issue you are experiencing

energy dashboard shows incorrect values.
In the energy dashboard, the power from the battery is also displayed as the consumption of the house load.
However, this is incorrect as there are conversion losses from low voltage to mains voltage.
Take a look at the screenshot, here you can see that 3kWh was drawn from the battery and 0.1kWh from the grid and this value is also displayed as the house load.
In the second screenshot you can see that the house load should only have been 1.8kWh, this value is correct and comes directly from the inverter.

Translated with DeepL.com (free version)

[image]

[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

14.1

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Raspberry 4

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
nein
```

### Anything in the Host logs that might be useful for us?

```txt
nein
```

michagro, 2025-02-02:
My Home Assistant 14.2 has the same problem with the latest patch Level.

closed 2025-02-03 by github-actions[bot]

Devil-67, 2025-02-15:
ok, I have now posted it here:

[https://github.com/home-assistant/core/issues/137266](url)

@michagro have you been able to solve the problem in the meantime