home-assistant/operating-system #3853: Network issues HA yellow after upgrade to CM5

memphi2, 2025-01-31:
### Describe the issue you are experiencing

I've updated my HA yellow yesterday from CM4 to CM5. So far so good, but I'm facing some weird network issues in different integrations.

-I’ve laggy RTSP Streams with the CM5 and Ethernet connected with 1Gbit. When I switch back! to 100MBit the streams are fine. Tested with Unifi, Cisco an Netgear Switches. No Problems with the CM4 at all.

-With the CM5 installed it seems not all UDP commands are send/lost when the destination is in a different VLAN. So my Anel devices do not respond. When HA and the device are in the same VLAN, no problem. Check it with Wireshark.
This is again not the case with the CM4. Tested it in different network setups.

-Telnet seems to be corrupted too (NETIO)

I only have one HA yellow (POE) and only one CM5, so I’m not sure wheather the HA yellow or the CM5 is the problem.

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

14.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Upgraded a Yellow from CM4 2GB to CM5 8GB

### Steps to reproduce the issue

1.Upgrade the CM to CM5
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
nothing useful
```

### Anything in the Host logs that might be useful for us?

```txt
nothing useful
```

sairon, 2025-02-04:
Unfortunately, I can't see anything like that on my end to look into it further. I assume you're running the install fully from NVMe? Would you be able to put up a minimal reproduction scenario?

memphi2, 2025-02-07:
Yes, my original installation ran via NVME. But I then did a reinstall on the internal flash of the CM5 (this was indeed a challenge). I then installed only one integration (Unifi Protect). Both in the app and in the browser on the desktop, the video streams no longer ran properly (stuttering, time delay). But further tests with devices in other VLANs (Phillips HUE) were also negative. A massive packetloss can be observed. This improves when the devices are in the same VLAN as the HA yelllow, but packetloss still exists, even if it is not as massive. 
As a test, I also connected the yellow to a dumb switch once, but that didn't help either.
For further testing I ordered a CM5 Dev board. Then perhaps I can say whether the problem is with the CM5 itself or with the yellow board.

closed 2025-05-16 by github-actions[bot]