home-assistant/operating-system #3855: After latest OS upgrade I'm stuck in a reboot loop

dirkhh, 2025-02-03:
### Describe the issue you are experiencing

I triggered the latest Home Assistant OS upgrade from the UI and the system didn't come back up. So I grabbed the RPi4b from the basement and connected it to a monitor. It shows the expected UBoot and then kernel boot messages
When it reaches the 'Starting Docker Application Container Engine' stage things go wrong.
After sitting there for a few seconds I get
```
[FAILED] Failed to start Docker Application Container Engiene
See 'systemctl status docker.service' for details
[DEPEND] Dependency failed for HassOS supervisor
```
and then it shuts down and reboots

I tried copying the OS to a fresh SD card - no change
I tried switching out the RPi4 board for a different one - no change

Not sure what to try next. Searching for a 'safe boot' option didn't find anything online

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

14.2

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

RPi4b

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
can't get to the logs - I have a video of partial boot log - transcribed the interesting bit above
```

### Anything in the Host logs that might be useful for us?

```txt
can't get to the logs - I have a video of partial boot log - transcribed the interesting bit above
```

sairon, 2025-02-04:
Unless the boot failure is caused by some data corruption of the Docker data, hard-resetting (pulling the power) three times in total should boot the old slot back (the currently booted slot is shown in one of the U-Boot messages). Actually, it should also revert to that when it's in the boot loop - only 3 attempts of the failing slot should be attempted then it should boot into the old version.

If the actual OS version is not the cause but it's data corruption, there are still ways, but I can't give you an universal answer. Let's see where this goes first.

dirkhh, 2025-02-04:
I let it boot about ten times in a row to get the video to read the log messages
I'm 99% sure the corruption is indeed in docker somewhere

cristianburrini, 2025-02-24:
I have the same problem and this happen in both booting slots so the system is stuck in a reboot loop.

dirkhh, 2025-02-24:
In my case it ended up being a corrupted file system. That affected components of docker (hence the reboot loop as starting docker failed), but also my backups. In the ended I ended up with a complete fresh install from scratch. Not my preferred outcome.
I am happy to see the remote backup options that were recently added (and am using them) :)