home-assistant/operating-system #3861: HAOS: Stuck in preparing - no supervisor internet connection

HrKnuddel, 2025-02-04:
### Describe the issue you are experiencing

I want to do a fresh install of HAOS to implement it into my smart home.
After I wrote it onto the Disk and boot from it, you get the info to finish the setup via the webinterface.
The Webinterface says: 
```
Preparing Home Assistant

This may take 20 minutes or more
```

With a loading bar that loads since (at the time of writing) one day
The supervisor Logs tells me, that the Host system is not connected to the internet which is proven to be false (e.g. via curl in the console)
With a bit of analysis I see, that the system is make DNS requests for `checkonline.home-assistant.io`
When i open this site one of two things happen: Either it gets the error that Discourse is offline with message `SSL handshake failed` or it gets the error that Discourse is offline with the message `Connection timed out`. This is reproduce able from multiple Devices and Locations (diffrent public IP-Addresses)
I have a feeling that there might be a problem with the Discourse. And maybe event not with the HAOS Image, but  i can't install HAOS with these problems
NTP is not the Issue. DNS is also not the issue. Connection to the internet is there but HAOS chooses to not have an internet connection

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

14.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Dell Optiplex 3050 SFF with a 128G Drive.
No USB Drives used

### Steps to reproduce the issue

For the Installation Problbems: 
1. Install on a new System
2. Open the WebGUI
3. Look at the logs

For the Discourse Problem:
1. Open `https://checkonline.home-assistant.io/` in any Webbrowser
2. Wait until the website loaded
3. Look at the Error

### Anything in the Supervisor logs that might be useful for us?

```txt
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/udev.sh
[17:25:48] INFO: Using udev information from host
cont-init: info: /etc/cont-init.d/udev.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun supervisor (no readiness notification)
services-up: info: copying legacy longrun watchdog (no readiness notification)
[17:25:48] INFO: Starting local supervisor watchdog...
[... 129 lines cut ...]
2025-02-04 17:29:25.786 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retrying in 30sec
```

### Anything in the Host logs that might be useful for us?

```txt
I do not have the possibility to copy paste the host logs here because I can not access the Webinterface for that
But I added a screenshot in the "Additional information" Section
```

### System information

I do not have Access to the linked page because homeassistant is not setup. But i can say that the `Home Assistant observer` says the following:

Supervisor: Connected
Supported: Supported
Healthy: Healthy

### Additional information

The Last two Lines from the supervisor logs are going to repeat themselves every 30 Seconds but i only added them once to keep the logs clean

Screenshot for the Host Logs:

[image]

sairon, 2025-02-10:
Checking only the domain wouldn't work, the full URL that is used for the connectivity check is `http://checkonline.home-assistant.io/online.txt`. Can you access that from the OS? Check it by typing `login` in the host console and then `curl -v http://checkonline.home-assistant.io/online.txt`.

HrKnuddel, 2025-02-10:
The request worked and I got a HTTP 200 back from the server

HrKnuddel, 2025-02-22:
I just tested it with OS version 14.1 and there is the same problem.

HrKnuddel, 2025-02-23:
I started to investigate further because i didn't get an answer and because i also couldn't open github on the Try ubuntu install for the installation. After some time I reset the BIOS of the PC and the reboot after it started working just like it should work. So I think there was a wrong config in the BIOS because i got the PC not fully reset and i had to change some things back from the Windows installed before. 
That sadly doesn't explain some of the problems, but I am happy, that it works now

closed 2025-02-23 by HrKnuddel