home-assistant/operating-system #3870: Raspberry 5 image won't boot.

ThomasHej, 2025-02-10:
### Feedback

I spend several hours trying to image and re-image the SD card, just to finally start looking for problems with the imagefile itself.  I found several sources claiming that the Raspberry 5 / HA OS 14.2 image is faulty and it can't boot and produce the web-page unattended.  The solution according the to articles is to find an older image (I used a 12.3 image) and then upgrade after installation.

I can confirm that the image can't boot unattended on a brand new Raspberry 5, and no matter what I tried it was stuck somewhere.  Getting the older image (HA 12.3) and flashing it using another tool (Etcher), worked like a charm.  Upgrade to 14.2 thereafter also worked.

Please remove the 14.2 image from the list in the Official Raspberry Imager (at least the Raspberry 5 image) so others wont get the impression that HA is broken from the start...

Thomas

### URL

https://www.home-assistant.io/installation/raspberrypi/

### Version

2025.2.1

sairon, 2025-02-10:
I can not reproduce the issue on my end. In the past there was an issue when the image was created from Windows (see #3437 and #3497) but that should be fixed now. What's the stepping of the SoC (the second line on the CPU will end either with `C1T` or `D0T`) and the RAM size? Can you attach an HDMI display and check what does it show? Also, what exact model of SD card are you using, have you tried a different one?

lattice0, 2025-02-12:
Same here. Raspberry pi 5, 4gb, written with dd, tried several SD cards with 14.2 (latest image). Raspberry Pi boots with HDMI, then I insert the SD and it boots real quick and shows a console for 0.3 seconds then goes blank. Raspberry Pi appears on the network as homeassistant but I cannot access its page. C1T model.

sairon, 2025-02-14:
@lattice0 I am not sure it's the same problem - if it appears on the network with its hostname, it means most of the system has started. Can you ping its IP and access it at `http://IP_ADDRESS:4357`?

lattice0, 2025-02-14:
@sairon I wasn t able to ping or access it. Ialso tried images 14.1 and 14.0 and they had the same problem

La-te, 2025-02-15:
I tried updating my Raspberry Pi 5 from 14.1 to 14.2. After first attempt it booted to 14.1. Then couple of days later I noticed there is again/still update available from 14.1 to 14.2 and proceeded. However, now the device stayed offline after the reboot. I verified from the router, network and arp as well that it really is offline. Next I'll take a 600 km drive to solve this :) 

There are a couple of issues reported with 14.2 and unable to boot or update. There seems to be something wrong.

marine-ecologist, 2025-02-19:
First time install of Home-Assistant iOS, persisted for hours until I came across this thread. 
Raspberry pi 5, 4gb, tried several SD cards with 14.2, same issue with 14.1 (image created on MacOS 15.1.1 not Windows). 
Not appearing on network.

IvanTC95, 2025-02-24:
Same here. Tried 14.2 and then 14.1. Same result with both.

Not sure if this is related, but when using any of the images from github, it asks if I want to edit the config, but it doesn't ask using the option in the Imager selector.

franfabrizio, 2025-03-04:
Can confirm same/similar issue. Brand new Pi 5 4GB, brand new 64GB microSD flashed with HOME ASSISTANT OS 14.2 image from the Raspberry Pi imager. I saw it get the DHCP lease, I could ping and scan the device on the network, but no web interface - port 8123 not open on the device. Tried booting several more times with same results. Hooked up to a monitor and rebooted again, it seems to get all the way to the ha> command line but still no web interface. Re-imaged the card. This time, it warned that Home Assistant CLI not starting! Jump into emergency console..." Rebooted that image a second time, and this time it booted successfully, and is responding on 8123! So I let it finish the initial Preparing Home Assistant... and then rebooted the Pi again to see if it would come back up. It did. So it seems that if you can get it to come up once, you can proceed, but definitely does not reliably boot the 14.2 image from the Pi Imager repository.

jaf69, 2025-03-29:
Hi everybody,
I was in the same situation tha you guys and I think than I’ve detected what is the way to walk around this problem. It’s very easy you could take out rpi5 from the case. This was my situation after more than 10 times intending upgrade from 13.2 to 15 HAOS because I’ve installed an SSD I’ve opened my RPI5 installed an SD Card with raspbian os at same time I’ve unplugged my SSD, after upgrade eeprom I plug again SSD without SD Card and without the case, all run perfectly. After that I installed the case and my rpi5 was blocked again, I release it from the case and again the rpi5 start perfectly without problem. Could someone test this?

closed 2025-07-06 by github-actions[bot]