home-assistant/operating-system #3875: Update from 14.1 to 14.2 looks successful but does not boot to 14.2

brianjmurrell, 2025-02-12:
### Describe the issue you are experiencing

I have tried the update several times and every time it just boots back to 14.1.  It seems to go through the motions but ultimately just doesn't update.

Starting state:

```
# ha os info
board: generic-x86-64
boot: A
boot_slots:
  A:
    state: booted
    status: good
    version: "14.1"
  B:
    state: inactive
    status: good
    version: null
data_disk: Generic-STORAGE-DEVICE-000000000206
update_available: true
version: "14.1"
version_latest: "14.2"
```

But after the restart, nothing has changed, still on 14.1.

```
# uptime
 13:06:46 up 4 min,  0 user,  load average: 4.14, 2.06, 0.83
# ha os info
board: generic-x86-64
boot: A
boot_slots:
  A:
    state: booted
    status: good
    version: "14.1"
  B:
    state: inactive
    status: good
    version: null
data_disk: Generic-STORAGE-DEVICE-000000000206
update_available: true
version: "14.1"
version_latest: "14.2"
```

I have gone through this motion several times and it's the same every time.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

14.1

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Generic x86_64 with a:

```
    ID_BUS: usb
    ID_MODEL: Bluetooth_Radio
    ID_MODEL_ENC: Bluetooth\x20Radio
[... 303 lines cut ...]
version_api | ok
installed_addons | go2rtc (1.9.8), Terminal & SSH (9.16.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 7
-- | --
resources | 2
views | 10
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | January 25, 2025 at 8:53 PM
-- | --
current_recorder_run | February 12, 2025 at 8:08 AM
estimated_db_size | 148.08 MiB
database_engine | sqlite
database_version | 3.47.1

</details>

GHBLoos, 2025-02-13:
Same here. 
But I have slightly different OS info (I never checked this info before). Both boot slots are ```bad```. 

```➜  ~ ha os info             
board: rpi5-64
boot: B
boot_slots:
  A:
    state: inactive
    status: bad
    version: "14.2"
  B:
    state: booted
    status: bad
    version: "14.1"
data_disk: Patriot-M.2-P310-240GB-P310HHBB241008002181
update_available: true
version: "14.1"
version_latest: "14.2"```

yognspoon, 2025-03-04:
Also, same here on my Raspberry Pi5.  Refuses to boot after update.

Like others, a cold restart seems to force a roll-back to 14.1.

davenelson, 2025-03-06:
Also for me on Raspberry Pi 5, requires turning the Pi off and back on to reboot with no update applied. I am going from 13.2 to 14.2.

bskitmor, 2025-03-14:
Same here -- after attempting to go from 14.1 to 14.2 HA would not reconnect via Chrome on WIndows 11.  Had to restart the Pi5 a couple of times to get HA back online, once it came back 14.1 showed as the HA OS so upgrade to 14.2 seems to have been inoperative.

brianjmurrell, 2025-03-22:
> Same here. But I have slightly different OS info (I never checked this info before). Both boot slots are `bad`.
> 
> ```
> board: rpi5-64
> boot: B
> boot_slots:
>   A:
>     state: inactive
>     status: bad
>     version: "14.2"
>   B:
>     state: booted
>     status: bad
>     version: "14.1"
> data_disk: Patriot-M.2-P310-240GB-P310HHBB241008002181
> update_available: true
> version: "14.1"
> version_latest: "14.2"```
> ```

This looks like a different issue.  Please open a new ticket and delete this comment so that it does not muddy up the original issue.

brianjmurrell, 2025-03-22:
I have this same problem after updating from 14.1 (because the 14.2 update failed as detailed above) to 15.0 also:

```
# ha os info
board: generic-x86-64
boot: A
boot_slots:
  A:
    state: booted
    status: good
    version: "14.1"
  B:
    state: inactive
    status: good
    version: null
data_disk: Generic-STORAGE-DEVICE-000000000206
update_available: true
version: "14.1"
version_latest: "15.0"
```

`ha os boot slot B` still does not boot 15.0 but if I simply reboot and interrupt GRUB and request Slot B it seems to successfully boot 15.0.  But on a reboot from that 15.0 boot on Slot B, it reverts back to Slot A on 14.1.

Something seems to have gone terribly sideways on 14.1+.  There are many people complaining about upgrades >14.0 but no suggestions anywhere on how to fix it.  I switched from a native python install of HA to HAOS on the belief that it would be a more stable platform but that does not seem to be playing out.

GHBLoos, 2025-03-28:
I used 'login' in SSH, but I don't know the credentials (or at least the credentials I wrote down don't work). 
From SSH `grub-editenv `is not a command
I cannot find `/mnt/ `(`/mnt/boot/EFI/BOOT/grubenv`)
It seems I cannot give the information.

GHBLoos, 2025-03-28:
> [@GHBLoos](https://github.com/GHBLoos) I was reacting to [@brianjmurrell](https://github.com/brianjmurrell). Your post has been marked as off-topic for a reason.

I deleted the comment. I will delete this comment too in a while

brianjmurrell, 2025-03-28:
> It sounds like something's wrong with the GRUB environment. When you're in the OS shell (i.e. through developer SSH or connected directly to the PC and typed `login`), can you check output of the following command when you boot into the different slots?
> 
> ```
> grub-editenv /mnt/boot/EFI/BOOT/grubenv list
> ```

I hope this is what you are looking for:
```
$ ssh homeassistant.local
X11 forwarding request failed on channel 0
Welcome to Home Assistant OS.

Use `ha` to access the Home Assistant CLI.
# cat /etc/os-release 
NAME="Home Assistant OS"
VERSION="14.1 (Generic x86-64)"
ID=haos
VERSION_ID=14.1
PRETTY_NAME="Home Assistant OS 14.1"
CPE_NAME=cpe:2.3:o:home-assistant:haos:14.1:*:production:*:*:*:generic-x86-64:*
HOME_URL=https://hass.io/
VARIANT="Home Assistant OS Generic x86-64"
VARIANT_ID=generic-x86-64
SUPERVISOR_MACHINE=generic-x86-64
SUPERVISOR_ARCH=amd64
# grub-editenv /mnt/boot/EFI/BOOT/grubenv list
MACHINE_ID=c17bf6c278974ee9b03f685234f7329d
A_OK=1
A_TRY=0
B_OK=1
B_TRY=0
ORDER=A B
# ha os info
board: generic-x86-64
boot: A
boot_slots:
  A:
    state: booted
    status: good
    version: "14.1"
  B:
    state: inactive
    status: good
    version: null
data_disk: Generic-STORAGE-DEVICE-000000000206
update_available: true
version: "14.1"
version_latest: "15.0"
# ha os boot-slot B
Post "http://supervisor/os/boot-slot": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
…
```

But it did reboot despite the timeout error posting.

But it rebooted to slot A again:

```
$ ssh homeassistant.local
X11 forwarding request failed on channel 0
Welcome to Home Assistant OS.

Use `ha` to access the Home Assistant CLI.
# cat /etc/os-release 
NAME="Home Assistant OS"
VERSION="14.1 (Generic x86-64)"
ID=haos
VERSION_ID=14.1
PRETTY_NAME="Home Assistant OS 14.1"
CPE_NAME=cpe:2.3:o:home-assistant:haos:14.1:*:production:*:*:*:generic-x86-64:*
HOME_URL=https://hass.io/
VARIANT="Home Assistant OS Generic x86-64"
VARIANT_ID=generic-x86-64
SUPERVISOR_MACHINE=generic-x86-64
SUPERVISOR_ARCH=amd64
# ha os info
board: generic-x86-64
boot: A
boot_slots:
  A:
    state: booted
    status: good
    version: "14.1"
  B:
    state: inactive
    status: good
    version: null
data_disk: Generic-STORAGE-DEVICE-000000000206
update_available: true
version: "14.1"
version_latest: "15.0"
# ha os boot-slot B
Post "http://supervisor/os/boot-slot": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
…
```

But this time I interrupted GRUB on the reboot and forced it to boot slot B.  It was going to boot slot A again.

```
$ ssh homeassistant.local
X11 forwarding request failed on channel 0
Welcome to Home Assistant OS.

Use `ha` to access the Home Assistant CLI.
# cat /etc/os-release 
NAME="Home Assistant OS"
VERSION="15.0 (Generic x86-64)"
ID=haos
VERSION_ID=15.0
PRETTY_NAME="Home Assistant OS 15.0"
CPE_NAME=cpe:2.3:o:home-assistant:haos:15.0:*:production:*:*:*:generic-x86-64:*
HOME_URL=https://hass.io/
VARIANT="Home Assistant OS Generic x86-64"
VARIANT_ID=generic-x86-64
SUPERVISOR_MACHINE=generic-x86-64
SUPERVISOR_ARCH=amd64
# grub-editenv /mnt/boot/EFI/BOOT/grubenv list
MACHINE_ID=6bac9abe93344614b782694ab02594b4
A_OK=1
A_TRY=0
B_OK=1
B_TRY=0
ORDER=B A
# ha os info
board: generic-x86-64
boot: B
boot_slots:
  A:
    state: inactive
    status: good
    version: "14.1"
  B:
    state: booted
    status: good
    version: null
data_disk: Generic-STORAGE-DEVICE-000000000206
update_available: false
version: "15.0"
version_latest: "15.0"
```

So, successfully booted on 15.0 on slot B, but if I reboot again, it will go back to 14.1 on slot A:

```
# ha host reboot
Post "http://supervisor/host/reboot": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
…
$ ssh homeassistant.local
X11 forwarding request failed on channel 0
Welcome to Home Assistant OS.

Use `ha` to access the Home Assistant CLI.
# cat /etc/os-release 
NAME="Home Assistant OS"
VERSION="14.1 (Generic x86-64)"
ID=haos
VERSION_ID=14.1
PRETTY_NAME="Home Assistant OS 14.1"
CPE_NAME=cpe:2.3:o:home-assistant:haos:14.1:*:production:*:*:*:generic-x86-64:*
HOME_URL=https://hass.io/
VARIANT="Home Assistant OS Generic x86-64"
VARIANT_ID=generic-x86-64
SUPERVISOR_MACHINE=generic-x86-64
SUPERVISOR_ARCH=amd64
# ha os info
board: generic-x86-64
boot: A
boot_slots:
  A:
    state: booted
    status: good
    version: "14.1"
  B:
    state: inactive
    status: good
    version: null
data_disk: Generic-STORAGE-DEVICE-000000000206
update_available: true
version: "14.1"
version_latest: "15.0"
``` 

> Also, is this a standard install on a UEFI system?

This is not a UEFI system.  I don't have the budget to go out and buy a new more modern (and way more computer than needed) computer just to run HA on.  It is a repurposed older (BIOS) system.

> Just asking because I saw people installing OS on systems supporting only legacy BIOS. In that case it's expected the updates won't work.

Is this something new/recent?  Because prior to the 14.2 OS update, OS updates worked correctly.  And ultimately the 15.0 OS update has also worked, it just doesn't seem to want to boot slot B without manual GRUB menu intervention.  It can't seem to remember that it should boot from the newly upgraded slot.  That sounds like some kind of bug rather than a fundamental flaw.

bskitmor, 2025-04-05:
I took the recommended route and upgraded the RPi 5 eeprom using a spare SD card and the Rasberry WIN 11 imaging app.  After doing that successfully, I was able to go to HA 15.1 with no trouble.  Thanks for the help from the HAS support team

________________________________
From: Brian J. Murrell
Sent: Friday, March 28, 2025 9:47 AM
To: home-assistant/operating-system
Cc: bskitmor; Comment
Subject: Re: [home-assistant/operating-system] Update from 14.1 to 14.2 looks successful but does not boot to 14.2 (Issue #3875)

[brianjmurrell]brianjmurrell left a comment (home-assistant/operating-system#3875)<https://github.com/home-assistant/operating-system/issues/3875#issuecomment-2761902457>

It sounds like something's wrong with the GRUB environment. When you're in the OS shell (i.e. through developer SSH or connected directly to the PC and typed login), can you check output of the following command when you boot into the different slots?

grub-editenv /mnt/boot/EFI/BOOT/grubenv list

I hope this is what you are looking for:

$ ssh homeassistant.local
X11 forwarding request failed on channel 0
Welcome to Home Assistant OS.

Use `ha` to access the Home Assistant CLI.
# cat /etc/os-release
NAME="Home Assistant OS"
VERSION="14.1 (Generic x86-64)"
ID=haos
VERSION_ID=14.1
PRETTY_NAME="Home Assistant OS 14.1"
CPE_NAME=cpe:2.3:o:home-assistant:haos:14.1:*:production:*:*:*:generic-x86-64:*
HOME_URL=https://hass.io/
VARIANT="Home Assistant OS Generic x86-64"
VARIANT_ID=generic-x86-64
SUPERVISOR_MACHINE=generic-x86-64
SUPERVISOR_ARCH=amd64
# grub-editenv /mnt/boot/EFI/BOOT/grubenv list
MACHINE_ID=c17bf6c278974ee9b03f685234f7329d
A_OK=1
A_TRY=0
B_OK=1
B_TRY=0
ORDER=A B
# ha os info
board: generic-x86-64
boot: A
boot_slots:
  A:
    state: booted
    status: good
    version: "14.1"
  B:
    state: inactive
    status: good
    version: null
data_disk: Generic-STORAGE-DEVICE-000000000206
update_available: true
version: "14.1"
version_latest: "15.0"
# ha os boot-slot B
Post "http://supervisor/os/boot-slot": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
…

But it did reboot despite the timeout error posting.

But it rebooted to slot A again:

$ ssh homeassistant.local
X11 forwarding request failed on channel 0
Welcome to Home Assistant OS.

Use `ha` to access the Home Assistant CLI.
# cat /etc/os-release
NAME="Home Assistant OS"
VERSION="14.1 (Generic x86-64)"
ID=haos
VERSION_ID=14.1
PRETTY_NAME="Home Assistant OS 14.1"
CPE_NAME=cpe:2.3:o:home-assistant:haos:14.1:*:production:*:*:*:generic-x86-64:*
HOME_URL=https://hass.io/
VARIANT="Home Assistant OS Generic x86-64"
VARIANT_ID=generic-x86-64
SUPERVISOR_MACHINE=generic-x86-64
SUPERVISOR_ARCH=amd64
# ha os info
board: generic-x86-64
boot: A
boot_slots:
  A:
    state: booted
    status: good
    version: "14.1"
  B:
    state: inactive
    status: good
    version: null
data_disk: Generic-STORAGE-DEVICE-000000000206
update_available: true
version: "14.1"
version_latest: "15.0"
# ha os boot-slot B
Post "http://supervisor/os/boot-slot": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
…

But this time I interrupted GRUB on the reboot and forced it to boot slot B. It was going to boot slot A again.

$ ssh homeassistant.local
X11 forwarding request failed on channel 0
Welcome to Home Assistant OS.

Use `ha` to access the Home Assistant CLI.
# cat /etc/os-release
NAME="Home Assistant OS"
VERSION="15.0 (Generic x86-64)"
ID=haos
VERSION_ID=15.0
PRETTY_NAME="Home Assistant OS 15.0"
CPE_NAME=cpe:2.3:o:home-assistant:haos:15.0:*:production:*:*:*:generic-x86-64:*
HOME_URL=https://hass.io/
VARIANT="Home Assistant OS Generic x86-64"
VARIANT_ID=generic-x86-64
SUPERVISOR_MACHINE=generic-x86-64
SUPERVISOR_ARCH=amd64
# grub-editenv /mnt/boot/EFI/BOOT/grubenv list
MACHINE_ID=6bac9abe93344614b782694ab02594b4
A_OK=1
A_TRY=0
B_OK=1
B_TRY=0
ORDER=B A
# ha os info
board: generic-x86-64
boot: B
boot_slots:
  A:
    state: inactive
    status: good
    version: "14.1"
  B:
    state: booted
    status: good
    version: null
data_disk: Generic-STORAGE-DEVICE-000000000206
update_available: false
version: "15.0"
version_latest: "15.0"

So, successfully booted on 15.0 on slot B, but if I reboot again, it will go back to 14.1 on slot A:

# ha host reboot
Post "http://supervisor/host/reboot": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
…
$ ssh homeassistant.local
X11 forwarding request failed on channel 0
Welcome to Home Assistant OS.

Use `ha` to access the Home Assistant CLI.
# cat /etc/os-release
NAME="Home Assistant OS"
VERSION="14.1 (Generic x86-64)"
ID=haos
VERSION_ID=14.1
PRETTY_NAME="Home Assistant OS 14.1"
CPE_NAME=cpe:2.3:o:home-assistant:haos:14.1:*:production:*:*:*:generic-x86-64:*
HOME_URL=https://hass.io/
VARIANT="Home Assistant OS Generic x86-64"
VARIANT_ID=generic-x86-64
SUPERVISOR_MACHINE=generic-x86-64
SUPERVISOR_ARCH=amd64
# ha os info
board: generic-x86-64
boot: A
boot_slots:
  A:
    state: booted
    status: good
    version: "14.1"
  B:
    state: inactive
    status: good
    version: null
data_disk: Generic-STORAGE-DEVICE-000000000206
update_available: true
version: "14.1"
version_latest: "15.0"

Also, is this a standard install on a UEFI system?

This is not a UEFI system. I don't have the budget to go out and buy a new more modern (and way more computer than needed) computer just to run HA on. It is a repurposed older (BIOS) system.

Just asking because I saw people installing OS on systems supporting only legacy BIOS. In that case it's expected the updates won't work.

Is this something new/recent? Because prior to the 14.2 OS update, OS updates worked correctly. And ultimately the 15.0 OS update has also worked, it just doesn't seem to want to boot slot B without manual GRUB menu intervention. It can't seem to remember that it should boot from the newly upgraded slot. That sounds like some kind of bug rather than a fundamental flaw.