home-assistant/operating-system #3879: No way to reset password for owner with Non-English (Korean) username

wooooooooooook, 2025-02-15:
### Describe the issue you are experiencing

When attempting to reset the password for a user with a Korean (or other non-English) username using the `auth reset` command, the system does not accept the username input. This makes it impossible to update the password for such users via the command line.

[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

14.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Running on Proxmox VM
Proxmox running of intel N100 CPU miniPC

### Steps to reproduce the issue

1.Create a user with a Korean name, e.g., 홍길동.
2. Try resetting the password via the command:
`auth reset --username '홍길동' --password 'new_password'`
3. I cannot input '홍길동'

### Anything in the Supervisor logs that might be useful for us?

```txt
.
```

### Anything in the Host logs that might be useful for us?

```txt
.
```

HarryVasanth, 2025-03-18:
Hello @wooooooooooook,

Does the terminal or console used to input the command supports UTF-8 encoding? This may resolve issues with inputting non-ASCII characters.

KevinCathcart, 2025-03-22:
Yeah, this is tricky. The `ha auth` commands only work from the host or from addons with admin level supervisor API access. 

This means the easiest way to run them is from the hardware console. But there are two issues. The first is support for utf-8. In theory the default Linux fbcon can support this, but it is limited to a 256 or 512 character font, which is problematic for some languages like Korean. Even if we ignore that issue, there is no IME support for the default framebuffer console. So I don't see this as especially likely to change, unless the devs are wiling to change the OS to include software like fbterm (which is a user-mode framebuffer console that can replace the kernel's one, which allows full unicode support, and has some support for IMEs), and configure it all correctly, and document how to enable various IMEs.

Other possible solutions might be:  enforcing ACII usernames (while still obviously allowing full Unicode display names), or adding an additional (perhaps numeric) identifier that can seen with list and used with reset. But technically neither of those approaches would be an OS level change, but rather either a core change, or a core+supervisor+cli change.

With the rest of this post, I'm going to document two workarounds that can be used in scenarios like this. Both are actually development features, not intended for end users, but which I have personally verified do work to allow resetting a password in this scenario.

### Workaround 1: host level ssh

The first workaround host level ssh daemon (not to be confused with the ssh addon). [The developer documentation suggests](https://developers.home-assistant.io/docs/operating-system/debugging/) using a USB stick with a specially named partition to do this. That can work, but can be inconvenient or even impossible from a virtual machine, depending on what the hypervisor (and any supporting software) supports. In that case, one can instead do it manually.

<details>
<summary>Setting up host SSH manually</summary>
On the system:
1. At the `ha> ` prompt type `login`. This will get you to a root shell.
2. Now we need to create `/root/.ssh/authorized_keys`, with the correct contents. The last I knew proxmox's novnc lacked the paste as keyboard input feature, so you might need to manually type the whole thing into vi (or into an `echo "">/root/.ssh/authorized_keys` line)? If so, then trying to pass though a USB thumb drive might be easier. 
3. Once the file has been created type `reboot`. After the system reboots, the host ssh server should be running.
</details>

Regardless of which approach you use to activate the host ssh, you can then access it via `ssh root@homeassistant.local -p 22222`, and run commands. I'd expect this to just work from Linux or macOS, as long as you do `ha auth reset ...`, since the ha is required from a normal shell prompt.

I was able to confirm this approach works even from Windows, as long as I used Windows Terminal, although with one slight catch. While running `ha auth list` showed the Korean name just fine, and pasting in `ha auth reset --username '홍길동' --password 'new_password'` worked, the pasted in name did show up as question marks. (Not really sure why the name can display fine in output but not when input, but the important thing is that it works.)

### Workaround 2: supervisor remote api

The second workaround is using the supervisor remote api plugin, along using docker on another computer.One can add the needed store, install, and run the needed addon, and find the needed auth token all from the device's `ha> ` prompt like so:
```
ha> store add https://github.com/home-assistant/addons-development
ha> addons install 77f1785d_remote_api
ha> addons start 77f1785d_remote_api
ha> addons logs 77f1785d_remote_api
```
One of the last lines of output should tell you the token. It is a bit obnoxiously long, but not as bad as an ssh key.

The from the other computer with docker you would run `docker run -it --rm ghcr.io/home-assistant/amd64-hassio-cli /bin/sh`. Then at the `#` prompt you would do the following (obviously replacing  `replace_this_with_remote_api_token` with the api token from the logs):
```sh
export SUPERVISOR_ENDPOINT=http://homeassistant.local
export SUPERVISOR_API_TOKEN=replace_this_with_remote_api_token
ha auth reset --username '홍길동' --password 'new_password'
```

wooooooooooook, 2025-03-25:
> Yeah, this is tricky. The `ha auth` commands only work from the host or from addons with admin level supervisor API access.
> 
> This means the easiest way to run them is from the hardware console. But there are two issues. The first is support for utf-8. In theory the default Linux fbcon can support this, but it is limited to a 256 or 512 character font, which is problematic for some languages like Korean. Even if we ignore that issue, there is no IME support for the default framebuffer console. So I don't see this as especially likely to change, unless the devs are wiling to change the OS to include software like fbterm (which is a user-mode framebuffer console that can replace the kernel's one, which allows full unicode support, and has some support for IMEs), and configure it all correctly, and document how to enable various IMEs.
> 
> Other possible solutions might be: enforcing ACII usernames (while still obviously allowing full Unicode display names), or adding an additional (perhaps numeric) identifier that can seen with list and used with reset. But technically neither of those approaches would be an OS level change, but rather either a core change, or a core+supervisor+cli change.
> 
> With the rest of this post, I'm going to document two workarounds that can be used in scenarios like this. Both are actually development features, not intended for end users, but which I have personally verified do work to allow resetting a password in this scenario.
> 
> ### Workaround 1: host level ssh
> The first workaround is using the host level ssh daemon (not to be confused with the ssh addon). [The developer documentation suggests](https://developers.home-assistant.io/docs/operating-system/debugging/) using a USB stick with a specially named partition to do this. That can work, but can be inconvenient or even impossible from a virtual machine, depending on what the hypervisor (and any supporting software) supports. In that case, one can instead do it manually.
> 
> Setting up host SSH manually
> Regardless of which approach you use to activate the host ssh, you can then access it via `ssh root@homeassistant.local -p 22222`, and run commands. I'd expect this to just work from Linux or macOS, as long as you do `ha auth reset ...`, since the ha is required from a normal shell prompt.
> 
> I was able to confirm this approach works even from Windows, as long as I used Windows Terminal, although with one slight catch. While running `ha auth list` showed the Korean name just fine, and pasting in `ha auth reset --username '홍길동' --password 'new_password'` worked, the pasted in name did show up as question marks. (Not really sure why the name can display fine in output but not when input, but the important thing is that it works.)
> 
> ### Workaround 2: supervisor remote api
> The second workaround is using the supervisor remote api plugin, along using docker on another computer.One can add the needed store, install, and run the needed addon, and find the needed auth token all from the device's `ha> ` prompt like so:
> 
> ```
> ha> store add https://github.com/home-assistant/addons-development
> ha> addons install 77f1785d_remote_api
> ha> addons start 77f1785d_remote_api
> ha> addons logs 77f1785d_remote_api
> ```
> 
> One of the last lines of output should tell you the token. It is a bit obnoxiously long, but not as bad as an ssh key.
> 
> The from the other computer with docker you would run `docker run -it --rm ghcr.io/home-assistant/amd64-hassio-cli /bin/sh`. Then at the `#` prompt you would do the following (obviously replacing `replace_this_with_remote_api_token` with the api token from the logs):
> 
> export SUPERVISOR_ENDPOINT=http://homeassistant.local
> export SUPERVISOR_API_TOKEN=replace_this_with_remote_api_token
> ha auth reset --username '홍길동' --password 'new_password'

Thank you very much. Although I reinstalled HA and restored it using the recovery function, I believe this will be helpful if the same situation occurs again.

closed 2025-03-25 by wooooooooooook