home-assistant/operating-system #3880: Realtek RTL8125 ethernet Lan not supported by HA OS (ASUS NUC 14 essential)

peletonduwer, 2025-02-15:
### Describe the issue you are experiencing

No Lan support on Realtek RTL8125 Lan controller. No IP address is assigned via Ethernet cable because the LAN controller is not supported by HA, driver r8125 is missing. With a USB to Ethernet dongle, an IP address is assigned.  The LAN port Realtek RTL8125 is not recognized by HA. It was a fresh install on an ASUS NUC 14 essential

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

14.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

It was a fresh install on an ASUS NUC 14 essential

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No logs, no ethernet connection
```

### Anything in the Host logs that might be useful for us?

```txt
No logs, no ethernet connection
```

### Additional information

[image]

sairon, 2025-02-15:
`r8125` is an out-of-tree driver from Realtek, RTL8125 should be supported in mainline `r8169` which is built-in for generic-x86-64. Can you type `login` and check `dmesg` why the card is not initialized?

NiVo28, 2025-02-15:
> `r8125` is an out-of-tree driver from Realtek, RTL8125 should be supported in mainline `r8169` which is built-in for generic-x86-64. Can you type `login` and check `dmesg` why the card is not initialized?

I have the same problem. I installed HA OS.
The commands 'login' and 'dmesg' doesn't work if I boot Ubuntu from USB. Or are these HA commands?
"login: Cannot possibly work without effective root"

My LAN-adapter doesn't work in a fresh Ubuntu 22.04.
When I run lspci -vv, it shows me the mainline r8169. Only as "Kernel modules: r8169", but it doesn't show up as "Kernal driver in use".
When I install the r8125 driver and activate it through modprobe, then my LAN-adapter works. But when installing HA OS, it doesn't seem to recognize my LAN hardware.
The only way to have HA OS work with a wired internet connection is through a ethernet-usb adapter and connect it to my PC's USB-port. Which is a dirty workaround.

jdg2025, 2025-02-25:
I'm having the same issue. Using a Dell Wyse 5070 ThinClient, I've installed both Ubuntu 24.04 and HOAS on two different internal drives, and regardless of which I boot to, the Realtek RTL8111/8168/8211/8411 PCI Express Gigabit Ethernet Controller (rev 15) won't work. Without an internet connection, all the various advice about downloading and installing different drivers doesn't help because there is a litany of capabilities not installed in ubuntu from the USB drive installer so anything I try attempts to connect to the internet to download screen after screen worth of items not installed that it needs. I'm trying the same with HOAS, but similar issues. Though lspci -vv does show Kernal driver in use: r8169

Hagren, 2025-02-28:
I ran in to the same problem with my new ASUS NUC 14 Essential. In Ubuntu 24.04 I ran 'lshw -C network' and the Realtek NIC shows up as unclaimed, so this is not an issue specific to HAOS as it is not recognised by Ubuntu either. If it should be picked up r8169 module this does not work for some reason and I have mainly used Intel NICs with Linux previously . I have downloaded the r8125 driver from Github and it works fine, but I don't know if I even have the option to install it in HAOS. I have temporarily solved my problem by running Ubuntu with VirtualBox and installing HAOS as a guest on VirtualBox. Performance is more than enough for now but I would prefer to run on bare metal as it is intended as a dedicated box for HAOS.

eggerjo1, 2025-03-11:
I have the same behaviour on a new ASUS NUC 14 Essential.
On another ASUS NUC 14 Essential unit with Ubuntu 20.02 running Idid the same solve as @Hagren, which is also described here:
https://discourse.ubuntu.com/t/rtl8125-2-5gbe-ethernet-port-not-working-in-ubuntu-24-04/55551`

bazyrush, 2025-03-30:
same issue :(
 
similar to eggerjo i managed to get the NIC working on the NUC (lol) in ubuntu with the drivers here https://www.realtek.com/Download/List?cate_id=584 but it doesn't seem like i can get it working in HAOS.

bazyrush, 2025-03-30:
by the way, is there a way to manually install a specific driver in HAOS? as mentioned, have found the same issue in ubuntu 24.04.2, but after manually installing the official driver it works fine

bazyrush, 2025-04-05:
> `r8125` is an out-of-tree driver from Realtek, RTL8125 should be supported in mainline `r8169` which is built-in for generic-x86-64. Can you type `login` and check `dmesg` why the card is not initialized?

sorry for the delayed response! not sure quite what i'm looking for but it seems not to recognise the controller, with an error "unknown chip XID 688, contact r8169 maintainers"

[image]

eggs-benny-t, 2025-04-09:
I can also confirm this issue.

Ubuntu 24.10 has the same issue, but following the directions at https://github.com/awesometic/realtek-r8125-dkms was a quick and easy fix.

Also, according to https://bugs.launchpad.net/ubuntu/+bug/2100350 Ubuntu 25.04/Plucky, due to be released in a week or so, already has r8125-dkms 9.014. So moving forward these NUC 14 Essentials will work out of the box for Ubuntu users.

I bought mine to run HAOS for me, so if there is any chance of getting these drivers included in HAOS in the future for a similar "works out of the box" experience that would be fantastic, but I'm very new to HA (this was my first attempt to play with it) so I'm not sure if that is a realistic goal or not.

Jassevanloe, 2025-04-09:
I have the same issue, and i tried the Ubuntu 25.04/Plucky beta and can confirm the issue is fixed in that version.

sairon, 2025-04-09:
Using the out-of-tree driver is not the way we want to go in HAOS, as the mainline `r8169` works for all previous Realtek NICs, and replacing it with `r8125` would mean that an alternative driver will be suddenly used there. But it seems the latest mainline contains support for more models, including the "XID 688" (thanks @bazyrush for the screenshot, it helped a lot), so we could eventually backport it here: https://github.com/torvalds/linux/commit/f75d1fbe7809bc5ed134204b920fd9e2fc5db1df

closed 2025-04-09 by sairon

sairon, 2025-04-09:
@cdce8p Thanks for checking it so quickly, and thanks a lot for identifying the missing patches! Patch incoming...

cdce8p, 2025-04-10:
If anyone else wants to test it, the build artifacts can be downloaded from: https://os-artifacts.home-assistant.io/index.html

commit fc984b94 (2025-04-09): Backport RTL8125D (rev C, XID 688) support (#4011)

* Backport RTL8125D (rev C, XID 688) support

Apply mainline patch adding support for NIC present e.g. on ASUS NUC 14
Essential.

Fixes #3880

* Update buildroot to add RTL8125D firmware

* buildroot 4cd211162d...5379c358bf (1):
  > linux-firmware: add RTL8125D firmware