home-assistant/operating-system #3889: Ethernet not connecting after Yellow CM5 upgrade

mheidinger, 2025-02-20:
### Describe the issue you are experiencing

After upgrading my Home Assistant Yellow from the CM4 (with eMMC, without WIFI) to the CM5 (with eMMC, without WIFI) the ethernet interface is no longer working. When connecting to the CM5 via Serial, I see the error `Error: Can't create config and activate end0: Connection 'Supervisor end0' is not available on device end0 because device has no carrier` and also no lights are blinking on the port itself.

The same setup (ethernet cable, router port, OS version, etc.) works after changing back to the CM4.

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

14.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Home Assistant Yellow Rev 1.2 (no PoE) with CM5 (eMMC, no WIFI) and NVME Disk (but also tested without)

### Steps to reproduce the issue

Basically following https://yellow.home-assistant.io/guides/reinstall-os-cm5/ after changing to a CM5.

1. Change Yellow to CM5
2. Run rpiboot
3. Flash Home Assistant with
  - Raspberry Pi Imager (probably 14.2)
  - Rufus (image pulled from CM4 with 14.2, 14.0 directly from GitHub and 14.2 directly from GitHub)
4. Boot
5. No lights on ethernet port + error message `Connection 'Supervisor end0' is not available on device end0 because device has no carrier`

### Anything in the Supervisor logs that might be useful for us?

```txt
# ha supervisor logs
2025-02-20 21:55:49.909 INFO (MainThread) [supervisor.docker.interface] Found ghcr.io/home-assistant/aarch64-hassio-audio versions: [<AwesomeVersion CalVer '2023.12.0'>]
2025-02-20 21:55:49.914 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-audio with version 2023.12.0
2025-02-20 21:55:49.928 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
2025-02-20 21:55:50.350 INFO (MainThread) [supervisor.docker.audio] Starting Audio ghcr.io/home-assistant/aarch64-hassio-audio with version 2023.12.0 - 172.30.32.4
2025-02-20 21:55:50.363 INFO (MainThread) [supervisor.docker.interface] Found ghcr.io/home-assistant/aarch64-hassio-observer versions: [<AwesomeVersion CalVer '2023.06.0'>]
2025-02-20 21:55:50.368 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-observer with version 2023.06.0
2025-02-20 21:55:50.387 INFO (MainThread) [supervisor.plugins.observer] Starting observer plugin
2025-02-20 21:55:50.760 INFO (MainThread) [supervisor.docker.observer] Starting Observer ghcr.io/home-assistant/aarch64-hassio-observer with version 2023.06.0 - 172.30.32.6
2025-02-20 21:55:50.774 INFO (MainThread) [supervisor.docker.interface] Found ghcr.io/home-assistant/aarch64-hassio-multicast versions: [<AwesomeVersion CalVer '2024.03.0'>]
2025-02-20 21:55:50.780 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-multicast with version 2024.03.0
2025-02-20 21:55:50.795 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
2025-02-20 21:55:51.007 INFO (MainThread) [supervisor.docker.multicast] Starting Multicast ghcr.io/home-assistant/aarch64-hassio-multicast with version 2024.03.0 - Host
2025-02-20 21:55:51.008 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 0 Home Assistant secrets
2025-02-20 21:55:51.017 INFO (MainThread) [supervisor.docker.interface] No version found for ghcr.io/home-assistant/yellow-homeassistant
2025-02-20 21:55:51.017 INFO (MainThread) [supervisor.homeassistant.core] No Home Assistant Docker image ghcr.io/home-assistant/yellow-homeassistant found.
2025-02-20 21:55:51.023 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/yellow-homeassistant with version landingpage
2025-02-20 21:55:51.023 INFO (MainThread) [supervisor.homeassistant.core] Using preinstalled landingpage
2025-02-20 21:55:51.031 INFO (MainThread) [supervisor.homeassistant.core] Starting HomeAssistant landingpage
2025-02-20 21:55:51.051 INFO (MainThread) [supervisor.homeassistant.module] Update pulse/client.config: /data/tmp/homeassistant_pulse
2025-02-20 21:55:51.325 INFO (MainThread) [supervisor.docker.homeassistant] Starting Home Assistant ghcr.io/home-assistant/yellow-homeassistant with version landingpage
2025-02-20 21:55:51.371 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 14.2 / BootSlot A
2025-02-20 21:55:51.374 WARNING (MainThread) [supervisor.jobs] 'GitRepo.clone' blocked from execution, no supervisor internet connection
2025-02-20 21:55:51.374 ERROR (MainThread) [supervisor.store] Can't add repository core due to 'GitRepo.clone' blocked from execution, no supervisor internet connection
2025-02-20 21:55:51.374 INFO (MainThread) [supervisor.resolution.module] Create new suggestion execute_reload - store / core
2025-02-20 21:55:51.374 INFO (MainThread) [supervisor.resolution.module] Create new issue fatal_error - store / core
2025-02-20 21:55:51.375 WARNING (MainThread) [supervisor.jobs] 'GitRepo.clone' blocked from execution, no supervisor internet connection
2025-02-20 21:55:51.375 ERROR (MainThread) [supervisor.store] Can't add repository https://github.com/hassio-addons/repository due to 'GitRepo.clone' blocked from execution, no supervisor internet connection
2025-02-20 21:55:51.375 INFO (MainThread) [supervisor.resolution.module] Create new suggestion execute_reload - store / a0d7b954
2025-02-20 21:55:51.375 INFO (MainThread) [supervisor.resolution.module] Create new issue fatal_error - store / a0d7b954
2025-02-20 21:55:51.376 WARNING (MainThread) [supervisor.jobs] 'GitRepo.clone' blocked from execution, no supervisor internet connection
2025-02-20 21:55:51.376 ERROR (MainThread) [supervisor.store] Can't add repository https://github.com/esphome/home-assistant-addon due to 'GitRepo.clone' blocked from execution, no supervisor internet connection
2025-02-20 21:55:51.376 INFO (MainThread) [supervisor.resolution.module] Create new suggestion execute_reload - store / 5c53de3b
[... 189 lines cut ...]
  enabled: false
  interface: end0
  ipv4:
    address: []
    gateway: null
    method: disabled
    nameservers: []
    ready: false
  ipv6:
    address: []
    gateway: null
    method: disabled
    nameservers: []
    ready: false
  mac: 2C:CF:67:DB:2E:82
  primary: false
  type: ethernet
  vlan: null
  wifi: null
supervisor_internet: false
```
```
# ha network update end0
Error: Can't create config and activate end0: Connection 'Supervisor end0' is not available on device end0 because device has no carrier
```

sairon, 2025-02-24:
Although I have not tested HAOS with CM5 without Wi-Fi, there should not be any differences regarding the ethernet functionality. Have you tried re-seating the CM5 module? If it doesn't help, could you try flashing Raspberry Pi OS to confirm it does work there and it's not a hardware fault?

mheidinger, 2025-02-25:
> Have you tried re-seating the CM5 module?

Yes but the seating looks good...

> Could you try flashing Raspberry Pi OS to confirm it does work there and it's not a hardware fault?

Good idea, tried that and the LAN port is still dark :/

I ordered now a generic Development Kit to check whether maybe the CM5 itself is broken. I don't suspect the Yellow to be the issue as it works with the CM4 and apparently works for others with the CM5.

So feel free to close this if you also think that it most probably is not related to the Yellow.

sairon, 2025-03-05:
Assuming hardware fault, I am closing this issue. Feel free to reopen/create a new one if it turns out it's software related.

mheidinger, 2025-03-14:
Want to keep this updated in case someone else has the same issue:

I now received a new PI and am still facing the same issue :/ Will investigate further and post here in case I find something...

mheidinger, 2025-04-01:
Ordered yet another Pi, this time with WIFI on board (debugging the other one without LAN and without a display was too much pain).
First tried it out on a Dev Board: LAN working. Then plugged it into the Yellow: Also working.

So no clue if I just got really bad luck and received two broken modules, got shipped the same broken module twice, destroyed them while I installed them on the Yellow or if there is an issue with the non-WIFI version. Last option still seems unlikely but I also can't prove the opposite.

At last I am happy that its working now :)