home-assistant/operating-system #3904: HA OS (CM5 / Home Assistant Yellow): Network fails to initialize (DHCP not working) at boot if any USB device is connected

Swalabrr, 2025-02-26:
### Describe the issue you are experiencing

**Description**

On Home Assistant Yellow (CM5), any USB device (e.g., a USB serial adapter, a simple USB mouse, or even a USB hub with nothing attached) causes the network to fail to initialize at startup. The system does not receive an IP address via DHCP and remains inaccessible (no ping, no web UI). Removing the USB device before boot allows the system to come up normally and obtain an IP via DHCP.
	•	Attempting a static IP reduces but does not completely eliminate the issue.
	•	This is not specific to a particular USB device—any device plugged into the USB port triggers the same behavior.

**Expected Behavior**

Home Assistant OS should successfully initialize the network and obtain an IP address (via DHCP) even with any USB device connected at boot.

**Request**
Please investigate why any USB device at boot leads to a failure in network initialization (DHCP unreachable). We’d expect the system to bring up the network regardless of USB devices. Thank you!

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

14.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Hardware: HomeAssistant Yellow with CM5 (upgraded). I have no idea if it would behave the same with the original CM4 and at this stage I'm not inclined to try that, unless instructed to do so.

### Steps to reproduce the issue

**Steps to Reproduce**
	1.	Connect any USB device (e.g., USB mouse, empty USB hub) to the Home Assistant Yellow (CM5).
	2.	Ensure DHCP is set for the network interface.
	3.	Power on or reboot the device.
	4.	Result:
	•	The system never obtains an IP address (DHCP fails), logs show network unreachable, and no ping or UI access is possible.
	5.	If you remove the USB device before boot, the system boots normally, obtains a DHCP IP, and is fully accessible.

### Anything in the Supervisor logs that might be useful for us?

```txt
**Relevant Logs**
	•	Errors in logs:

`ERROR (MainThread) [async_upnp_client.ssdp] Received error: [Errno 101] Network unreachable`

	•	dmesg shows normal enumeration for the USB device, with no direct error messages about the USB.
	•	Once the system is fully booted (without USB connected), re-inserting the USB device does not cause network issues.

`2025-02-26 19:01:43.616 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-5074' coro=<Addon.watchdog_container() done, defined at /usr/src/supervisor/supervisor/addons/addon.py:1530> exception=AddonsJobError('Rate limit exceeded, more than 10 calls in 0:30:00')>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/addons/addon.py", line 1544, in watchdog_container
    await self._restart_after_problem(event.state)
  File "/usr/src/supervisor/supervisor/jobs/decorator.py", line 291, in wrapper
    raise on_condition(
        f"Rate limit exceeded, more than {self.throttle_max_calls} calls in {self.throttle_period(group_name)}",
    )
supervisor.exceptions.AddonsJobError: Rate limit exceeded, more than 10 calls in 0:30:00`
```

### Anything in the Host logs that might be useful for us?

```txt
none that I can find.
```

### System information

## System Information

version | core-2025.2.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.6.62-haos-raspi
arch | aarch64
timezone | Europe/Bucharest
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 1557
Downloaded Repositories | 29

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.02.1
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 916.2 GB
disk_used | 102.6 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | yellow
supervisor_api | ok
version_api | ok
installed_addons | Matter Server (7.0.0), MariaDB (2.7.2), OpenThread Border Router (2.13.0), Glances (0.21.1), InfluxDB (5.0.2), Grocy (0.24.0), Hikvision Doorbell (3.0.19), Log Viewer (0.17.1), File editor (5.8.0), Grafana (10.4.2), ESPHome Device Builder (2025.2.0), Advanced SSH & Web Terminal (20.0.1), Node-RED (19.0.2), Mosquitto broker (6.5.0), Tailscale (0.25.0), Silicon Labs Multiprotocol (2.4.5), RPC Shutdown (2.5), Studio Code Server (5.18.3), Whisper (2.4.0), Speech-to-Phrase (1.3.0), Piper (1.5.2), Paradox Alarm Interface (3.6.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 10
-- | --
resources | 20
views | 46
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), end0 (enabled, default, auto), hassio (disabled), docker0 (disabled), vethd69981e (disabled), vethef7a99d (disabled), veth82c07ff (disabled), vethb2d843c (disabled), veth75eba30 (disabled), vetha1ce1dd (disabled), veth8fe7a1e (disabled), veth88f024c (disabled), veth2630736 (disabled), wpan0 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), end0 (192.168.8.201/24), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), vethd69981e (), vethef7a99d (), veth82c07ff (), vethb2d843c (), veth75eba30 (), vetha1ce1dd (), veth8fe7a1e (), veth88f024c (), veth2630736 (), wpan0 ()
ipv6_addresses | lo (::1/128), end0 (fd1e:c8fa:cf3f:bb4a:63ca:2d8f:1c73:89c7/64, fe80::858b:4052:fa12:9a88/64), hassio (fe80::42:a9ff:fecd:25ba/64), docker0 (fe80::42:89ff:fe76:35f7/64), vethd69981e (fe80::d048:6aff:fead:e51d/64), vethef7a99d (fe80::bc2f:e2ff:fe03:b09c/64), veth82c07ff (fe80::fc54:91ff:fed0:f89f/64), vethb2d843c (fe80::584f:baff:fecb:43c8/64), veth75eba30 (fe80::fcd3:98ff:fee1:2a36/64), vetha1ce1dd (fe80::e831:11ff:fefa:df07/64), veth8fe7a1e (fe80::805d:feff:fecc:7268/64), veth88f024c (fe80::b442:46ff:fe06:4dfa/64), veth2630736 (fe80::7000:61ff:fefa:3737/64), wpan0 (fdb2:d4a8:20a3:1ab0:0:ff:fe00:fc11/64, fd52:4246:bb69:1:ed9e:861d:e99c:bcb8/64, fdb2:d4a8:20a3:1ab0:0:ff:fe00:fc10/64, fdb2:d4a8:20a3:1ab0:0:ff:fe00:fc38/64, fdb2:d4a8:20a3:1ab0:0:ff:fe00:fc00/64, fdb2:d4a8:20a3:1ab0:0:ff:fe00:c00/64, fdb2:d4a8:20a3:1ab0:226a:a36:9d62:f46/64, fe80::30f9:c4d7:caa5:9553/64)
announce_addresses | 192.168.8.201, fd1e:c8fa:cf3f:bb4a:63ca:2d8f:1c73:89c7, fe80::858b:4052:fa12:9a88

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | February 15, 2025 at 18:26
-- | --
current_recorder_run | February 26, 2025 at 19:11
estimated_db_size | 5042.56 MiB
database_engine | mysql
database_version | 10.11.6

</details>

### Additional information

	•	Setting a static IP can help sometimes but still fails occasionally. At least with a static IP it is always responding to ping (even if when its not accessible through web browser or ssh)
	•	This is not limited to a USB serial adapter; any USB peripheral (mouse, hub, etc.) triggers the same network failure at boot.
	•	A “wait for network” approach (ha network update ... --wait) is not recognized or available in this HA OS version.
	•	Suggests a deeper conflict in USB enumeration vs. DHCP initialization on CM5-based HA OS.
	•	Network: DHCP environment (works fine on other devices)
	•	Behavior: Only fails if any USB device is connected at boot time
-  I'm using https with self-signed certificates. No external access configured. No problem with that in the past year.

sairon, 2025-03-03:
I am not able to reproduce such behavior - no matter what I connect to the USB ports, the Yellow always gets a DHCP lease. Can you connect to your Yellow over the serial console (see [Windows](https://yellow.home-assistant.io/guides/use-serial-console-windows/) or [Linux/macOS](https://yellow.home-assistant.io/guides/use-serial-console-linux-macos/) instructions) and check the output of `systemctl status NetworkManager`? Also, full `dmesg` output of the boot would help.

Swalabrr, 2025-03-09:
Meanwhile I've uninstalled o lot of addons and HACS addons and at some point this behaviour stopped. If there are logs that I can recover, I'd gladly do so, but I need a little help with where should I look for them.

HarryVasanth, 2025-03-18:
@Swalabrr: Seems that the issue is caused probably due to the race condition between USB initialization and DHCP requests. The DHCP client may start before USB controllers are fully initialized, causing network setup to fail. Also, some HACS add-ons may attempt premature network access, triggering cascading failures (most likely the case by looking at the Supervisor rate-limit errors).

### After you re-enable/re-add your add-ons and booting back up, can you please check the following:

#### Network status
```sh
ha network info
```
This will show active interfaces and DHCP status.

#### Boot logs for USB/Network timing

```sh
dmesg | grep -E 'usb|NetworkManager|eth0'
```
Look for USB enumeration completion time vs. network interface activation.

or like @sairon requested just do the following to get the entire boot log

```sh
dmesg
```

closed 2025-06-28 by github-actions[bot]