home-assistant/operating-system #3911: Unable to boot Home Assistant on a Pi4 with SD Card using 11 Feb 2025 EEPROM firmware

ric355, 2025-03-03:
### Describe the issue you are experiencing

While trying to migrate from an SD card on a Pi4 (2Gb model) to a USB SSD, I upgraded to the latest EEPROM boot firmware. After I did this, I wanted to go back to booting from the SD card, and found that the device refused to boot.  I got 7 green flashes, which translates to the kernel can't be found.

On restoring the eeprom to factory firmware, I could boot from the SD card again. I did not change the SD card in any way (same physical device throughout, no changes to the OS or anything else on the card at all).

So it looks like somewhere along the way, RPi have broken the ability to boot Home Assistant. This might be specific to my device being a reasonably early one; I don't know. 

I haven't tried other recent versions of the firmware as they don't seem to provide a means to do that, so I can't tell when it was broken.

Chronology;
- Before upgrading the firmware, everything was fine - I could reboot at will.
- After upgrading the firmware, the SD I had been using would not boot. It was shut down properly previously.
- After upgrading the firmware, I could install a PiOS build on another SD card, same size and brand (Sandisk Ultra 32gb) and that would boot without issue.
- I could partially boot from the USB3 SSD after the firmware upgrade, but there were other hardware issues stopping me from getting it working properly. However, it did actually go through the boot process albeit with errors.
- Once I downgraded the firmware, the original, untouched SD card would boot again.

It seems like the latest firmware does not like the boot mechanism HA uses when it's on an SD card, as it's fine with a standard Raspberry Pi image and fine with the HA system on a USB SSD.

I don't really know who (if anyone) should fix this - it seems like RPi are the ones who broke it? Just to be clear, if you use the image provided in the Raspberry Pi Imager to do the eeprom firmware upgrade, and then use the latest Home Assistant also from the Imager, it won't boot (on my Pi4 at least; it's a reasonable early one).

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 14.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Pi4 2Gb early model
32gb Sandisk Ultra SD card

### Steps to reproduce the issue

1. Install Home assistant on SD card using pi imager
2. Boot and get it all up and running
3. Upgrade Pi4 to latest firmware using their boot image method via RPi Imager - for me that's 11 Feb 2025. Any of the boot options is the same either USB then SD or SD then USB. Obviously you only want the SD inserted and not a USB bootable device.
4. Try to boot the SD card again - it will fail with green LED showing 7 flashes (kernel missing) and rainbow screen.
5. Revert the eeprom firmware to factory using raspi-config. SD card will now boot fine.

### Anything in the Supervisor logs that might be useful for us?

```txt
No, this is a boot issue so nothing to show here.
```

### Anything in the Host logs that might be useful for us?

```txt
No, this is a boot issue so nothing to show here.
```

alexbatis, 2025-03-04:
im experiencing the same on raspberry pi 4, booting from 128gb sd card flashed with raspberry pi imager v1.8.5 using Home Assistant OS 14.2 (RPI 4/4000)

i was on the latest eeprom bootloader (2025/02/11)
```
$ vcgencmd bootloader_version
2025/02/11 17:00:13
version 75c1e570e99e14684e79e6ab8f5355d831440c71 (release)
timestamp 1739293213
update-time 1741029879
capabilities 0x0000007f
```

i flashed another SD card with raspberry pi OS. it booted with no issues.

In raspberry pi os i manually downloaded/updated the eeprom boot loader to [an arbitrarily older version from September 2024](https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware-2712/old/latest/pieeprom-2024-09-05.bin)

```
sudo rpi-eeprom-update -d -f ./pieeprom-2024-09-05.bin
```

after restarting my system (`sudo restart`) i observed that my `bootloader_version` had been successfully been downgraded to the September 2024 version

```
$ vcgencmd bootloader_version
2024/09/05 19:52:07
version 7fe8d0296644ec57f6c29eaa7ea2255571147af4 (release)
timestamp 1725562327
update-time 1741116378
capabilities 0x0000007f
```

finally, i swapped the sd card flashed with raspberry pi os with the sd card flashed with home assistant and it was able to successfully boot home assistant os.

sairon, 2025-03-05:
It seems to be a known issue triggered by the partition layout that HAOS uses, more details are in the issue linked above. I am hoping for a quick stable release of firmware that would fix it, if not, I'll come up with a way how to pull the needed files selectively (we can't simply bump the rpi-firmware revision to the latest one, as it will break other things).

Downgrading to an older version is one possible workaround (I have updated @alexbatis's post, it was incorrectly linking firmware binary for BCM2712/RPi5), only 2025-02-11 appears to be breaking so far, [2024-12-07](https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware-2711/latest/pieeprom-2024-12-07.bin) still works. Another way is taking `start4.elf` and `fixup4.dat` from https://github.com/raspberrypi/firmware/commit/9995946dce284505d3dd46c7563d74a691f38a70 and replacing the ones on the first (boot) partition on the SD card.

PR #3913 (2025-03-05): Update RPi firmware to fix boot with 2025-02-11 bootloader
* buildroot 7d5c3b5e70...b079a02a9a (1):
  > package/rpi-firmware: bump to 1.20250305 with updated firmware binaries

Fixes #3911

## Summary by CodeRabbit

- **Chores**
  - Updated an external dependency to its latest stable version, ensuring the project tracks current improvements and maintains overall system stability.
commit d4e11afd (2025-03-05): Update RPi firmware to fix boot with 2025-02-11 bootloader (#3913)

* buildroot 7d5c3b5e70...b079a02a9a (1):
  > package/rpi-firmware: bump to 1.20250305 with updated firmware binaries

Fixes #3911
commit 6e8c3900 (2025-03-13): Update RPi firmware to fix boot with 2025-02-11 bootloader (#3913)

* buildroot 7d5c3b5e70...b079a02a9a (1):
  > package/rpi-firmware: bump to 1.20250305 with updated firmware binaries

Fixes #3911

(cherry picked from commit d4e11afd3c98619bb83c19bee876902f794f8376)
closed 2025-03-05 by sairon