home-assistant/operating-system #3915: 15.0rc1: RaspberryMatic add-on stops working due to `Operation not permitted`

jens-maus, 2025-03-05:
### Describe the issue you are experiencing

While testing the upcoming 15.0 version of HAos by installing the current 15.0rc1 in a Proxmox (OVA) environment I recognized that the RaspberryMatic add-on does not start up correctly anymore and returns errors upon startup:

``` 
Identifying Homematic RF-Hardware: ....HmRF: HMIP-RFUSB/eQ-3 HmIP-RFUSB@usb-0000:01:1b.0-1, HmIP: HMIP-RFUSB/eQ-3 HmIP-RFUSB@usb-0000:01:1b.0-1, OK
Updating Homematic RF-Hardware: HMIP-RFUSB: 4.4.18, not necessary, OK
Starting hs485dLoader: disabled
Starting xinetd: OK
Starting eq3configd: OK
Starting lighttpd: OK
Starting ser2net: disabled
Starting ssdpd: OK
Starting ha-proxy: OK
Starting NUT services: disabled
Initializing Third-Party Addons: OK
Starting LGWFirmwareUpdate: not required
Setting LAN Gateway keys: not required
Starting hs485d: disabled
Starting multimacd: ..............ERROR
Starting rfd: ....................ERROR
Starting HMIPServer: .......................................................................................................................................................ERROR
```

After some investigation/debugging, I found out that these errors seem to be related to permission issues within the add-on docker container of the RaspberryMatic add-on when trying to access its special `/dev/eq3loop` device which is generated by the addon itself when doing e.g. a `modprobe eq3_char_loop`. In fact when starting the internal `multimacd` service to generate `/dev/mmd_bidcos` and `/dev/mmd_hmip`, the `/dev/eq3loop` seem to end up in a situation where the docker container will lose permission to access the loop device. See here:

[image]

When downgrading / reinstalling 14.2 the add-on works immediately correct again and when performing the same test sequence no `Operation is not permitted` error appears anymore:

[image]

So to me it seems, that something in the host OS is blocking access to the `/dev/eq3loop` device as soon as the `multimacd` daemon creates its secondary `/dev/mmd_bidcos` and `/dev/mmd_hmip` devices.

Thus, it seems some changes in the upcoming 15.0 (docker bump to v28 ?) seem to have changed the permission scheme which now prevents the eq3loop device to work correctly anymore.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

15.0rc1

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

VM installed in Proxmox 8.3.3

### Steps to reproduce the issue

1. Install fresh HassOS VM based on 15.0rc1
2. Install RaspberryMatic Addon
3. Start Add-on and monitor startup log

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
n/a
```

Baxxy13, 2025-03-06:
This should be checked and fixed before HA-OS 15 is released.
Otherwise, many users of the RaspberryMatic add-on will be left with a non-functioning RaspberryMatic.

jens-maus, 2025-03-06:
> Unfortunately, I don't have the hardware available, but I will get in touch with Jens to track it down. It's definitely something that needs to be resolved for the release.

Thanks! Just check your Discord messages ;-)

> (tagging as board/ova, but other boards are very likely affected too)

This is indeed board unrelated because the issue recently referenced up there related to HomeAssistant Yellow, thus rpiX platform.

commit 8aae5c05 (2025-03-11): Fix runtime device permissions update in runc v1.2.x

Add missing patch and update for latest runc version to fix losing device
permissions when new devices are added in runtime.

* buildroot b079a02a9a...3914f8cad5 (2):
  > package/runc: add patch for extended default allowed devices in v1.2.4
  > package/runc: add missing patch to fix device permissions update

Fixes #3915
PR #3921 (2025-03-11): Fix runtime device permissions update in runc v1.2.x
Add missing patch and update for latest runc version to fix losing device permissions when new devices are added in runtime.

* buildroot b079a02a9a...3914f8cad5 (2):
  > package/runc: add patch for extended default allowed devices in v1.2.4
  > package/runc: add missing patch to fix device permissions update

Fixes #3915

## Summary by CodeRabbit

- **Chores**
  - Updated a core underlying component to its latest version, helping maintain the system's current state and ensuring continued stability.
commit 04debe2f (2025-03-11): Fix runtime device permissions update in runc v1.2.x (#3921)

Add missing patch and update for latest runc version to fix losing device
permissions when new devices are added in runtime.

* buildroot b079a02a9a...3914f8cad5 (2):
  > package/runc: add patch for extended default allowed devices in v1.2.4
  > package/runc: add missing patch to fix device permissions update

Fixes #3915
commit 7e6e4119 (2025-03-13): Fix runtime device permissions update in runc v1.2.x (#3921)

Add missing patch and update for latest runc version to fix losing device
permissions when new devices are added in runtime.

* buildroot b079a02a9a...3914f8cad5 (2):
  > package/runc: add patch for extended default allowed devices in v1.2.4
  > package/runc: add missing patch to fix device permissions update

Fixes #3915

(cherry picked from commit 04debe2f539c4275fd68fe4d38fee3f8b8bab84b)
closed 2025-03-11 by sairon