home-assistant/operating-system #3920: Disabling IPv4 causes problems with default components

gittygitnice, 2025-03-10:
### Describe the issue you are experiencing

Disabling IPv4 in the settings of Home Assistant (Settings->System->Network) and leaving only IPv6 enabled doesn't disable IPv4 completly. Two virtual network interfaces of Home Assistant ("docker0" and "hassio") still get private IPv4 addresses from the Class B range assigned (as can be seen in the attached screenshot). This leads to problems as other components like SSDP pick up these addresses, try to work with them and throw an error message ("[Errno 101] Network unreachable") in the log as the main NIC no longer handles IPv4 addresses.

I first posted  about this issue here: https://github.com/StevenLooman/async_upnp_client/issues/259. At that time it wasn't clear what causes the problems.

[image]

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

14.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Running HA in Proxmox 8.3

### Steps to reproduce the issue

1. Disable IPv4 and leave only IPv6 enabled in the settings of Home Assistant. Restart the system.
2. The command "ip a" shows that two virtual network interfaces still get assigned IPv4 addresses.
3. The log shows the error "[Errno 101] Network unreachable, transport: <_SelectorDatagramTransport fd=22 read=polling write=<idle, bufsize=0>>, socket: <asyncio.TransportSocket fd=22, family=2, type=2, proto=0, laddr=('0.0.0.0', 52238)>", repeating every ten minutes, as SSDP picks up the private IPv4 addresses and tries to work with them which no longer is possible.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

### Additional information

Are the IPv4 addresses for the two virtual network interfaces internally needed by Home Assistant? Otherwise it would be helpful if they are disabled when disabling IPv4 in the settings as they cause problems otherwise. Thank you!

agners, 2025-07-01:
@davidrapan recently implemented IPv6 for Docker, you can enable it using `ha docker options --enable-ipv6=true`. Currently there is no way to disable IPv4, but I guess we could add a similar flag `--enable-ipv4`, which the user then can set to false.

But this is really a feature request.

@gittygitnice can you create a feature request at https://github.com/orgs/home-assistant/discussions/categories/home-assistant-os-supervisor? Thanks!

gittygitnice, 2025-07-25:
Thanks, I created a feature request here: https://github.com/orgs/home-assistant/discussions/441