home-assistant/operating-system #3935: Building addons from dockerfile doesn't use docker buildkit

felipecrs, 2025-03-17:
### Describe the issue you are experiencing

When building addons from dockerfiles that relies on certain instructions, it fails because the dockerfile isn't being built with buildkit.

Buildkit is the default builder in docker for several years already, since [23.0.0](https://github.com/moby/moby/releases/tag/v23.0.0) (we are on 28.0.1).

I would say everyone expects buildkit to be in use at this point by whatever system that is kept up to date.

Another inconsistency is that Supervised installs do have buildkit enabled, which may lead some addon authors to write dockerfiles which are not compatible to all installations.

Yet another one is that buildkit is enabled when running `docker build` from the [Advanced Terminal addon](https://github.com/hassio-addons/addon-ssh).

Buildkit brings huge benefits either through [new features and performance improvements](https://docs.docker.com/build/buildkit/#overview).

Also, it should be 99.9% backwards compatible. The risk of it not being compatible with some existing dockerfile is **very low**.

To fix this, either [`docker buildx` component](https://github.com/docker/buildx) should come installed in HAOS (I believe `docker` automatically uses `buildx` for `docker build` when it's found installed, and `buildx` always uses `buildkit`).

Or `buildkit` can be enabled in docker `/etc/docker/daemon.json` `{"features": {"buildkit": true}}`, or `DOCKER_BUILDKIT=1` environment variable can be set while running `docker build`.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

15.0

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

i5-12100T

### Steps to reproduce the issue

Create a local addon with a dockerfile that depends on some buildkit feature:

```dockerfile
FROM alpine

RUN --mount=type=bind,source=scripts/prepare_image.sh,target=/tmp/prepare_image.sh \
  /tmp/prepare_image.sh
```

And attempt to build it from supervisor. It should fail with something like:

```
The --mount option requires BuildKit. Refer to https://docs.docker.com/go/buildkit/ to learn how to build images with BuildKit enabled
```

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System information

## System Information

version | core-2025.3.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.12.18-haos
arch | x86_64
timezone | America/Sao_Paulo
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4925
Installed Version | 2.0.5
Stage | running
Available Repositories | 1593
Downloaded Repositories | 15

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.0
-- | --
update_channel | stable
supervisor_version | supervisor-2025.03.3
agent_version | 1.7.2
docker_version | 28.0.1
disk_total | 234.0 GB
disk_used | 47.3 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | AdGuard Home (5.2.5), Advanced SSH & Web Terminal (20.0.2), Asterisk (5.1.0), Cloudflared (5.2.10), ESPHome Device Builder (2025.2.2), Frigate (0.15.0), Glances (0.21.1), MariaDB (2.7.2), Microsoft STT (1.2.1), Microsoft TTS (1.2.0), Mosquitto broker (6.5.0), Netdata (2.2.6-addon.0), Nginx Proxy Manager (1.0.1), Studio Code Server (5.18.3), Traccar (0.25.0), WireGuard (0.11.0), Duck DNS (1.18.0), sshuttle-server (0.1.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 7
-- | --
resources | 4
views | 10
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), enp2s0f1 (enabled, default, auto), hassio (disabled), docker0 (disabled), vethf0f319c (disabled), vethdda6579 (disabled), vethf734ce4 (disabled), veth03db1a6 (disabled), vethc15cb0f (disabled), vethe1f9690 (disabled), vethc20a87e (disabled), veth71e2d33 (disabled), vethf4d648e (disabled), veth2f56a81 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp2s0f1 (192.168.1.10/24), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), vethf0f319c (), vethdda6579 (), vethf734ce4 (), veth03db1a6 (), vethc15cb0f (), vethe1f9690 (), vethc20a87e (), veth71e2d33 (), vethf4d648e (), veth2f56a81 ()
ipv6_addresses | lo (::1/128), enp2s0f1 (fe80::ed3f:7aae:fe43:4e2d/64), hassio (fe80::fc87:88ff:fe3d:945e/64), docker0 (fe80::1c59:54ff:fe73:9e83/64), vethf0f319c (fe80::104e:96ff:fe84:d12e/64), vethdda6579 (fe80::c0fc:54ff:feee:4059/64), vethf734ce4 (fe80::3409:88ff:fe7d:7ec3/64), veth03db1a6 (fe80::2cb5:34ff:fe92:a04b/64), vethc15cb0f (fe80::d88d:d6ff:feed:dead/64), vethe1f9690 (fe80::184a:25ff:feb9:ed95/64), vethc20a87e (fe80::c4da:49ff:feaa:315b/64), veth71e2d33 (fe80::5cf7:f8ff:fe03:f957/64), vethf4d648e (fe80::d068:61ff:fe23:b36c/64), veth2f56a81 (fe80::9003:11ff:fe63:4ea/64)
announce_addresses | 192.168.1.10, fe80::ed3f:7aae:fe43:4e2d

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | March 6, 2025 at 2:13 AM
-- | --
current_recorder_run | March 17, 2025 at 11:01 AM
estimated_db_size | 852.49 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

<details><summary>Sonoff</summary>

version | 3.8.2 (c4b6fda)
-- | --
cloud_online | 1 / 1
local_online | 1 / 1

</details>

sairon, 2025-03-28:
The documentation and ADRs are quite vague about this, so I've discussed this with @pvizeli and it's not something we really want to support. Building add-ons on the fly should ideally only be used for development, published add-on should have a pre-built image. GH Actions for public repositories are free so it shouldn't be much limiting.

Apart from that, the buildx plugin is around 65MB - that's something we can't currently fit in the OS image itself and caters only a rather niche use case, so any effort at doing it is hardly justifiable. However, I can imagine such functionality could be provided through a plugin (see for example how the audio stack is implemented). But that still requires some architecture decisions and planning.

> Or `buildkit` can be enabled in docker `/etc/docker/daemon.json` `{"features": {"buildkit": true}}`, or `DOCKER_BUILDKIT=1` environment variable can be set while running `docker build`.

This would still require the buildx plugin to be installed, so the above applies.

felipecrs, 2025-03-28:
@sairon, I appreciate the time you've taken on this.

However, this will become unavoidable eventually:

```
DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            Install the buildx component to build images with BuildKit:
            https://docs.docker.com/go/buildx/
```

And I would like to mention that for some add-ons it makes perfect sense to not publish an image and instead rely on local building.

Imagine an add-on that extends from an existing image but adds some script file to it.

The cost of building it is pretty much the same as pulling the image, but to pull the image:

1. An extra step from add-on developer has to be taken (to either automate build/push or do it manually)
2. If automated, CI cycles are spent (which may be free for the author, but it's not free to the world)
3. Disk consumption in the container registry (which may also be free for the author, but it's not free to the world)