home-assistant/operating-system #3938: Update to HAOS 15.0 fails on Home Assistant Blue / Hardkernel ODROID-N2/N2+

nagi121, 2025-03-17:
### Describe the issue you are experiencing

Update fails with these log entries at Supervisor log:

`2025-03-17 17:35:39.542 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-15.0.raucb
2025-03-17 17:35:40.618 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed updating slot boot.0: failed to run slot hook: Child process exited with code 1`

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

6.6.66-haos

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

USB-Devices:
usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0
/dev/disk/by-id/ata-Intenso_SSD_3832440-532110003
/dev/disk/by-id/mmc-SC32G_0x2c72efda
external Bluetooth adapter

### Steps to reproduce the issue

1. Click on Settings > Update > 
2. Select "Home Assistant Operating System Update"
3. Click "Update"
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2025-03-17 17:35:39.542 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-15.0.raucb
2025-03-17 17:35:40.618 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed updating slot boot.0: failed to run slot hook: Child process exited with code 1
```

### Anything in the Host logs that might be useful for us?

```txt
2025-03-17 16:35:39.546 ha rauc[406]: input bundle: /mnt/data/supervisor/tmp/hassos-15.0.raucb
2025-03-17 16:35:39.583 ha kernel: loop0: detected capacity change from 0 to 290284
2025-03-17 16:35:39.607 ha kernel: loop0: detected capacity change from 290284 to 290280
2025-03-17 16:35:39.639 ha kernel: device-mapper: verity: sha256 using implementation "sha256-ce"
2025-03-17 16:35:39.879 ha kernel: loop1: detected capacity change from 0 to 49152
2025-03-17 16:35:39.899 ha rauc[7757]: cp: can't stat '/mnt/boot/EFI/BOOT/grubenv': No such file or directory
2025-03-17 16:35:40.237 ha kernel: I/O error, dev mmcblk0, sector 16812 op 0x1:(WRITE) flags 0x800 phys_seg 3 prio class 2
2025-03-17 16:35:40.237 ha kernel: I/O error, dev mmcblk0, sector 16856 op 0x1:(WRITE) flags 0x800 phys_seg 15 prio class 2
2025-03-17 16:35:40.245 ha kernel: Buffer I/O error on dev mmcblk0p1, logical block 472, lost async page write
2025-03-17 16:35:40.253 ha kernel: Buffer I/O error on dev mmcblk0p1, logical block 473, lost async page write
2025-03-17 16:35:40.261 ha kernel: Buffer I/O error on dev mmcblk0p1, logical block 474, lost async page write
2025-03-17 16:35:40.269 ha kernel: Buffer I/O error on dev mmcblk0p1, logical block 475, lost async page write
2025-03-17 16:35:40.289 ha rauc[7758]: cp: write error: Input/output error
2025-03-17 16:35:40.291 ha kernel: I/O error, dev mmcblk0, sector 16812 op 0x1:(WRITE) flags 0x800 phys_seg 1 prio class 2
2025-03-17 16:35:40.409 ha kernel: I/O error, dev mmcblk0, sector 16436 op 0x1:(WRITE) flags 0x800 phys_seg 1 prio class 2
2025-03-17 16:35:40.409 ha kernel: Buffer I/O error on dev mmcblk0p1, logical block 52, lost sync page write
2025-03-17 16:35:40.483 ha kernel: I/O error, dev mmcblk0, sector 16956 op 0x1:(WRITE) flags 0x800 phys_seg 2 prio class 2
2025-03-17 16:35:40.578 ha rauc[7758]: cp: write error: Input/output error
2025-03-17 16:35:40.579 ha kernel: I/O error, dev mmcblk0, sector 16996 op 0x1:(WRITE) flags 0x800 phys_seg 1 prio class 2
```

### System information

## System Information

version | core-2025.3.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.6.66-haos
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 1594
Downloaded Repositories | 16

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 25

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.1
-- | --
update_channel | stable
supervisor_version | supervisor-2025.03.3
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 234.7 GB
disk_used | 43.4 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.16.0), AppDaemon (0.16.7), File editor (5.8.0), Mosquitto broker (6.5.0), Node-RED (19.0.2), Zigbee2MQTT (2.1.3-1), Advanced SSH & Web Terminal (20.0.2), ESPHome Device Builder (2025.2.2), Home Assistant Google Drive Backup (0.112.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 13
views | 13
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), end0 (enabled, default, auto), docker0 (disabled), hassio (disabled), veth10fdb76 (disabled), vethbbddc5b (disabled), veth9e097f2 (disabled), veth89e8d4b (disabled), veth927c8ef (disabled), vethe77c196 (disabled), vethc742fed (disabled), vethd72aeaf (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), end0 (192.168.12.12/24), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), veth10fdb76 (), vethbbddc5b (), veth9e097f2 (), veth89e8d4b (), veth927c8ef (), vethe77c196 (), vethc742fed (), vethd72aeaf ()
ipv6_addresses | lo (::1/128), end0 (fe80::ff36:7a34:a8d5:d34d/64), docker0 (fe80::42:bfff:fe00:b9f9/64), hassio (fe80::42:4ff:fe90:3ec3/64), veth10fdb76 (fe80::701b:fcff:fe36:7d6b/64), vethbbddc5b (fe80::a00f:1aff:fe54:9c08/64), veth9e097f2 (fe80::6435:13ff:fe01:7b24/64), veth89e8d4b (fe80::6071:3aff:fe43:d317/64), veth927c8ef (fe80::ec6e:faff:fe74:28e6/64), vethe77c196 (fe80::f0b7:e6ff:fea3:9324/64), vethc742fed (fe80::8a7:a3ff:febc:a8f1/64), vethd72aeaf (fe80::a45e:64ff:fe0e:e1ee/64)
announce_addresses | 192.168.12.12, fe80::ff36:7a34:a8d5:d34d

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 30. Januar 2025 um 06:11
-- | --
current_recorder_run | 17. März 2025 um 17:02
estimated_db_size | 4567.84 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

ortegaga, 2025-03-17:
Same here: after updating Home Assistant Blue to HAOS 15.0, it stopped responding; it doesn't even respond to pings. If you unplug it, wait, and plug it back in, it responds to pings again, but I assume it's trying to update again because after a while, it stops responding again.

SeByDocKy, 2025-03-17:
Yes update is quite long .... But on my odroid M1, it's solving the USB problem ... and seems more stable :)

nagi121, 2025-03-17:
> Same here: after updating Home Assistant Blue to HAOS 15.0, it stopped responding; it doesn't even respond to pings. If you unplug it, wait, and plug it back in, it responds to pings again, but I assume it's trying to update again because after a while, it stops responding again.

Sorry to say that: I think that's not the same.

My issue occurs during the update and the update is not complete.

nagi121, 2025-03-17:
> Yes update is quite long .... But on my odroid M1, it's solving the USB problem ... and seems more stable :)

It's good to hear that, but it doesn't solve this issue.

nagi121, 2025-03-17:
Directly after the failed update, HA was working normally and I created this issue.

Now, two hours later, HA stopped responding. No response to ping requests.

No change after a power cycle.

formatBCE, 2025-03-17:
Same problem. It's dead, no ping, no HDMI signal after update (i didn't touch it for 2 hours before trying to reboot, so it's not reboot in the middle).
That's crazy that we have things like this in 2025, guys :)

bfgiant, 2025-03-17:
I also have this issue. App is unable to connect to home assistant, and the device isn't even showing up on my network despite being wired in.

ginkage, 2025-03-18:
Upgrade to HassOS 15.0 killed my ODroid N2+, with symptoms similar to https://github.com/home-assistant/operating-system/issues/3227

ginkage, 2025-03-18:
Yes, gets stuck in U-Boot, with a screen just like in this comment: https://github.com/home-assistant/operating-system/issues/3227#issuecomment-2018130825 ("no mmc device at slot 2").
I'm using a Foresee 256 GB module.

nagi121, 2025-03-18:
I tried several haos image versions on two different SD-cards with no success.

I also tried to flash the SD-cards with SPI OTG. No go.

Flashing with balena, I can see on an an attached HDMI this screen:

[image]

This screen appears several times in a row and then stucks after 3 times.

nagi121, 2025-03-18:
Now, I flashed the 15.0 image with Win32DiskImager and the boot goes one step beyond:

[image]

albaintor, 2025-03-18:
> Now, I flashed the 15.0 image with Win32DiskImager and the boot goes one step beyond:
> 
> [image]

I had the exact same screen as yours : cycling TFTP from server..., then Loading:  T T T T T "
Reboots / unplugging the device would not resolve this.

It's like the MMC boot device could not be found or loaded, as if it was faulty.

Strangely, at one point I managed to make him boot but I don't know how.

sairon, 2025-03-18:
@nagi121 There's definitely something wrong with your boot device of flashing process. It actually didn't go "one step beyond" but rather back - in the second case it's not even able to find the kernel partition. In the first case, it finds it, but if the data in the kernel partition isn't flashed properly, it may lock up at the point you posted. Also a key information is missing here, what boot device are you using? An SD card or an eMMC module? What brand and model?

nagi121, 2025-03-18:
> [@nagi121](https://github.com/nagi121) There's definitely something wrong with your boot device of flashing process. It actually didn't go "one step beyond" but rather back - in the second case it's [...]

I use a SD card. Three different brands (amazon basics 64 GB, Noname 64 GB, SanDisk Ultra 32 GB).

ginkage, 2025-03-18:
Actually, regarding https://github.com/home-assistant/operating-system/pull/3878 I have a question: I see that the eMMC patch was moved from

`buildroot-external/board/hardkernel/patches/uboot/0001-HACK-mmc-meson-gx-limit-f_max-to-24-MHz-on-the-first.patch`

to

`buildroot-external/board/hardkernel/patches-meson/0001-HACK-mmc-meson-gx-limit-f_max-to-24-MHz-on-the-first.patch`.

But shouldn't it be, rather,

`buildroot-external/board/hardkernel/patches-meson/uboot/0001-HACK-mmc-meson-gx-limit-f_max-to-24-MHz-on-the-first.patch`

(note the `uboot` subfolder)?

sairon, 2025-03-18:
@ginkage 🤦Of course, that one's on us. It explains the other N2 related problems. Very good catch, thank you!

Unfortunately, even though I have a "problematic" setup in my test farm, the boot sometimes proceeds without errors, so I missed that 😢

albaintor, 2025-03-18:
Good news ! 
@sairon  one question please for users who already are on OS 15.0, will the next update revert the bootloader or do we have to flash them back to 14.2 manually  ?

sairon, 2025-03-18:
> Unfortunately, even though I have a "problematic" setup in my test farm, the boot sometimes proceeds without errors, so I missed that 😢

Just realized I only have the problematic setup with a Kingston module. But the missing patch triggered the regression on the other eMMC modules that misbehaved in the past (originally thought to be only legacy Micron 128 GB modules which are not available anymore, yet it seems the Foresee are broken too), and Kingston modules actually work fine in vanilla U-Boot without this patch...

@nagi121 For the sake of completeness, the issue you initially reported and problems you're seeing are unrelated to the one uncovered now.

@albaintor If you managed to boot after couple of tries, you should be fine until another reboot - when you'd eventually have to go through this again. You can downgrade to 14.2 with `ha os update --version 14.2` to get the fixed bootloader (note that using `ha os boot-slot other` wouldn't do that - U-Boot will stay the same as it's installed to the boot partition on OS update). Alternatively, you could also replace just the U-Boot binary in the boot partition and run OS 15 with the working U-Boot. Eventually, that also means that running `ha os boot-slot other` after downgrading to 14.2 will result in the same - you'll have old U-Boot with OS 15 running from the current slot.

sairon, 2025-03-18:
Let's track @nagi121's issue here and move to another one for the problem discovered by @ginkage:
- https://github.com/home-assistant/operating-system/issues/3942

---

@nagi121 As I wrote in the previous post, your issue appears to be unrelated. Have you tried flashing an older HAOS version with the same procedure to check it's only the OS 15 that doesn't work for you?

albaintor, 2025-03-18:
> Just realized I only have the problematic setup with a Kingston module. But the missing patch triggered the regression on the other eMMC modules that misbehaved in the past (originally thought to be [...]

Thank you it worked :

1. `ha os update --version 14.2` => reinstall the previous (working) U-boot version
2. `ha os boot-slot other` => lets upgrade to the new OS 15.0 while keeping working release of U-boot

nagi121, 2025-03-18:
Yes, I did. 14.1 and 11.4.

Now, it doesn't boot any more. 
With the switch to SPI, it shows petitboot. With switch to MMC and SD-Card containing haos inserted, red and blue light are solid lit, no output on HDMI.

sairon, 2025-03-18:
> Yes, I did. 14.1 and 11.4.
> 
> Now, it doesn't boot any more.

Sorry, I don't follow. That means that 14.1 and 11.4 have the same issue or they boot fine?

> With the switch to SPI, it shows petitboot. With switch to MMC and SD-Card containing haos inserted, red and blue light are solid lit, no output on HDMI.

It definitely needs to be switched to the MMC position, I suggest omitting that variable from your testing.

nagi121, 2025-03-18:
> Sorry, I don't follow. That means that 14.1 and 11.4 have the same issue or they boot fine?

All versions have the same issue.

> It definitely needs to be switched to the MMC position, I suggest omitting that variable from your testing.

I know. I tried SPI to flash a SD-Card. Also, I did a 'pb-update' there and changed the configuration "Boot Order = Any Device".

After that I could not boot from SD-Card any more (with switch at MMC position).

Then, I crated a SD-Card on a Raspberry with 
```
sudo dd if=/dev/zero of=/dev/sdd bs=4M
unxz -c haos_odroid-n2-11.4.img.xz | sudo dd of=/dev/sdd bs=4M status=progress
```

This SD boots to ha! After staring a while to the screen, several error messages appeared. The SD-Card seems to be bad. I have to try another one.

[image]

Or maybe an eMMC as alternative.

nagi121, 2025-03-18:
Which eMMC module should I buy?

ronaldtveen, 2025-03-18:
I'm confused on what to do now to fix it? 

I'm also running Home Assistant Blue and updated the OS late last night (eur/amsterdam). Now Home Assitant doesn't work anymore nor can I SSH into it. After a restart I can ping it for 30 seconds then it dies again.

I don't have a screen or keyboard connected (never had) which makes this difficult, but I understand that these two commands will fix it when I'm able to log into it?

> [..]
> 
> Thank you it worked :
> 
> 1. `ha os update --version 14.2` => reinstall the previous (working) U-boot version
> 2. `ha os boot-slot other` => lets upgrade to the new OS 15.0 while keeping working release of U-boot

What should I do to fix this? I hope it is not bricked?
Should I first revert back to a Raspberry Pi and install a recent backup or Is there hope for me to fix it via a couple of reboots or is there something different in the making?

Andrewletsdig18, 2025-03-18:
In same situation here Dell Wyse 5060 with ssd card installed  wont reboot left for 20mins powered off and on same problem 

strangly have a Dell Wise 5070 with same ssd type ( testing and dev ) that took v15 without problems

vb7445, 2025-03-18:
(I guess) Has HAOS 15 has been pulled back? 
My plan was to update this morning but it isn't showing anymore  on the Settings page.
I'm running HA on a Blue.

[image]

formatBCE, 2025-03-18:
> (I guess) Has HAOS 15 has been pulled back? 
> My plan was to update this morning but it isn't showing anymore  on the Settings page.
> I'm running HA on a Blue.
> 
> [image]

Well cheer for avoiding hassle :)

sairon, 2025-03-18:
@nagi121 All the clues currently point to a hardware issue - while we can't completely rule out bad SD card or problems in the flashing procedure, it looks less likely with follow up posts. It can be either problem with the SD card slot (e.g. damaged slot causing signal integrity issues) or damaged SoC on the board itself. If it's the latter, using eMMC likely wouldn't help either. At this point I'm pretty confident to say it's really unrelated to other software issues in OS 15.0.

As it was written couple of posts back, issues booting on some eMMC modules after the update to OS 15.0 are being discussed in #3942 (in case someone wonders why the posts were marked as off-topic here).

thesaofficer, 2025-03-18:
@sairon I'm getting exactly the same as @nagi121 on HA blue Odroid N2+ with eMMC immediately following attempted update to HAOS 15 - "No partition table MMC 0".   So surely it is related to the OS 15 update afterall.  I have no idea how to recover from this, except by removing eMMC and re-flashing.  Is that right?

nagi121, 2025-03-18:
> [@nagi121](https://github.com/nagi121) All the clues currently point to a hardware issue - while we can't completely rule out bad SD card or problems in the flashing procedure, it looks less likely [...]

Okay. I can follow you in some parts. But it's annoying that 4 SD-Cards, 3 flashing tools, the SD-Card slot or the SoC is damaged at the same time as the OS 15.0 Update was attempted to install.

I ordered an eMMC to exclude some of the possible reasons and will report the results. I also ordered a Pi 5 to make my wife happy again.

Thanks for your patience and your help so far.

ronaldtveen, 2025-03-18:
I've connected my Blue to HDMI and got the `** No partition table - mmc 1 **` error too. 
This happened right after updating to OS 15.0. I can't believe all the problems are because of faulty eMMC's and that it is unrelated to the OS 15.0 update, right? What are the odds...

[image]

If I understand correctly I can just keep unplugging power and plugging it back in until the boot succeeds after several attempts?

(It is literally getting dark in our house and I would like a game plan)

vb7445, 2025-03-18:
What about all the previous HAOS updates that went successfully and now all of sudden update 15 uncovers the issue? What's so special about 15?

thesaofficer, 2025-03-18:
Okay.  Thank you @sairon and @nagi121.  Not knowing much, but I was thinking the lack of a partition table explained why I couldn't even access the GRUB menu.  Sorry for introducing a red herring which others have picked up!  I tried a very great number of hard boots with no change.  But now, for anyone else in my position, I currently seem to be successfully following [this](https://www.home-assistant.io/installation/odroid#flashing-an-odroid-n2) guide to get a working HA OS and restore things back to my most recent backup

nagi121, 2025-03-19:
Some news.

I verified my SD cards with H2testw and Rufus (Rufus is great!) . All SD cards are error free.

`Warning: Only 30429 of 30430 MByte tested.
Test finished without errors.
You can now delete the test files *.h2w or verify them again.
Writing speed: 19.1 MByte/s
Reading speed: 19.7 MByte/s
H2testw v1.4
`

I flashed some other images (ubuntu and CoreElec) to SD card and they do not boot.
So, I can verify that the SD slot of my Odroid n2+ is defect. 

Then, I put the SD card with CoreElec into an USB adapter, sticked it into the Odroid, switched the Odroid to SPI and booted successfuly into CoreElec.

Same thing with haos (tried 8.2 and 15.0) does not work. Stucks at "[sda1] Processing new USB device".
Too bad, because this would be a nice solution. Perhaps you can get some inspiration from there ;-)

Tomorrow, I will get a new eMMC card and will try that.

ldharmon, 2025-03-19:
I wish I would have known about this.  I just updated my odroid m1 and it is not responding. (8gb 256 gb ssd)
I believe I tried running off an EMMC rather than SD-NVME and it didn't boot!
( I had used etcher to load HA via the bootloader.  But HA didn't boot for me.)  Perhaps I will try again

ldharmon, 2025-03-19:
I removed my sd and ssd then installed a 32gb emmc and flashed it with etcher.  after a reboot petiboot did not see the OS ???  Petiboot was acting funny, cursor keys not working???
I put my ssd and sd card back and rebooted.  To my surprise it booted and I was able to connect to HA with a web browser.
The web page says Preparing home assistant   may take 20 minutes...

closed 2025-03-21 by nagi121