home-assistant/operating-system #3940: after Update to OS15 HA is brutally slow, reconnects intermittenlty

vrtsman, 2025-03-17:
### Describe the issue you are experiencing

after updating to the latest HAOS15 my HA is so slow switching between the panels, config, etc. it takes so long you can grab a cup of coffee. Has never happend before in the last 2 years.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

15.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

intel i3 running as a vhdx (16GB RAM with SSD) 

### Steps to reproduce the issue

1. cant reproduce
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
access almost impossible
```

### Anything in the Host logs that might be useful for us?

```txt
right now hard to access
```

### Additional information

cant really get to the logs as the system is totally unresponsive and reconnects during access

vrtsman, 2025-03-17:
after downgrading to 14.2 issue resolved

vrtsman, 2025-03-17:
any advise before trying to update to OS15.0? I guess it will happen again with 15.0; What info do you need to troubleshoot on 15.0?

sairon, 2025-03-18:
Is the CPU hogged with 15.0? Even if you can't reliable access the Home Assistant frontend, you should be able to connect to it e.g. with the "Virtual Machine Connection" tool and check what's going wrong in the OS shell.

Parollo, 2025-03-18:
My HA green was also very difficult to reach after the update. The CPU load was over 30%. But after one night, the HA calmed down again by itself 😀. It is now running at 2% load again and everything is working normally again.

Ocramius, 2025-03-19:
Similar behavior since a reboot earlier today: `ha core restart --safe-mode` kinda gives me an usable environment, so I'm not 100% this isn't caused by custom integrations, at this moment.

sairon, 2025-03-20:
You can refer to this bdraco's guide about how to identify instability issues in Core: https://community.home-assistant.io/t/2024-5-tracking-down-instability-issues-caused-by-integrations/724441/1

chheiss, 2025-03-22:
Same problem here, OS 15 slow and several integrations loose connections (Fujitsu, Ondilo, Weatherflow, …) after a while. Downgrade to OS 14.2 solves issues.

vrtsman, 2025-03-23:
i dont konw if i want to ake the risk and update again. At the moment i dont have time to spend on troubleshooting. Will think about doing it again next week. I guess i will just create a VM snapshot before do some testings and then i will see

sairon, 2025-03-24:
@vrtsman Can you please at least share [system information](https://my.home-assistant.io/redirect/system_health/) from the working OS version? As the root cause of your case hasn't been identified, there's a risk that upcoming versions won't fix issues with your install.

WladyTee, 2025-03-25:
@sairon a similar behaving is still occurring to me, without VM installation.

My case is based on the out of memory crash that appears in less than 1 hour, depending on the integrations.
Random execution for Bluetooth scanning, Speedtest, Mobile Companion updates from remote are impacting the overall stability slowing down until reboot.
Please consider that Bluetooth integration has been modified (fixed?) and will be provided with 2025.4.

If it could help you, when comparing OS14.2 vs OS15, the System Monitor Processor Use is 10-15% higher with OS15.
I hope it helps even with a NUC installation without VM.

[image]

marshalleq, 2025-04-24:
I just upgraded to 15.2 last night and have been hit with this issue on Raspberry Pi 5.  I would describe it as a slow to react / slow to transmit kind of situation.  I'm using the official HA thread USB device with Nanoleaf downlight which have been working very well.  It seems to be far worse when operating via my Shelly display - as in it seems to work on the phone a lot better, so that could be a clue - it was bad enough that the Shelly device was actually going offline, the HA web page was unavailable, rebooting would bring inconsistent results before dying again that sort of thing.  I've downgraded to 15.1 and I'd say it's better but not perfect.  I am not sure, but I may have been one version behind when I updated.  Something weird going on here.