home-assistant/operating-system #3941: after update to OS 15 CPU load extremely high

rybackisback, 2025-03-17:
### Describe the issue you are experiencing

After updating to OS 15 CPU load has gone from 15-16% to 95%

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

15

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Intel RNUC11PAHI70000, Intel Core i7-1165G7 Processor
32 GB ram
Google Coral

### Steps to reproduce the issue

1. cant reproduce. noticed on undegrading to OS 15.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
none
```

### Anything in the Host logs that might be useful for us?

```txt
none
```

sairon, 2025-03-18:
Just to confirm, is this a bare-metal install of generic-x86-64 image? Can you check `top` (after typing `login` with keyboard and display connected) what eats the CPU?

dangel666, 2025-03-18:
I'm running on an N100 but with proxmox and did an upgrade to check CPU usage (easy to visualize in prox) and it looks pretty much the same even when I rollback via a snapshot.

Philap95, 2025-03-18:
I can confirm that the CPU load has increased after the update to OS 15. I use the qcow2 image via Proxmox on NUC13i5.
(See picture below)
<details>
  <summary>CPU-Load in PVE</summary>

[image]
(Update performed at about 7 pm.)
</details>

sairon, 2025-03-18:
@Philap95 And here's the CPU load for a Proxmox instance that runs on the latest RC (15.0.rc1 was released 2 weeks ago):

[image]

As you can see, it's fairly steady for the whole time. The update or the associated reboot possibly triggered something in your setup. There's already a suggestion above on how to identify what's causing it.

rybackisback, 2025-03-18:
> I'd recommend `top -co %CPU` instead so it's sorted by CPU and shows arguments. This can also be toggled by pressing `P`/`c` respectively. Alternatively install the `glances` addon, open it and press `z` and `c`.

See below screenshot from glances. Compreface addon seems to be consuming all of the the cpu load and frigate a second with os 15. I never saw this high cpu load in OS 14.2 with these addons.

[image]

sairon, 2025-03-18:
@rybackisback Well, there you go. Check the add-on logs to see what's going on there. Maybe it used some hardware acceleration that isn't working anymore in OS 15? If it's not the case, you should report an issue in the add-ons' issue trackers.

Philap95, 2025-03-18:
@sairon You may be right, yes I added a sensor via Zha just before the update but I can't imagine that's it. I'll restore a backup, look at the differences and carry out the update again, I'll let you know if anything comes up, thank you.

pszypowicz, 2025-03-18:
My high traffic was coming from the homeassistant container (`docker stats`)

tredick, 2025-03-18:
Can't see any CPU usage increase (the peak was the update and restart of the system at 14:32) - so I guess it might be an addon which is not used by me:

[image]

But the memory consumption has increased ... which I don't mind since there's enough room on my Proxmox server:

[image]

gfn256, 2025-03-18:
So I've also got an increased CPU usage after update to 15.0.

I'm running HAOS VM in Proxmox on an Intel i7-13800H. Update performed without incident. I then noticed a very small increase in  VM CPU usage going up by about 0.1 to 0.2% - so nothing to complain about.

BUT! - a few hours later I had to do some electrical maintenance (nothing related to HA!) so I had to power down the server for about an hour. Powered back on - HAOS VM booted perfectly, but CPU has now gone up by about 1.1%. You may think this is little, but this causes the CPU temp to have reason by 7-8 degrees Celsius and power draw has gone up by about 9-10 watts. That is not good in my books. I checked `top` but couldn't see much standing out except for `python3` using between 0.7 & 2.1%.

[image]

dangel666, 2025-03-19:
> Can't see any CPU usage increase (the peak was the update and restart of the system at 14:32) - so I guess it might be an addon which is not used by me:
> 
> [image]
> 
> But the memory consumption has increased ... which I don't mind since there's enough room on my Proxmox server:
> 
> [image]

Proxmox doesn't really show memory consumption correctly and this is a known 'feature' - so mine always looks like this:

[image]

but in reality it's this:

[image]

And this has been the case for me for some years of use now - it slowly climbs to max in proxmox and stays there.  In reality it doesn't do any such thing.

gfn256, 2025-03-19:
I actually now suspect that the increase in CPU load is smaller than I suggested above. What makes me believe so is the fact that after the actual HAOS 15.0 update the load only rose slightly, but after a complete host/node system shutdown & restart it rose significantly (see my above image). This leads me to believe that the major increase in load is actually caused by a recent Proxmox update to qemu 9.2.

I am investigating the matter further on this [thread](https://forum.proxmox.com/threads/qemu-9-2-available-on-pvetest-and-pve-no-subscription-as-of-now.161849/post-756793) on the Proxmox forums.

sairon, 2025-03-19:
@gfn256 It's not hard to perform couple of regression tests - you can reboot the VM and select the other boot slot in the GRUB menu in the Proxmox console to see if it changes in the different versions. Or you can use the `ha os boot-slot other` command to switch back and forth.

As already said above, memory usage view in Proxmox says nothing about Linux VMs. It shows the total memory allocation including buffers and caches which will be eventually freed by the Linux kernel for running processes. Furthemore, it means the upper green portion of the graph is memory that is currently allocated to the VM without any use whatsoever (but if you don't need it for other VMs, it doesn't do any harm).

eavogels, 2025-03-20:
I'm not sure who to ask, but since I'm not a Linux expert—just a Home Assistant user—I need some guidance.

I tried updating to version 15 on both my NUC and Dell Wyse, but each time, the update only forced a switch to another boot slot instead of completing successfully. I attempted the update multiple times on both machines, hoping it was just a temporary issue, but the result was always the same.

Would it be best to just wait for version 15.1? Reading about blacklists and bootloaders is making me a bit uneasy. Any advice would be greatly appreciated.

sairon, 2025-03-24:
The OP's issues is now better tracked in #3960, let's follow-up there.

closed 2025-03-24 by sairon