home-assistant/operating-system #3942: OS 15 on ODROID-N2 unable to boot from Foresee 256 GB eMMC

sairon, 2025-03-18:
> Yes, gets stuck in U-Boot, with a screen just like in this comment: https://github.com/home-assistant/operating-system/issues/3227#issuecomment-2018130825 ("no mmc device at slot 2").
> I'm using a Foresee 256 GB module. 

 _Originally posted by @ginkage in [#3938](https://github.com/home-assistant/operating-system/issues/3938#issuecomment-2731948367)_

ginkage, 2025-03-18:
That would be FEMDNN256G-A3A56 — and indeed it works just fine with OS 14.2, so yeah, seems like an U-Boot issue.

ginkage, 2025-03-18:
Yes, that's the one.

josemaX, 2025-03-18:
Same here but ODroid 128 eMMC "red square". After 4 or 5 "hard resets" without any video output it started and (seems) it is working as it must, EXCEPT create backup failed, then restarted HA and was able to create a backup. If helps someone.

sairon, 2025-03-18:
I adjusted title of this issue to track issues caused by the misplaced patch, as pointed out in https://github.com/home-assistant/operating-system/issues/3938#issuecomment-2732517130

This is known to affect not just the Foresee module but also older 128 GB eMMC modules and manifests as intermittent boot problems, with device stuck in U-Boot.

PR #3946 (2025-03-18): Move ODROID-N2/C2/C4 U-Boot patch to correct folder to fix eMMC issues
As pointed out by @ginkage, in #3878 the eMMC patch was moved to a wrong path. Move it to uboot subfolder so it's correctly applied.

Fixes #3942
commit fd2c16d5 (2025-03-18): Move ODROID-N2/C2/C4 U-Boot patch to correct folder to fix eMMC issues (#3946)

As pointed out by @ginkage, in #3878 the eMMC patch was moved to a wrong path.
Move it to uboot subfolder so it's correctly applied.

Fixes #3942